fwupd/firmware-lenovo #449: Failed to update ME firmware of ThinkPad T14s Gen 3

ITUNUOLUWA0, 2024-06-17:
Please i need help in fixing my laptop i have checked every plaform on google including the lenovo form and i really would the assistant 
i have trying to fix but anytime i on the i says reading ME firmware please dont power off and at 48 % is fails

kmauleon, 2024-06-18:
Hi @ITUNUOLUWA0 ,  i forwarded your issue to ODM owner... in the meantime can you please try the following...
SS BIOS Setup Menu so we can know current BIOS/EC version
Set BIOS to default settings
Place Linux-Firmware-Update to lowest boot priority

kmauleon, 2024-06-18:
as per ODM owner...         

May we know the bios version on customer system, thank you.
I guess bios version is less than N3CET55W(1.36).

ITUNUOLUWA0, 2024-06-18:
Actually the bios N3CET62W(1.43).

ITUNUOLUWA0, 2024-06-18:
Should I attempt a roll back or what cause it is the latest uefi Bois version 🤔

kmauleon, 2024-06-18:
u can try to downgrade but i don't think it's allowed... let me inform ODM regarding your BIOS version... BTW, can we have the SS of BIOS Menu?

ITUNUOLUWA0, 2024-06-18:
[image]

kmauleon, 2024-06-18:
got it... forwarded to ODM owner...

have you tried these..
- Set BIOS to default settings
- Place Linux-Firmware-Update to lowest boot priority

ITUNUOLUWA0, 2024-06-19:
Have done that already still the same thing

kmauleon, 2024-06-19:
how about fwupdmgr reinstall?

following up ODM owner...

kmauleon, 2024-06-20:
oh.. you can't boot into the OS itself...

how about reinstalling BIOS via USB bootable >> https://pcsupport.lenovo.com/ph/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-3-type-21br-21bs/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C

ITUNUOLUWA0, 2024-06-20:
Have download the latest bios already or is why for me roll back and then update the bios fron usb

kmauleon, 2024-06-20:
yes please try

kmauleon, 2024-08-21:
hi @ITUNUOLUWA0 any changes from your side?  are you still encountering the issue?  please advise... thank you very much

kmauleon, 2024-10-24:
@ITUNUOLUWA0 have you tried updating BIOS via USB?  Please advise.  Thank you very much.

kmauleon, 2024-12-23:
no update... closing for now... please let me know if need to reopen.... thank you very much.

closed 2024-12-23 by kmauleon