fwupd/firmware-lenovo #450: Missing firmware 1.23 for ThinkPad P1 Gen 5

Iridias, 2024-06-24:
Security-advisory lists 1.22 as affected: https://support.lenovo.com/de/de/product_security/LEN-158632

Official download page offers 1.23:
https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p1-gen-5-type-21dc-21dd/downloads/driver-list/component?name=BIOS
With release-date: 05 Jun 2024

Devicepage on LVFS only lists 1.22: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JET.firmware

mrhpearson, 2024-06-24:
1.23 is available on LVFS: https://fwupd.org/downloads/92278eb2755b25055f14651901653bcf60ae80782853a64974416d72eb06bfa3-Lenovo-ThinkPad-P1Gen5X1Extreme5-SystemFirmware-1.23.cab

Let me know if I'm missing something.
Mark

closed 2024-06-24 by mrhpearson

franck-iaropoli-arm, 2024-06-27:
Is this version installing properly for you @Iridias ?
It doesn't get installed on one of our device.

`$ sudo fwupdmgr get-history                                                                                        
21DDS6SP00                                                                                                                                                  
│                                                                                                                                    
├─Embedded Controller:                                            
│ │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e                                                                                          
│ │   Previous version:   0.1.11                                                                                                                            
│ │   Update State:       Failed                                              
│ │   Update Error:       failed to run update on reboot                                                                                                                                   
│ │   Last modified:      2024-06-27 14:11                                                                                                                                                 
│ │   GUID:               f8963b51-51b6-4059-87c4-722967c16745                                                       
│ │   Device Flags:       • Internal device                                                                                          
│ │                       • Updatable                                         
│ │                       • System requires external power source                                                                    
│ │                       • Supported on remote server                        
│ │                       • Needs a reboot after installation                                                                                                                              
│ │                       • Device is usable for the duration of the update                                                                                 
│ │                                                                                                                                                                                                                                       
│ └─ThinkPad P1 Gen 5/X1 Extreme Gen 5:                                                                                                                                                                                                   
│       New version:      0.1.12                                                                                                     
│       Remote ID:        lvfs                                                                                       
│       Release ID:       93501
│       Summary:          Lenovo ThinkPad P1 Gen 5/X1 Extreme Gen 5 System Firmware                                                                         
│       License:          Proprietary                                                                                                                        
│       Size:             1.1 MB                                              
│       Created:          2024-05-14                                          
│       Urgency:          High                                                
│       Vendor:           Lenovo                                              
│       Release Flags:    • Is upgrade                                        
│       Description:                                                          
│       Changes in this releae version 1.12                                   
│                                                                             
│       Important updates - Nothing                                           
│                                                                             
│       New functions or enhancements - Updated PDFW                          
│                                                                             
│       Problem fixes - Nothing                                               
│                                                                             
└─System Firmware:                                                            
  │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038                                                                                          
  │   Previous version:   0.1.22                                                                                                                             
  │   Update State:       Failed                                              
  │   Update Error:       failed to run update on reboot                      
  │   Last modified:      2024-06-27 14:11                                    
  │   GUID:               3a6cdf97-cfb9-479d-9c4c-135e0d4505c9                
  │   Device Flags:       • Internal device                                   
  │                       • Updatable                                         
  │                       • System requires external power source             
  │                       • Supported on remote server                        
[... cut ...]
        Urgency:          High                                                
        Vendor:           Lenovo                                              
        Release Flags:    • Is upgrade                                        
        Description:                                                          
        Changes in this release - version 1.23                                
                                                                              
        New functions or enhancements - Updated the Diagnostics module to version 04.34.001.`

Iridias, 2024-06-27:
Yes, it worked on my device.