fwupd/firmware-lenovo #454: Battery capacity reported as 0% after firmware upgrade on ThinkPad X13 Gen 1

Shunpoco, 2024-07-01:
I also reported at #329, but opened this issue for easy to find this issue for Gen 1.

Hi teams,

I'm using ThinkPad X 13 Gen 1 AMD (20UF), and I had the issue described in the title.
After upgrading its firmware, 
- The battery capacity reported as 0%
- Working without plugged in power is not possible
- I also booted with WIndows 11, but it also reported the battery capacity as 0% and can't work without plugged in power

Here is the all of the information when I ran `fwupdmgr get-devices`:

```bash
$ fwupdmgr get-devices
20UFS0KE00
│
├─Integrated Camera:
│     Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:    59.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              17d214a8-3e31-5f50-94b5-45bf35278459
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:       • Updatable
│   
├─Ryzen 5 PRO 4650U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         1f2e0c2f-e078-5fcb-a8de-f78eff9a470a
│                         5fd5eb41-dd4a-5742-8448-861eaaa61cf8
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.39
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   Update Error:       Cannot install update when system battery is not at least 25%
│ │   GUIDs:              428ba8ba-40c8-4446-8d7a-394d6c28b26a
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  312
│       Minimum Version:  312
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       Update Error:     Cannot install update when system battery is not at least 25%
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341
│                         5971a208-da00-5fce-b5f5-1234342f9cf7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Needs a reboot after installation
│                         • Updatable
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    1.257.0.0
│     Vendor:             ST Microelectronics (TPM:STM)
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         84df3581-f896-54d2-bd1a-372602f04c32
│                         bfaed10a-bbc1-525b-a329-35da2f63e918
│                         70b7b833-7e1a-550a-a291-b94a12d0f319
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48
│     Device Flags:       • Internal device
│   
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    1057227777
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUID:               83e0eecb-29be-468c-ba06-0b011e9d2d74
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    65575
│     Minimum Version:    65575
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUID:               4bea12df-56e3-4cdb-97dd-f133768c9051
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    65793
│     Minimum Version:    65793
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUID:               f7476d9b-7b01-4850-b255-0cc5227cdae6
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
│     Summary:            UEFI ESRT device
│     Current version:    22808
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUID:               ae4cd32d-732f-4170-a8f5-8cb4c5c4d0b8
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─UMIS RPJTJ256MEE1OWX:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    2.1C0628
      Vendor:             Union Memory (Shenzhen) (NVME:0x1CC4)
      Update Error:       Cannot install update when system battery is not at least 25%
      GUIDs:              0317581b-58d7-52b1-8ad7-6983aa66d3db
                          56c329eb-587d-5325-a46b-ce9be3cc1d37
                          66bde5bc-c9c2-562f-8ac0-45e1019e2578
                          0e5096da-b17e-5128-bce0-83f4b6773c4d
                          83f5c44e-578a-5a7a-8755-6e99234be8ba
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
 ```

mrhpearson, 2024-07-03:
For my reference: Tracking this issue with internal ticket LO-3154

mrhpearson, 2024-07-09:
Did you update the battery FW?
I checked with the FW team and there were battery FW update failure issues and the update got pulled (back last year - I checked and the FW is not in the stable channel so shouldn't be accessible).

For getting your system running - you need to check with the Lenovo support team to get your system fixed - hopefully that goes smoothly.

mrhpearson, 2024-07-11:
Confirmed the battery FW update was pulled as it caused issues if three updates in a row failed.
Please check with Lenovo support team on getting the system fixed.
If you can let me know where the update came from that would be appreciated.

Shunpoco, 2024-07-11:
@mrhpearson 
Thanks!
> If you can let me know where the update came from that would be appreciated.
I'm sorry, but I got this issue last year (and since then, I'm always using my ThinkPad with a charger), and I forgot how I did my update at that time.

I will contact to Lenovo support team, but how can I contact them? In addition, although the hardware warranty was already expired last year, can I post this issue?

mrhpearson, 2024-07-12:
I think you need to go through https://pcsupport.lenovo.com/us/en/contact-us and once you put in the serial number it gives you options?

Shunpoco, 2024-09-28:
Hi @mrhpearson ,

Sorry for the late reply because.
I tried to contact the support, but it said that I should send my X13 laptop to a technical center and should pay for postage, diagnosis, and repair. It is almost the same cost as the price of my laptop and it doesn't guarantee the problem will be solved. I don't think it is a good option for me.

Do you know when the firmware will be released on fwupd? Or, will it never be released for any reason?
I think the problem is clear (reinstall the battery firmware), if I get the program, I can try it myself.

mrhpearson, 2024-10-03:
Hi - sorry for the slow reply.

Afraid it's unlikely to be on LVFS - especially as it's older and caused problems.
I'm surprised they're giving you grief as the update itself caused the problem. Might be worth escalating...

As you are able to boot Windows - can you do the update from there?

Mark

Shunpoco, 2024-10-04:
Thanks, 
> As you are able to boot Windows - can you do the update from there?

Yes, I can also boot on Windows, I will try. Can I update the entire Firmware version or do you think there is a battery specific file?

mrhpearson, 2024-10-07:
Afraid I don't know (I don't use Windows at all except under duress). 
Can you see if Vantage offers you an update? If it doesn't I'll ask the battery team for guidance.
Mark

Shunpoco, 2024-10-11:
Hi Mark, sorry for the update

I tried to update BIOS through Windows, and the BIOS version was updated but the battery problem was not solved.

At first, I tried to update BIOS through Lenovo Commercial Vantage, but it fails with the error like "The update fails because the user canceled, the PC wasn't plugged into a power supply, or low battery." (I'm sorry because I'm using Vantage in Japanese, the message was translated by me).
So I decided to update BIOS manually. The steps are like:
1. Download exe file from [the support page](https://superuser.com/questions/1569417/how-to-update-lenovo-bios-on-a-t460-without-battery)
2. Execute the exe file only for extracting the update program.
3. Execute `Winuptp64.exe` with `-sp` option. The option makes the update skip checking battery percentage.

Eventually, the update was succeeded and the BIOS version is now 1.48 (current latest one), but my laptop still doesn't detect the battery.

mrhpearson, 2024-10-18:
Battery team asked to double confirm if this condition occurred by "the user unplugged the adapter when the system was updating?". Do you know if this was the case or not?

Shunpoco, 2024-10-18:
Hi, thanks for asking the battery team.

If I remember correctly, during the update I didn't unplugged the adapter and there was no temporary electricity disruption.

crab2313, 2025-09-25:
I'm hit by this issue today, why don't you just remove the firmware update from the testing channel ? If you know it will cause issue.

Shunpoco, 2025-10-06:
@mrhpearson 
Hi, I've received your email.
I tried, but so far it doesn't work well with `No supported devices found`
```bash
$ sudo fwupdmgr install --force Lenovo-ThinkPad-T14s-Gen-1-BatteryFirmware-258.0.0.37573.cab -vv
(fwupdmgr:13150): GLib-DEBUG: 21:39:16.686: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:13150): GLib-GIO-DEBUG: 21:39:16.707: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.707: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.707: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.707: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.707: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.707: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:13150): GLib-GIO-DEBUG: 21:39:16.707: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.715: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.715: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.715: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.715: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:13150): dconf-DEBUG: 21:39:16.715: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:13150): Fwupd-DEBUG: 21:39:16.762: Emitting ::status-changed() [idle]
(fwupdmgr:13150): Fwupd-DEBUG: 21:39:16.767: Emitting ::status-changed() [decompressing]
Decompressing…           [ -                                     ](fwupdmgr:13150): Fwupd-DEBUG: 21:39:16.768: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
No supported devices found
```
And get-devices doesn't have Battery info:

```bash
$ sudo fwupdmgr get-devices --show-all 
20UFS0KE00
│
├─Integrated Camera:
│     Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:    59.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              17d214a8-3e31-5f50-94b5-45bf35278459 ← USB\VID_04F2&PID_B6D0&REV_5918
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│     Device Flags:       • Updatable
│   
├─MassStorageClass:
│     Device ID:          c5c832237a38d9222f33d4db8cd17dca3b7da72e
│     Summary:            SCSI device
│     Current version:    1539
│     Vendor:             Generic (SCSI:Generic)
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              d66cc4ea-bb8b-548d-919a-c31be466803c ← SCSI\VEN_Generic&DEV_MassStorageClass&REV_1539
│                         304d0264-1b0b-55ec-bc62-c79aca351420 ← SCSI\VEN_Generic&DEV_MassStorageClass
│   
├─MassStorageClass:
│     Device ID:          6283c0e30e239931ab82f9d59e7dfdf16bbfc501
│     Summary:            SCSI device
│     Current version:    1539
│     Vendor:             Generic (SCSI:Generic)
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              d66cc4ea-bb8b-548d-919a-c31be466803c ← SCSI\VEN_Generic&DEV_MassStorageClass&REV_1539
│                         304d0264-1b0b-55ec-bc62-c79aca351420 ← SCSI\VEN_Generic&DEV_MassStorageClass
│   
├─Ryzen 5 PRO 4650U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
│                         1f2e0c2f-e078-5fcb-a8de-f78eff9a470a ← CPUID\PRO_0&FAM_17&MOD_60
│                         5fd5eb41-dd4a-5742-8448-861eaaa61cf8 ← CPUID\PRO_0&FAM_17&MOD_60&STP_1
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
[... 99 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─UMIS RPJTJ256MEE1OWX:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    2.1C0628
      Vendor:             Union Memory (Shenzhen) (NVME:0x1CC4)
      Serial Number:      SS1B60641Z1CD1A8300D
      Update Error:       Cannot install update when system battery is not at least 25%
      GUIDs:              0317581b-58d7-52b1-8ad7-6983aa66d3db ← NVME\VEN_1CC4&DEV_6202&SUBSYS_1CC41CC4&REV_03
                          56c329eb-587d-5325-a46b-ce9be3cc1d37 ← NVME\VEN_1CC4&DEV_6202&SUBSYS_1CC41CC4
                          66bde5bc-c9c2-562f-8ac0-45e1019e2578 ← NVME\VEN_1CC4&DEV_6202&REV_03
[... cut ...]
                          83f5c44e-578a-5a7a-8755-6e99234be8ba ← UMIS RPJTJ256MEE1OWX
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

In addition, let me just confirm: Is the cap file for X13 Gen1 AMD? The xml says it is for T14s. I remember that T14s Gen1 and X13 Gen1 are similar product, but does it also work on my laptop?