fwupd/firmware-lenovo #456: Getting "UEFI BIOS settings update pending reboot" after succesful update.

verpejas, 2024-07-08:
**Describe the bug**
Hi! I am getting the following error in fwupdmgr "UEFI BIOS settings update pending reboot" after successfully updating the bios from v1.25 to v1.26 on a ThinkPad T14 G2 AMD (model 20XK). At first,  fwupd did not want to work properly. I updated from a USB stick. Bios update succeeded but the above mentioned message still remained. I run OpenSUSE. I disabled secure boot and secure rollback - downgraded bios to 1.25 from USB - fwupd shows no errors. Updated bios from fwupd to 1.26 - it succeeded without issues, but the above mentioned message still remains in `fwupdmgr get-devices`

**Steps to Reproduce**
1. Check firmware info using `fwupdmgr get-devices`  - no errors shown on bios 1.25
2. On a Lenovo 20XK update bios from 1.25 to 1.26 either through USB, fwupd or from Windows.
3.  Check firmware info usin `fwupdmgr get-devices` - an error is shown: "Update Error:       UEFI BIOS settings update pending reboot"

**Expected behavior**
Just as with version 1.25, it should not show any errors as the bios has been updated succesfully and there does not apper to be any issues. Fwupd won't allow for future updates or downgrades due to this error being shown.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
verpejas@ThinkPad:~$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.20
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.20
runtime   org.kernel                    6.9.1-1-default

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed from OpenSUSE main repo:

```shell
verpejas@ThinkPad:~$ zypper info fwupd
Loading repository data...
Reading installed packages...

Information for package fwupd:
------------------------------
Repository     : Main Repository (OSS)
Name           : fwupd
Version        : 1.9.20-1.1
Arch           : x86_64
Vendor         : openSUSE
Installed Size : 6.4 MiB
Installed      : Yes (automatically)
Status         : up-to-date
Source package : fwupd-1.9.20-1.1.src
Upstream URL   : https://fwupd.org/
Summary        : Device firmware updater daemon

```

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20XK0016US
│
├─AMD Ryzen 5 PRO 5650U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a50000d
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│ │                       20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│ │   Device Flags:       • Internal device
│ │ 
│ └─Secure Processor:
[... 923 lines cut ...]
│       ├── xzio.mod
│       └── zstd.mod
├── initrd -> initrd-6.9.1-1-default
├── initrd-6.8.9-1-default
├── initrd-6.9.1-1-default
├── sysctl.conf-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/sysctl.conf
├── sysctl.conf-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/sysctl.conf
├── System.map-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/System.map
├── System.map-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/System.map
├── vmlinuz -> vmlinuz-6.9.1-1-default
├── vmlinuz-6.8.9-1-default -> ../usr/lib/modules/6.8.9-1-default/vmlinuz
└── vmlinuz-6.9.1-1-default -> ../usr/lib/modules/6.9.1-1-default/vmlinuz

57 directories, 499 files

```

**Additional questions**

- Operating system and version: Opensuse Tumbleweed
- Have you tried rebooting?: Yes
- Is this a regression?
- Are you using an NVMe disk?: Yes. WD SN570 2TB
- Is secure boot enabled?: It wouldn't update with SB enabled, disbled it for the duration of update
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?: No

verpejas, 2024-07-11:
Update on 2024.07.11 - the bug still occurs on Bios v 1.27 - I was unable to update using fwupdmgr or Discover - had to boot back up into a windows drive and perform the update there. The "UEFI BIOS settings update pending reboot" still persists on 1.27

fkru, 2024-07-13:
JFYI: Given ThinkPad T14 Gen 1 AMD and fwupd 1.9.21, I am unable to update BIOSS to version 1.49 with the error message "UEFI BIOS settings update pending reboot" (secure boot disabled).

fkru, 2024-07-14:
> JFYI: Given ThinkPad T14 Gen 1 AMD and fwupd 1.9.21, I am unable to update BIOS to version 1.49 with the error message "UEFI BIOS settings update pending reboot" (secure boot disabled).

Let me add that installing the update manually by downloading the .cab file works fine, but the error message "UEFI BIOS settings update pending reboot" still remains. By the way, BIOS version 1.49 fixes CVE-2023-20593 (Zenbleed) at last!

immanuelfodor, 2024-08-12:
I also can't apply a BIOS update on "ThinkPad T14/P14s Gen 1 AMD". It says "UEFI BIOS settings update pending reboot" for 0.1.47, so it doesn't let me install 0.1.49.

freefd, 2024-08-12:
Hi @immanuelfodor, I was able to install 0.1.49 using [Gnome Software GUI](https://apps.gnome.org/Software/)  as a workaround.
After the update, there is no longer anything about "update pending reboot" in the CLI output of fwupdmgr.

immanuelfodor, 2024-08-12:
I have Manjaro KDE, so no Gnome, and in Manjaro package manager (pamac-manager), there is no lvfs integration, my only option was fwupdmgr so far  :face_holding_back_tears:

immanuelfodor, 2024-09-03:
As a workaround, I managed to update the BIOS by creating a bootable USB stick from the ISO file provided by Lenovo.

wj25czxj47bu6q, 2024-10-31:
I also have this issue on the T14 AMD Gen 1 (20UD) since upgrading to 0.1.49. I was able to successfully update to 0.1.50 by downloading the .cab file from LVFS and using `fwupdmgr local-install`.

The "UEFI BIOS settings update pending reboot" error persists after updating to 0.1.50.

awehrfritz, 2024-11-04:
This still seems to be an issue in version 0.1.28 on my ThinkPad P14s Gen 2a. `fwupdmgr update` just found an update to version 0.1.29, but issues an `Update Error:       UEFI BIOS settings update pending reboot`.

I had the same issue with the 0.1.27 -> 0.1.28 update, but worked around it by creating a USB drive and manually updated it. Obviously, not a very elegant solution.

I'm on Fedora 40 with fwupd version 1.9.26.

awehrfritz, 2024-11-04:
> If you manually read that file I expect it's "1".

@superm1 I just checked, and the file has a "0":
``` sh
root@linux-P14s:~# cat /sys/devices/virtual/firmware-attributes/thinklmi/attributes/pending_reboot
0
```

Any other ideas how to debug/fix this?

superm1, 2024-11-04:
That's quite odd.  I wonder if we have a bug in our BIOS settings handler in fwupd.  You can see that the thinklmi plugin just gets it from context:
https://github.com/fwupd/fwupd/blob/1_9_X/libfwupdplugin/fu-context.c#L347

Context gets it from the bios settings handler: https://github.com/fwupd/fwupd/blob/1_9_X/libfwupdplugin/fu-bios-settings.c#L572

Can you tar up all of your /sys/class/firmware-attributes directory?  I can try to emulate your directory and see if it reproduces.

superm1, 2024-11-04:
@awehrfritz shared it to me offline, but I can't seem to get any UpdateError errors on my uefi devices tripped with this.  I tried it on my workstation.

Here is 2.0
```sh
❯ FWUPD_SYSFSFWATTRIBDIR=~/thinklmi/sys/devices/virtual/firmware-attributes/ fwupdtool get-devices 94d6edee-8f19-40a9-be14-d881e2e3d55c
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Selected device: System Firmware
LENOVO 30E1S68903
│
└─System Firmware:
      Device ID:          557960baa042cab815d0c30d318fa147f32cdd55
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    65627
      Minimum Version:    65627
      Vendor:             LENOVO (DMI:LENOVO)
      Update State:       Success
      GUID:               94d6edee-8f19-40a9-be14-d881e2e3d55c
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
      Device Requests:    • Message
```

Here is 1.9.x.
```
❯ FWUPD_SYSFSFWATTRIBDIR=~/thinklmi/sys/devices/virtual/firmware-attributes/ fwupdtool get-devices 94d6edee-8f19-40a9-be14-d881e2e3d55c
Loading…                 [                                       ]19:43:40.088 FuHistory            schema version 14 is unknown
Loading…                 [************************************** ]                                                                                                                                                                                                                      WARNING: This package has not been validated, it may not work properly.
Selected device: System Firmware
LENOVO 30E1S68903
│
└─System Firmware:
      Device ID:          557960baa042cab815d0c30d318fa147f32cdd55
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    65627
      Minimum Version:    65627
      Vendor:             LENOVO (DMI:LENOVO)
      Update State:       Success
      GUID:               94d6edee-8f19-40a9-be14-d881e2e3d55c
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
      Device Requests:    • Message
```

It's reading the pending_reboot properly as well.
```
❯ FWUPD_SYSFSFWATTRIBDIR=~/thinklmi/sys/devices/virtual/firmware-attributes/ fwupdtool get-bios-settings pending_reboot
Loading…                 [                                       ]19:43:58.015 FuHistory            schema version 14 is unknown
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
pending_reboot:
  Current Value:        0
  Description:          Settings will apply after system reboots
  Read Only:            True
```

I also tried to stage an update, but don't see anything come up.
```
❯ FWUPD_SYSFSFWATTRIBDIR=~/thinklmi/sys/devices/virtual/firmware-attributes/ fwupdtool update
Loading…                 [                                       ]19:44:25.918 FuHistory            schema version 14 is unknown
Loading…                 [************************************** ]                                                                                                                                                                                                                      WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates:
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVL21T0HCLR-00BL7
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.91 to 1.98?                                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║                                                                              ║
║ LENOVO 30E1S68903 must remain plugged into a power source for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Do not turn off your computer or remove the AC adapter while the update is in progress.
An update requires a reboot to complete. Restart now? [y|N]: n
Do not turn off your computer or remove the AC adapter while the update is in progress.
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]: n
```

awehrfritz, 2024-11-04:
Might be that I'm doing something wrong, but checking for `pending_change` give me this:

``` sh
root@linux-P14s:~# fwupdmgr get-bios-settings pending_change
Authenticating…          [ -                                     ]
Unable to find attribute

Use fwupdmgr --help for help
```

While I still get:
``` sh
root@linux-P14s:~# cat /sys/devices/virtual/firmware-attributes/thinklmi/attributes/pending_reboot 
0
```

This is on Fedora 40 with kernel 6.10.12-200.fc40.x86_64 and fwupd 1.9.26.

superm1, 2024-11-04:
> Thanks a lot @superm1 for looking into this! Much appreciated!
> 
> Might be that I'm doing something wrong, but checking for `pending_change` give me this:
> 
> ``` sh
> root@linux-P14s:~# fwupdmgr get-bios-settings pending_change
> Authenticating…          [ -                                     ]
> Unable to find attribute
> 
> Use fwupdmgr --help for help
> ```
> 
> While I still get:
> ``` sh
> root@linux-P14s:~# cat /sys/devices/virtual/firmware-attributes/thinklmi/attributes/pending_reboot 
> 0
> ```
> 
> This is on Fedora 40 with kernel 6.10.12-200.fc40.x86_64 and fwupd 1.9.26.
> 

It's "pending_reboot"

awehrfritz, 2024-11-05:
> It's "pending_reboot"

Arrrg, that was just me playing around with this. I actually tried `pending_reboot` first, but also that is not available:
``` bash
root@linux-P14s:~# fwupdmgr get-bios-settings pending_reboot
Authenticating…          [ -                                     ]
Unable to find attribute
```

superm1, 2024-11-05:
> > It's "pending_reboot"
> 
> Arrrg, that was just me playing around with this. I actually tried `pending_reboot` first, but also that is not available:
> 
> ```shell
> root@linux-P14s:~# fwupdmgr get-bios-settings pending_reboot
> Authenticating…          [ -                                     ]
> Unable to find attribute
> ```

Hmm, that's really odd!  I wonder if this is a race condition of thinklmi loading after fwupd is done because that was totally in the drop you gave me.

awehrfritz, 2024-11-05:
Same thing:
``` sh
root@linux-P14s:~# systemctl restart fwupd.service
root@linux-P14s:~# fwupdmgr get-bios-settings pending_reboot
Authenticating…          [ -                                     ]
Unable to find attribute
```

superm1, 2024-11-05:
Can you please attach the output of `sudo fwupdtool get-bios-settings -vv 2>&1 | tee fwupdtool.txt`?  I'd like to see if anything else complains that stands out why this wouldn't be working.

julianbrost, 2024-11-05:
Looks like I've found a workaround: after `echo 1 > /sys/devices/virtual/firmware-attributes/thinklmi/attributes/TFTPWindowSize/current_value` (which itself actually resulted in an error, `echo: write error: invalid argument`), reading that file works and `fwupdmgr get-devices` no longer shows that "Update Error: UEFI BIOS settings update pending reboot" error.

superm1, 2024-11-06:
Great sleuth work!
Have a try with this: https://github.com/fwupd/fwupd/pull/8041

closed 2024-11-06 by superm1

vwbusguy, 2024-11-14:
> reading that file works and `fwupdmgr get-devices` no longer shows that "Update Error: UEFI BIOS settings update pending reboot" error.

This workaround worked for me *after* a reboot.  I still got the error with `fwupdmgr get-bios-settings pending_reboot` until I restarted the systemd service and then got a return value of 1.  The 1 sets it to proper pending_reboot.  On reboot, I saw a message about a config reload on the Lenovo logo before the OS was loaded and after reboot, that had changed to a 0 and the error cleared.

superm1, 2024-11-14:
> This workaround worked for me *after* a reboot.  I still got the error with `fwupdmgr get-bios-settings pending_reboot` until I restarted the systemd service and then got a return value of 1.  The 1 [...]

So do you mean to say after installing the workaround and doing a fresh boot everything is fine? Or is there still a strateful bug?

vwbusguy, 2024-11-15:
> So do you mean to say after installing the workaround

It looks like the patch ignores the invalid state where that workaround actually fixes the invalid state:

> Looks like I've found a workaround: after `echo 1 > /sys/devices/virtual/firmware-attributes/thinklmi/attributes/TFTPWindowSize/current_value`

I haven't tested with the patch fix and unfortunately now can't because the state itself is now fixed on my P14S with that workaround.

```
$ rpm -q fwupd
fwupd-1.9.26-1.fc41.x86_64
```

awehrfritz, 2024-12-14:
Just as a follow-up on this, with the release of 1.9.27, I was able to test this and can confirm that fwupd/fwupd#8041 fixes the issue.

I just upgraded my firmware from 0.1.28 to 0.1.29 on a ThinkPad P14s Gen 2a.