fwupd/firmware-lenovo #458: Lenovo X1 Nano Gen 2 - Failed to recover ME Firmware

mtnstar, 2024-07-22:
Hi

just updated some firmware on my X1 Nano Gen 2, but now when I start my notebook, ME Firmware cannot be updated. Now sound is not working anymore, only dummy device available.

1. Switch on Notebook
2. Reading ME Firmware ... Please do not power off! (2-3mins)
3. Failed to recover ME Firmware
4. System Reset
5. Then Step 2+3 again
6. Then the system boots

tried downgrade:

`fwupdmgr --allow-older --allow-reinstall install 9b6f56c7d761d1b8cd2ae7ea441335834e4a1d9c4f0d74de36aba0bbabc1dfa5-Lenovo-ThinkPad-X1NanoGen2-CorporateMEFirmware-0.27.2176.cab`

but no effect.

[get-devices.txt](https://github.com/user-attachments/files/16339550/get-devices.txt)

Pop!_OS 22.04 LTS

```
~/tmp/x1nano $ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.5
runtime   org.kernel                    6.9.3-76060903-generic
```

thx for any help!

mtnstar, 2024-08-01:
rehi

I also tried to update bios with USB iso, but not effect. Still the same message when I boot up my notebook.

any help would be appreciated !

kmauleon, 2024-08-01:
hi @mtnstar can you please share your BIOS Setup Menu... thanks!
I will forward your concern to ODM CO....

mtnstar, 2024-08-02:
@kmauleon thx

[image]

kmauleon, 2024-08-05:
Hi @mtnstar based on the SS, your system has the latest MEFW already... no need to update MEFW

kmauleon, 2024-08-05:
can you please try the following...
1. BIOS default settings
2. Linux Firmware Update boot on lowest priority
3. Reinstall BIOS via USB >> [link](https://pcsupport.lenovo.com/ph/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-nano-gen-2-type-21e8-21e9/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C)

mtnstar, 2024-08-12:
Hi @kmauleon 

tried all those 3 steps, but still the same :cry: 

Created a bootable usb stick with bios upgrade, then applied the update, but after the restart, same ME firmware error ...

anything else I could try?

kmauleon, 2024-08-12:
I am still waiting for advise from ODM CO....
Can you please check this link if anything could work >> https://github.com/fwupd/firmware-lenovo/issues/303#issuecomment-1373192951

kmauleon, 2024-08-21:
hi @mtnstar can you please update your system BIOS to [1.27](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3IET.firmware)

mtnstar, 2024-08-26:
@kmauleon just tried to update to 1.27, but it looks like the update is blocked by the pending ME Firmware upgrade :cry: 
tried to update system BIOS with `fwupdmgr update` but also with USB stick. both attempts did not update BIOS from 1.21 to 1.27.

[image]

anything else I could try to get rid of the ME Firmware update process?

kmauleon, 2024-08-27:
hi @mtnstar did you choose number 2 from the option?
I will ask ODM CO for advise and feedback... thank you

mtnstar, 2024-09-02:
hi @kmauleon thx for your support!

1. `sudo fwupdmgr update`
2. reboot
3. [image]
4. [image]
5. small text `Reset System` on the upper right
6. step 3.+4. again

boot

bios not updated!

kmauleon, 2024-09-09:
hi @mtnstar are you willing to update MEFW via USB?  ODM CO can share the procedure for this.  please email me
kmauleon at lenovo dot com

mtnstar, 2024-09-09:
thanks a lot @kmauleon for sending me the correct files for flashing and fixing the pending ME firmware upgrade!
it works now again :partying_face: :pray: :+1:

closed 2024-09-09 by mtnstar

kmauleon, 2024-09-10:
Glad it did work! thanks for the feedback @mtnstar !!

PauloMorgs, 2025-11-23:
same problem here on t14 gen 2

kmauleon, 2025-11-24:
@PauloMorgs let's discuss in #556 for proper tracking... thank you