fwupd/firmware-lenovo #459: Need reinstallation every time I reboot

ThomasBerneHCSE, 2024-07-24:
Everytime I reboot my computed, the lenovo system firmware  needs the same update again and again:

[image]

I'm already several versions behind. How can I fix this?

[image]

hughsie, 2024-07-24:
Can you fill in this please:

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

ThomasBerneHCSE, 2024-07-24:
```
❯ fwupdmgr --version

compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-38-generic
runtime   org.freedesktop.fwupd         1.9.16
```
```
❯ efibootmgr -v

BootCurrent: 0009
Timeout: 1 seconds
BootOrder: 0009,0008,0005,0004,000B,0001,0006,0002,0000,0003
Boot0000* Windows Boot Manager  VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d0000000c000100000010000000040000007fff0400
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
    data: 57 49 4e 44 4f 57 53 00 01 00 00 00 88 00 00 00 78 00 00 00 42 00 43 00 44 00 4f 00 42 00 4a 00 45 00 43 00 54 00 3d 00 7b 00 39 00 64 00 65 00 61 00 38 00 36 00 32 00 63 00 2d 00 35 00 63 00 64 00 64 00 2d 00 34 00 65 00 37 00 30 00 2d 00 61 00 63 00 63 00 31 00 2d 00 66 00 33 00 32 00 62 00 33 00 34 00 34 00 64 00 34 00 37 00 39 00 35 00 7d 00 00 00 0c 00 01 00 00 00 10 00 00 00 04 00 00 00 7f ff 04 00
Boot0001* endeavouros   VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot0002* Generic Usb Device    VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot0003* CD/DVD Device VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot0004* UEFI: PXE IPv4 Intel(R) Ethernet Connection (17) I219-LM      PciRoot(0x0)/Pci(0x1f,0x6)/MAC(9c2dcd110d4f,0)/IPv4(0.0.0.00.0.0.0,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 9c 2d cd 11 0d 4f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot0005* UEFI: PXE IPv6 Intel(R) Ethernet Connection (17) I219-LM      PciRoot(0x0)/Pci(0x1f,0x6)/MAC(9c2dcd110d4f,0)/IPv6([::]:<->[::]:,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 9c 2d cd 11 0d 4f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot0006* Linux Boot Manager    VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot0008* Linux Boot Manager    HD(1,GPT,5aa1265f-8f76-45bc-a256-35f9df68af2c,0x1000,0x1f4000)/File(\EFI\SYSTEMD\SYSTEMD-BOOTX64.EFI)
      dp: 04 01 2a 00 01 00 00 00 00 10 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 5f 26 a1 5a 76 8f bc 45 a2 56 35 f9 df 68 af 2c 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 53 00 59 00 53 00 54 00 45 00 4d 00 44 00 5c 00 53 00 59 00 53 00 54 00 45 00 4d 00 44 00 2d 00 42 00 4f 00 4f 00 54 00 58 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
Boot0009* Ubuntu        HD(1,GPT,5aa1265f-8f76-45bc-a256-35f9df68af2c,0x1000,0x1f4000)/File(\EFI\UBUNTU\SHIMX64.EFI)
      dp: 04 01 2a 00 01 00 00 00 00 10 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 5f 26 a1 5a 76 8f bc 45 a2 56 35 f9 df 68 af 2c 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 55 00 42 00 55 00 4e 00 54 00 55 00 5c 00 53 00 48 00 49 00 4d 00 58 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
Boot000B* UEFI OS       HD(1,GPT,5aa1265f-8f76-45bc-a256-35f9df68af2c,0x1000,0x1f4000)/File(\EFI\BOOT\BOOTX64.EFI)0000424f
      dp: 04 01 2a 00 01 00 00 00 00 10 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 5f 26 a1 5a 76 8f bc 45 a2 56 35 f9 df 68 af 2c 02 02 / 04 04 30 00 5c 00 45 00 46 00 49 00 5c 00 42 00 4f 00 4f 00 54 00 5c 00 42 00 4f 00 4f 00 54 00 58 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
```

No output for `efivar -l | grep fw`.

```
❯ tree /boot

/boot
├── config-6.8.0-36-generic
├── config-6.8.0-38-generic
├── efi
│   ├── e7352681962248f598b4de79e28eec32
│   │   ├── 6.6.32-1-lts
│   │   │   ├── initrd
│   │   │   ├── initrd-fallback
│   │   │   └── linux
│   │   └── 6.9.2-arch1-1
│   │       ├── initrd
│   │       ├── initrd-fallback
│   │       └── linux
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fbx64.efi
│   │   │   └── mmx64.efi
│   │   ├── Linux
│   │   ├── systemd
[... 302 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-6.8.0-38-generic
├── initrd.img-6.8.0-36-generic
├── initrd.img-6.8.0-38-generic
├── initrd.img.old -> initrd.img-6.8.0-36-generic
├── lost+found  [error opening dir]
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.8.0-36-generic
├── System.map-6.8.0-38-generic
├── vmlinuz -> vmlinuz-6.8.0-38-generic
├── vmlinuz-6.8.0-36-generic
├── vmlinuz-6.8.0-38-generic
└── vmlinuz.old -> vmlinuz-6.8.0-36-generic
```
Operating system and version is defined in the first message : Kubuntu 24.04.
The problem occurs when rebooting: I upgrade the lenovo system firmware, reboot and I have to do it again like if the update didn't hapen. 
Secure boot isn't enabled in the BIOS
Boot Order Lock isn't enabled in the BIOS

ThomasBerneHCSE, 2024-07-24:
```
❯ fwupdmgr get-devices --show-all-devices
LENOVO 30FA000PFR
│
├─Unknown Device:
│     Device ID:          f11ae0c22cb95f9e05ab0725c266bf3d029141d1
│     Serial Number:      1
│     GUID:               690573e8-228a-5d48-ab03-9678dcc9ee4e ← DRM\VEN_LHC&DEV_2700
│   
├─Unknown Device:
│     Device ID:          10b89bd2d0a93eea2c45428e980d3bf8ee4ad42b
│     Serial Number:      1
│     GUID:               690573e8-228a-5d48-ab03-9678dcc9ee4e ← DRM\VEN_LHC&DEV_2700
│   
├─Unknown Device:
│     Device ID:          0524415e314e4a07b83ff6e54c04d9f08ea2c84f
│     Serial Number:      1
│     GUID:               690573e8-228a-5d48-ab03-9678dcc9ee4e ← DRM\VEN_LHC&DEV_2700
│   
├─12th Gen Intel Core™ i7-12700T:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000035
│     Vendor:             Intel
│     GUIDs:              615995d1-bdd0-5048-a291-83eaef0de820 ← CPUID\PRO_0&FAM_06&MOD_97
│                         c9697a05-dbd8-58ac-833c-5a4fa8fdbcd7 ← CPUID\PRO_0&FAM_06&MOD_97&STP_2
│     Device Flags:       • Internal device
│   
├─Micron MTFDKBA512TFH:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    7006P7LN
│     Vendor:             Micron Technology Inc (NVME:0x1344)
│     Serial Number:      222037F2B876
│     GUIDs:              b70b3b98-b8a6-5e07-8a97-138d66b0dded ← NVME\VEN_1344&DEV_5407
│                         07e304bc-bb66-563d-bb53-011f94039748 ← NVME\VEN_1344&DEV_5407&SUBSYS_13440100
│                         71266491-97a4-5af1-8bfd-257b33a970d5 ← Micron MTFDKBA512TFH
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          52fd7e8d4539254282359c12d6f2ce90ba653eba
│ │   Summary:            UEFI System Resource Table device (Updated via caspule-on-disk)
│ │   Current version:    0.1.29
│ │   Minimum Version:    0.1.29
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               15fae072-6129-4306-bd08-22194b7855ba
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ ├─AMT [unprovisioned]:
│ │     Device ID:        8d5470e73fd9a31eaa460b2b6aea95483fe3f14c
│ │     Summary:          Hardware and firmware technology for remote out-of-band management
│ │     Current version:  16.0.15.1662
│ │     Bootloader Version:16.0.15.1662
│ │     Vendor:           Intel Corporation (MEI:0x8086)
│ │     GUIDs:            12f80028-b4b7-4b2d-aca8-46e0ff65814c
│ │                       3244e2f4-fbe1-5e67-8c66-1c9936061c21 ← MEI\VEN_8086&DEV_7AE8
│ │                       7c6adfbd-f851-5647-82df-f6f0052c3d2c ← MEI\VEN_8086&DEV_7AE8&SUBSYS_17AA330E
│ │     Device Flags:     • Internal device
[... 82 lines cut ...]
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          5209ea5ea8047a13e95ba630a1db2172b51ba2eb
│     Summary:            UEFI System Resource Table device (Updated via caspule-on-disk)
│     Current version:    356527616
│     Minimum Version:    356527616
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               0c208ff1-6a42-45e1-887a-7694815a9981
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            LENOVO
      GUID:               3110ca52-44a7-5bdd-ba77-8a8de8eb6a2b ← UEFI\CRT_64ADF813F63132B2F8E8499FBD00D90BCF762DC3
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • Micron MTFDKBA512TFH (7005P7LN → 7006P7LN)```

ThomasBerneHCSE, 2024-07-26:
@hughsie There should be it. Tell me if you need anymore informations

hughsie, 2024-07-26:
I think having multiple Linuxes [partially] installed is causing confusion -- you've also got Ubuntu stuff there too. I'm not sure where to start tbh.

jeremyrobert-hellocse, 2024-07-26:
Hello, i have the same problem, but i haven't multiple linuxes installed, only Ubuntu 24.04.

```
❯ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.16
compile   com.hughsie.libxmlb           0.3.17
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-39-generic
runtime   org.freedesktop.fwupd         1.9.16
```

```
❯ efibootmgr -v
BootCurrent: 0004
Timeout: 1 seconds
BootOrder: 0004,0002,0001,0003,0000
Boot0000* Windows Boot Manager	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000000000100000010000000040000007fff0400
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
    data: 57 49 4e 44 4f 57 53 00 01 00 00 00 88 00 00 00 78 00 00 00 42 00 43 00 44 00 4f 00 42 00 4a 00 45 00 43 00 54 00 3d 00 7b 00 39 00 64 00 65 00 61 00 38 00 36 00 32 00 63 00 2d 00 35 00 63 00 64 00 64 00 2d 00 34 00 65 00 37 00 30 00 2d 00 61 00 63 00 63 00 31 00 2d 00 66 00 33 00 32 00 62 00 33 00 34 00 34 00 64 00 34 00 37 00 39 00 35 00 7d 00 00 00 00 00 01 00 00 00 10 00 00 00 04 00 00 00 7f ff 04 00
Boot0001* UEFI: PXE IPv4 Intel(R) Ethernet Connection (17) I219-LM	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(e88088a47f36,0)/IPv4(0.0.0.00.0.0.0,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 e8 80 88 a4 7f 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot0002* UEFI: PXE IPv6 Intel(R) Ethernet Connection (17) I219-LM	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(e88088a47f36,0)/IPv6([::]:<->[::]:,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 06 1f / 03 0b 25 00 e8 80 88 a4 7f 36 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot0003* Linux-Firmware-Updater	HD(1,GPT,90c7abe4-c3ef-4cf4-9952-a2f27e068bc0,0x800,0x219800)/File(\EFI\ubuntu\fwupdx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 98 21 00 00 00 00 00 e4 ab c7 90 ef c3 f4 4c 99 52 a2 f2 7e 06 8b c0 02 02 / 04 04 36 00 5c 00 45 00 46 00 49 00 5c 00 75 00 62 00 75 00 6e 00 74 00 75 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0004* Ubuntu	HD(1,GPT,90c7abe4-c3ef-4cf4-9952-a2f27e068bc0,0x800,0x219800)/File(\EFI\ubuntu\shimx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 98 21 00 00 00 00 00 e4 ab c7 90 ef c3 f4 4c 99 52 a2 f2 7e 06 8b c0 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 75 00 62 00 75 00 6e 00 74 00 75 00 5c 00 73 00 68 00 69 00 6d 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
```
```
❯ fwupdmgr get-devices --show-all-devices
LENOVO 30H0002HFR
│
├─Unknown Device:
│     Device ID:          f52c67be84319b3a7a97466582ec85ca88892153
│     Serial Number:      VNA8592Z
│     GUID:               0be1fbc0-2c3e-5de2-b8d2-21ea86b33c6a ← DRM\VEN_LEN&DEV_62B0
│   
├─Unknown Device:
│     Device ID:          78b0c9d3cbc4d10ee1d91ac02f140a5f96a9f263
│     Serial Number:      VNA8584N
│     GUID:               0be1fbc0-2c3e-5de2-b8d2-21ea86b33c6a ← DRM\VEN_LEN&DEV_62B0
│   
├─13th Gen Intel Core™ i7-13700T:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000123
│     Vendor:             Intel
│     GUIDs:              3d4645d8-13fe-541a-ac5c-7d9d944e4500 ← CPUID\PRO_0&FAM_06&MOD_B7
│                         988c4cf1-ace4-5eb5-b26c-905ff236f412 ← CPUID\PRO_0&FAM_06&MOD_B7&STP_1
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          32216088075afb6a4d591c6c612ab8d2b064aec8
│ │   Summary:            UEFI System Resource Table device (Updated via caspule-on-disk)
│ │   Current version:    0.1.42
│ │   Minimum Version:    0.1.42
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               8a15883b-8f20-4aaf-a7d8-ceb35316fb3d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
[... 447 lines cut ...]
│       └── zstd.mod
├── initrd.img -> initrd.img-6.8.0-39-generic
├── initrd.img-6.8.0-38-generic
├── initrd.img-6.8.0-39-generic
├── initrd.img.old -> initrd.img-6.8.0-38-generic
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.8.0-38-generic
├── System.map-6.8.0-39-generic
├── vmlinuz -> vmlinuz-6.8.0-39-generic
├── vmlinuz-6.8.0-38-generic
├── vmlinuz-6.8.0-39-generic
└── vmlinuz.old -> vmlinuz-6.8.0-38-generic

11 directories, 311 files
```

**Additional questions**

- Operating system and version: **Ubuntu 24.04**
- Have you tried rebooting? **Yes**
- Is secure boot enabled? **No**
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? **No**