fwupd/firmware-lenovo #460: ThinkStation P330 BIOS update did not update CSME

kepstin, 2024-08-05:
I recently set up a ThinkStation P330 (Gen2; i7-9700) which had been sitting idle/unused for quite a long time. I installed Fedora 40, and gave me an update to BIOS version 0.1.117, which I applied. I don't recall the exact original BIOS version; I think it may have been 0.1.57.

However, even though the BIOS update was applied, the ME was not updated. Both fwupd and the BIOS "System Status" screen report ME version 12.0.6.1120, which appears to be very old. (The release notes for BIOS version 0.1.76 say "Update ME FW to 12.0.49.1534", so I would have expected either that version or something newer.) I do not recall seeing any screens during the update process that mentioned applying an ME update.

I then attempted to apply BIOS M1VKT77A (0.1.119) via the Linux capsule update package available from the Lenovo support website via fwupd. The BIOS appears to have updated correctly, but the ME version remained unchanged.

As a final attempt, I re-applied BIOS M1VKT77A (0.1.119) via the USB Drive Package. Although the BIOS flash screen appeared and it looked like the re-flash was successful, again the ME version remained unchanged.

Is there something I'm missing? Is there any other option available to apply the ME updated?

<details>
<summary>fwupdmgr get-devices --show-all</summary>

```
LENOVO 30D10016US
│
├─Unknown Device:
│     Device ID:          5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│     Serial Number:      1492V6130HCL
│     GUID:               c7d085af-e82f-56d0-b0cd-d60072b075e8 ← DRM\VEN_DEL&DEV_A0BE
│   
├─Unknown Device:
│     Device ID:          a6c25fea64794a797f39d32a3277659be669963d
│     Serial Number:      G3D7F041161B
│     GUID:               c7d085af-e82f-56d0-b0cd-d60072b075e8 ← DRM\VEN_DEL&DEV_A0BE
│   
├─BIOS1:
│     Device ID:          a41715edae830e26b4c4a9246ffa62cddf8f7338
│     Summary:            Memory Technology Device
│     Vendor:             DMI:LENOVO
│     GUIDs:              cc083504-3412-5e26-ad02-66c3fb6335ac ← MTD\NAME_BIOS1
│                         18cd26a0-3a62-5fd7-a9f3-e92d1af1f858 ← MTD\VENDOR_LENOVO&NAME_BIOS1
│                         508c7811-f3c2-533e-beba-ed3d07112141 ← MTD\VENDOR_LENOVO&PRODUCT_30D10016US&NAME_BIOS1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
├─CoffeeLake-S GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              8655a49b-b4b2-54ff-a6c6-06009dd64f13 ← PCI\VEN_8086&DEV_3E98
│                         d32cfd2a-eb63-5f46-85d4-23c756a119c2 ← PCI\VEN_8086&DEV_3E98&SUBSYS_17AA314F
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-9700 CPU @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000fa
│     Vendor:             Intel
│     GUIDs:              809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         50a811ae-a8fd-5cd0-90f4-33583974b789 ← CPUID\PRO_0&FAM_06&MOD_9E&STP_D
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          175ab462ce746650ec3f14c6d0c95315cb53a72c
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    0.1.119
│ │   Minimum Version:    0.1.119
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               6bfea39f-87d3-4be0-9afb-39c8632dea45
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
[... 68 lines cut ...]
│     Current version:    RQR24.11_B0036
│     Bootloader Version: BOT03.02_B0009
│     Vendor:             Logitech, Inc. (HIDRAW:0x046D, USB:0x046D)
│     Install Duration:   30 seconds
│     GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                         279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
└─WD BLACK SN770 1TB:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    731100WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      24143A807914
      GUIDs:              1524d43d-ed91-5130-8cb6-8b8478508bae ← NVME\VEN_15B7&DEV_5017
                          87cfda90-ce08-52c3-9bb5-0e0718b7e57e ← NVME\VEN_15B7&DEV_5017&SUBSYS_15B75017
                          914bfa00-b683-532c-8c3c-71a59e7ae800 ← WD_BLACK SN770 1TB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```
</details>

hughsie, 2024-08-05:
BIOS and CMSE (ME) updates are separate on Lenovo platforms I'm afraid.

kepstin, 2024-08-06:
Hmm. Ok, so a little more research and I think I've figured out what's going on.

First, the BIOS release notes for [0.1.76 on LVFS](https://fwupd.org/lvfs/devices/com.lenovo.P330.BIOS.firmware#0.1.76) aka [M1VKT4CA from Lenovo](https://download.lenovo.com/pccbbs/thinkcentre_bios/m1vjt77usa.txt) explicitly state "Update ME FW to 12.0.49.1534". Either this is false, or maybe the Windows exe download (which I can't find) for this version included a copy of the ME update tool in the zip?

Second, the support website for the [P330 Workstation 2nd Gen](https://pcsupport.lenovo.com/us/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p330-2nd-gen/downloads/driver-list/component?name=Motherboard%20Devices%20%28Backplanes,%20core%20chipset,%20onboard%20video,%20PCIe%20switches%29&id=FB762CAD-DE8B-49C2-A04B-809B4958F608), which is what I get when I enter my system's serial number, does *not* list any separate ME firmware update tools. But the support page for the [P330 Workstation](https://pcsupport.lenovo.com/us/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p330/downloads/driver-list/component?name=Motherboard%20Devices%20%28Backplanes,%20core%20chipset,%20onboard%20video,%20PCIe%20switches%29&id=FB762CAD-DE8B-49C2-A04B-809B4958F608) - which is identical hardware except using an 8th gen Intel cpu instead of 9th gen - *does* list the ME update tool. (Specifically, the "Corporate Intel Management Engine Firmware Update Tool" version 12.0.90.2072).

So I guess there's 2 separate problems?

* Lenovo's support website is not listing some downloads that are applicable to my hardware.
* Lenovo isn't providing ME updates for this hardware via LVFS for Linux.

hughsie, 2024-08-06:
@mrhpearson when you're back from PTO can you shed any light on this please. Thanks!

mrhpearson, 2024-08-07:
Created internal ticket LO-3244 for the FW team - looks like the BIOS update on LVFS hasn't been moved from embargo to stable; and I can't see any ME updates issued either.
Have asked for these to be provided.

mrhpearson, 2024-08-13:
BIOS M1VKT77A FW has been released to LVFS.
The FW team told me this includes the ME FW (v12.0.49.1534)

Let me know if any further issues

kepstin, 2024-08-13:
The updated BIOS does not resolve this issue - installing the BIOS updates does *not* update the ME firmware.

The latest currently available ME firmware from the Lenovo site for this device (available for download via a separate updater for Windows) is 12.0.90.2072.

mrhpearson, 2024-08-13:
I'll double check with the FW team. Thanks for letting me know.

mrhpearson, 2024-08-14:
Had a bit of back and forth with the FW team. Not great news I'm afraid

 - The note about ME FW update being included in the BIOS update was incorrect and misleading.

 - This platform apparently did not have LVFS updates for the ME FW in scope (it was added for the P340 and newer). They are not planning on doing an LVFS update, and pointed me at the Windows updater as the supported method for doing the update. Sorry - I know that sucks :(

Mark

kepstin, 2024-08-14:
Ah, that really is unfortunate, especially since it was a system that was available with a Linux preload :/

I've managed to do the update manually by extracting the firmware update binary from the windows updater tool and locating a copy of the FWUpdLcl tool for Linux.