fwupd/firmware-lenovo #461: Make firmware for Lenovo LOQ series laptop available on LVFS/fwupmgr

UnforgivenIsAlive, 2024-08-08:
Currently, there is no firmware available for the Lenovo LOQ series laptops (a budget variant of the Legion series, which are also not available.

Would be cool if they can be made available via fwupdmgr
Currently, new BIOS firmware is [available](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/gaming-series/loq-16aph8/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C) from June 2024 which now cannot be installed via fwupdmgr.

hughsie, 2024-08-08:
@mrhpearson i appreciate this model isn't in the linux program, but is there anything you can share? e.g. a Lenovo-approved way of getting the capsule file out of the win32 exe perhaps -- then we can write up a nice doc about using `sudo fwupdtool install-blob file.cap --plugin uefi_capsule`

mrhpearson, 2024-08-08:
My guess is that Lenovo isn't going to approve anything that hasn't been signed off on by QA. This has some validity - if you used a 'Lenovo sanctioned' method, but it's not tested, and then it goes wrong and your laptop gets bricked - Lenovo get's yelled at (and on the hook for the expense of repairs etc).
Trying to get this from a team that don't have a Linux program...I don't think I'd have a hope of any success I'm afraid. I'll check next time I meet with the FW leads for their opinion in case I'm wrong.
It would be really nice if we could have platforms that don't necessarily get the full Linux treatment (preloads/certification/regulatory paperwork/etc) but a middle ground where FW can be updated and best effort support provided for running Linux. 
I don't have that kind of framework to operate under currently I'm afraid.

UnforgivenIsAlive, 2024-08-08:
Please help me understand...The LVFS is full with Lenovo ThinkPad laptops. Are these all sanctioned by Lenovo because Lenovo has them in their Linux program ? And because the LOQ, IdeaPa and Legiond series are not,(Windows only) Lenovo won't sanction them to be in the LVFS  ?

mrhpearson, 2024-08-09:
Yes - that's it.
www.lenovo.com/linux might be helpful?

darius-m, 2025-08-07:
This may be a bit off-topic for fwupd (also in an older discussion), but is there any technical reason why the firmware updates are not released as binary files that can be used to update the firmware from the UEFI menus, similar to how desktop motherboard manufacturers release their updates?

I think in a different discussion on the repo, @hughsie also mentioned that most of the actual work (which I assume includes version checking, system compatibility, cryptographic signatures, etc.) is actually performed by the UEFI and the OS's job in the update process is to copy some binaries and then reboot the system to allow UEFI to do its job. This makes sense, as it's imperative from a security point of view to not trust anything coming from lower privilege levels, which in the case of UEFI includes the admin account on the OS.
If that is the case, it seems like the only piece missing is a page in the UEFI UI that allows copying the binary blob to the correct location to kick off the update process (as well as the binary files, of course). I'm not familiar with low level UEFI functionality, so things may actually be much more complicated than this.

I'm sure most Linux users would appreciate having fwupd being able to update the firmware automatically, but I think people would agree that being able to manually update the UEFI would be a reasonable compromise (and some might even prefer it).

c8dhjp4tyv-bit, 2026-10-09:
Hi @mrhpearson and @hughsie,

I own a Lenovo LOQ 15AHP10 (83JG) and developed an unofficial local fwupd remote for the original Lenovo R8CN26WW BIOS.

On September 13, 2026, I successfully performed a same-version BIOS reinstall on real hardware using fwupd.

The recorded results confirm:

- UEFI capsule update protocol
- `UpdateState: 2` (Success)
- `LastAttemptStatus: 0x0`
- Correct firmware GUID and AppStream ID
- Successful boot after the update

This was a same-version reinstall on one machine, not a full upgrade-path or manufacturer QA test.

I understand Lenovo's concerns regarding QA and support responsibility. My goal is to help bring official LVFS BIOS updates to the LOQ series, so Linux users can simply run `fwupdmgr update`.

Would Lenovo consider reviewing these results or involving the appropriate firmware engineering team?

I can provide sanitized logs, metadata and packaging tools without redistributing Lenovo's proprietary BIOS.

Thank you!

c8dhjp4tyv-bit, 2026-10-10:
https://github.com/c8dhjp4tyv-bit/lenovo-loq15ahp10-fwupd-tools

mrhpearson, 2026-10-13:
Afraid I don't have any hooks into the LOQ team - I'll see if I can find out more but chances of success are pretty small I'm afraid.

c8dhjp4tyv-bit, 2026-10-24:
Thanks @mrhpearson, I really appreciate you checking.

I understand you don't have a direct line into the LOQ team. Would you be able to point me to a name, role, or team I could contact directly — someone on the consumer/Legion BIOS side, or whoever handled Linux enablement for Legion Go S? I'm happy to do the legwork myself; an intro or just a name would be a huge help.

The ask itself is small: not full Linux certification, just permission to publish already-released, Lenovo-signed capsules on LVFS with a "no official support" label.

Thanks again!

awsms, 2026-11-07:
Not having to boot Winblows PEE every single time I need to update the BIOS of my Legion Slim 5 would be very nice.

Firi0n, 2026-11-15:
Data point: **upgrade** (not same-version) on a **LOQ 15AHP9 (83DX)**, done from Linux with fwupd, no Windows involved.

- BIOS NZCN28WW → **NZCN42WW** (EC 1.28 → 1.42), CachyOS, fwupd 2.1.9, 2026-11-15
- Payload: `JLG50042.ROM` extracted with `7z x nzcn42ww.exe` from the official Lenovo download (Lenovo-signed PE image, `CN=Certificate_NZ`, contains the full 32 MB SPI image incl. EC firmware; the bundled `platform.ini` has `[UpdateEC] Flag=0` and `[SecureUpdate] viaESP=3`, so the firmware itself does the flash)
- Command: `sudo fwupdtool install-blob JLG50042.ROM <System Firmware device ID>`, then reboot
- Result: flash completed, `bios_version` = NZCN42WW, `ec_firmware_release` = 1.42, fwupd Update State = Success
- Side effect: **BIOS settings were reset to factory defaults.**
- Nothing else regressed so far (lenovo-wmi-other attributes, platform profiles, NVIDIA RTD3/D3cold all fine).

Unofficial, single unit, use at your own risk, but it confirms the capsule path works on the LOQ 15AHP9 for real version upgrades.