fwupd/firmware-lenovo #462: Thinkpad Carbon X1 Gen 8 bricked after botched Fedora firmware update

Waldorf3, 2024-08-17:
**Describe the bug**
I had this X1 Gen 8 with Win 11 running as it should, but decided I wanted to change to Linux. So I installed Fedora Workstation 40, and the updater then prompted me to first install two firmware updates. One was something to do with battery management, I can't recall what the other was about. Anyway, I installed them, and there was this red line showing the update process on the screen, until it suddenly shut off.

Since then it has been dead. I tried to disconnect the power and press a paperclip in the reset hole, but no cigar. Have I bricked it? Any suggestions to get it going again?

**Steps to Reproduce**
It's dead. I can't reproduce it.

**Expected behavior**
It would be nice if it would boot. Even into BIOS.

**fwupd version information**
I don't know.  Standard Fedora image downloaded and installed, happened from there.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

The standard Fedora GUI tool. I think they call it "software", but I'm not sure.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
Oh yes, many times.  It doesn't. 

- Is this a regression?

hughsie, 2024-08-17:
@mrhpearson help!

hughsie, 2024-08-17:
There's some more debugging available in https://www.reddit.com/r/thinkpad/comments/1eucmz8/thinkpad_carbon_x1_gen_8_bricked_after_fedora/ I think.

Waldorf3, 2024-08-17:
Sorry, not directly a fwupd problem, I'll close this and continue back on Reddit if anyone is interested.

closed 2024-08-17 by Waldorf3