fwupd/firmware-lenovo #465: Updating docking station (4OBO) firmware in Linux, then switching to windows to perform additional updates gives error "MCU check mode timeout"

rustyscottweber, 2024-09-04:
Think-pad thunderbolt 4 dock (4OBO) . Firmware was updated initially using linux (ubuntu 22.04). However, after the firmware was updated to 10.13, the operating system was swapped out to Windows. Attempting to upgrade the firmware, in Windows, results in the following error: "MCU check mode response timeout". Additional upgrades and downgrades in Linux report success.

hughsie, 2024-09-04:
So it only fails in Windows? I think you should report a bug there I'm afraid.

rustyscottweber, 2024-09-04:
The problem is that they are going to point the fingers right back to fwupd since the windows problem only ever happens after an upgrade conducted by fwupd.  It's the whole volleyball problem where different people go bouncing the problem between two groups saying "It's not my Problem". Ideally, the bug should lie closest to where the solution gets implemented. If fwupd is causing a problem in the firmware which causes windows to fail, it seems logical that the change would have go into fwupd since the problem is not reciprocal; updates in windows do not cause problems for fwupd.  If there is a change that's needed on the windows side of things, then I would like to have an iron clad case or smoking gun of something that they are doing wrong or differently that they should be doing but aren't.

hughsie, 2024-09-04:
Right, but it's not like we can can grep for "MCU check mode response timeout" in the windows updater.

rustyscottweber, 2024-09-04:
I agree that the situation isn't ideal and I'm still trying to engage the Windows side of the firmware update to give me an exact nature of the problem, but I'm running into issues with finding someone who knows what they are talking about or who has the power to help. Is there a better or more direct windows firmware resource or point of contact that we can use to sus out which side of the fence this problem should land on?

hughsie, 2024-09-04:
If this is `USB\VID_17EF&PID_30B4` then @victor-cheng is the right person to help.

rustyscottweber, 2024-09-04:
```
├─ThinkPad Thunderbolt 4 Dock:
│ │   Device ID:          edd70214171cb2cf6eca28973f9b5dd26e746d1d
│ │   Current version:    10.16
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      1S40B0ZDTL3JMN
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       275e4695-9b4e-5263-835e-8681bec8cd1a ← USB\VID_17EF&PID_30B4&CID_40B0
│ │                       09f2cd6f-de67-5046-91ab-7de482e67877
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │   Device Requests:    • Message
│ │ 
│ ├─Audio Controller:
│ │     Device ID:        7405a651aaa9ab3b5f5665f20c185b2ac6e7dd3d
│ │     Current version:  49-0E-41
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             e1443fec-2b49-5bd0-b921-662ed11e88df ← USB\VID_17EF&PID_30B4&CID_AUDIO
│ │   
│ ├─Display Port 5:
│ │     Device ID:        920c38586a88745a1dfcb9026f9b2410eda9c98b
│ │     Current version:  5.07.001
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             755e347f-4ae7-53f8-ad69-45b3e9145ed8 ← USB\VID_17EF&PID_30B4&CID_DP5x
│ │   
│ ├─Display Port 6:
│ │     Device ID:        a0e0e61e3203362c603cb9e6cca46dad99cbbb1b
│ │     Current version:  6.05.003
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             a7b49eae-1c1d-5b83-ac90-9bbe78ee39c1 ← USB\VID_17EF&PID_30B4&CID_DP6x
│ │   
│ ├─Dock Management Controller:
│ │     Device ID:        f89ae2ce277603f7f389c133864e2dc4590adfe5
│ │     Current version:  0.1.97
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             3edd934f-9800-59c9-b02a-11be480d72e5 ← USB\VID_17EF&PID_30B4&CID_DMC
│ │   
│ ├─Dock Management Controller:
│ │     Device ID:        095dd0f7f9f2fad67fcf47e2236cb1b8097e3498
│ │     Current version:  21.15
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             c287973a-cc6b-5d2a-ac71-3956a512f73d ← USB\VID_17EF&PID_30B4&CID_MCU
│ │   
│ ├─Dock Management Controller Information:
│ │     Device ID:        6dfcab9946044070936c3aa15176f5cfa987a595
│ │     Current version:  10.16
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     Serial Number:    1S40B0ZDTL3JMN
│ │     GUID:             697a6a95-464e-5b90-a46e-73beab89316e ← USB\VID_17EF&PID_30B5
│ │   
│ ├─Ethernet Adapter:
│ │     Device ID:        9c23388c7f670d3c2f73238cba8f21800a7dfe7d
│ │     Current version:  1.73.4
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             abbc9b3e-43fe-538f-bfaf-b7e74b3fd47c ← USB\VID_17EF&PID_30B4&CID_I255
│ │   
│ ├─Power Delivery:
│ │     Device ID:        ac03bb2ede10a2bce41c65161deac8010cde8bea
│ │     Current version:  12.5.49
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             e83aba6c-99b3-5e87-b982-ba57a1bf750f ← USB\VID_17EF&PID_30B4&CID_PD
│ │   
│ ├─Thunderbolt 4 Controller:
│ │     Device ID:        1a8a444be38156d045a25684d11031ce738688ca
│ │     Current version:  41.81.06
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             d6408693-09a5-5f0f-b492-275b5ed16d10 ← USB\VID_17EF&PID_30B4&CID_TBT4
│ │   
│ ├─USB 2 Hub:
│ │     Device ID:        ea5edabd1b89209486f2be2331ee482e599d1260
│ │     Current version:  V1003
│ │     Vendor:           Lenovo (USB:0x17EF)
│ │     GUID:             4eab6a92-0ec3-5221-ac0b-c841655f7989 ← USB\VID_17EF&PID_30B4&CID_USB2
│ │   
│ └─USB 3 Hub:
│       Device ID:        1896db1f7dc216a3a7c54a29870169178cedc06d
│       Current version:  5563
│       Vendor:           Lenovo (USB:0x17EF)
│       GUID:             0e5b868b-34a4-5753-8f2e-fe5326494d7a ← USB\VID_17EF&PID_30B4&CID_USB3

```

rustyscottweber, 2024-09-05:
This issue is consistent with 1.0.13,  1.0.16, and 1.0.18.  So, I don't think this is actually related to this specific version (1.0.18). I've actually also tested installing/re-installing drivers and even using a whole other lenovo laptop to attempt the update. However, this is the first sign of someone else having the same issue as what I am seeing. Let me follow up in that thread to see if there might be some similarities or something that might either absolve or incriminate fwupd.

marcel151, 2024-09-06:
Mh, I was able to update to 1.0.13 and 1.0.16 just fine in the past... This is the first time I had this issue.

rustyscottweber, 2024-09-09:
@olli1337 , Are you running Linux at all?

victor-cheng, 2024-09-11:
If you power cycle and perform the operation again, does the error still occur?

marcel151, 2024-09-11:
Yes, either power cycling or holding the button to reset dock. Still won't update.

villeturpeinen, 2024-09-13:
Same issue here, update utility gives "MCU check mode timeout" on ThinkPad P14s Gen 3, but works on ThinkPad X1 Carbon G12. Both laptops have Windows 11 with latest 2024-09 CU installed, and every update from Lenovo Commercial Vantage.

rustyscottweber, 2024-09-13:
It should be noted as well, that this firmware upgrade works just fine in Linux.

closed 2024-09-13 by rustyscottweber

marcel151, 2024-09-13:
> It should be noted as well, that this firmware upgrade works just fine in Linux.

Nope it didn't. At least not for all revisions of the dock. https://github.com/fwupd/firmware-lenovo/issues/452#issuecomment-2338311278

tundratier, 2024-09-19:
Has this been solved by any windows user? This thread is the only search result on Google when searching for "MCU check mode response timeout!"

mcdowellster, 2024-10-22:
Can confirm the same issue on Windows 11 with this dock. The firmware update utility tries every single day and it's driving me nuts.

ClaudiuPetrut, 2025-01-11:
The same problem occurs with the ThinkPad TBT4 Dock (40B0), Firmware Update Utility 1.0.18, and the Lenovo ThinkPad E16 Gen 1, Windows 11.

ThinkPad TBT4 Dock Firmware Update Utility 1.0.18
===================================================================
Date: 2025/01/12
Time: 01:01

Dock Name = ThinkPad Universal Thunderbolt 4 Dock
Product = 40B0
SN = 1S40***********
MAC Address = 00-00-00-00-00-00
MCU CMD timeout in Check Mode1

.Retry in MCU CMD timeout in Check Mode1

MCU CMD timeout in Check Mode1

.Retry in MCU CMD timeout in Check Mode1

MCU CMD timeout in Check Mode1

MCU check mode response timeout!

dechamps, 2025-04-15:
As discussed on the [Lenovo forum](https://forums.lenovo.com/t5/Displays-Options-and-Accessories/ThinkPad-Universal-Thunderbolt-4-Dock-40B0-Update-Loop/m-p/5185713?page=4#6576334), this seems to be a widespread issue that is not specific to fwupd: I encountered the same issue despite never having used fwupd.

Ironically, it works if you use a non-Lenovo computer to flash the dock.

elajoie, 2025-06-12:
Worked when I also did on my non Lenovo system

marcel151, 2026-09-15:
Just to do a follow up. Logitech seems to be the issue here, the following helped in my case. 
`Figured this one out. Logitech/LogiOptions+ or something Logitech related was somehow locking access to the dock MCU. Killing anything related to Logitech in the task manager allowed the FW Update Utility to read the MCU and update the dock.`
[https://forums.lenovo.com/t5/Displays-Options-and-Accessories/Thinkpad-Universal-TB4-Dock-firmware-1-18-persistent-failure/m-p/5318197?page=3#10163923](url)

Tried to update to 1.0.20, only worked killing all Logitech related beforehand.

Residents-evil, 2026-10-14:
For those still looking for answers
Kill the service named "Logitech LampArray Service"
Source: [This post](https://gist.github.com/LuCkEr-/c454d7995f97412af9cc5d9544c24d62)

Solution worked in W11 Enterprise 25H2 on a Lenovo P16v G1 AMD and Lenovo 40B0 dock