fwupd/firmware-lenovo #468: Fwupd not detect dock update When boot up device without the dock attached (TBT4 40B0 - firmware upgrade).

rcilto1, 2024-09-12:
When boot up the device without having the dock attached to the device and plug the dock to the device after everything booted up the fwupd will not detect the dock update. If we boot up while the dock is attached to the device the fwupd will detect the update and everything works as expected.

Test realized connecting the dock after the system up.

1 - When we run "fwupdmgr get-devices" the dock is recognized
2 - When we run "fwupdmgr get-updates" the fwupdmgr doesn't detect any updates of the dock

To reload the firmware we need to run "fwupdtools get-updates", after that we run again "fwupdmgr get-updates" and everything works as expected.

Setup:
Thinkpad T14 Gen5 - Ubuntu 22.04 OEM - Fwupd version 1.9.23 (snap)
Thinkpad T14 Gen2 - Debian derivative (testing) - Fwupd version 1.9.20 (debian repository)

Feel free to request some information and logs.

Regards,
Rodnei Cilto

mrhpearson, 2024-09-12:
@hughsie - afraid need your help with this one - I wasn't sure how to debug further. Any suggestions on next steps please?
Thanks!

hughsie, 2024-09-12:
No worries at all. So it sounds like there's somehow a difference in what fwupd sees from hotplug to coldplug. Can you grab the output of `fwupdmgr get-devices --json` for the hotplug and coldplug cases please. Thanks!

rcilto1, 2024-09-13:
Hello!!!

Follows the output using the --json, I filtered using the Guid:

When the dock is attached while the OS start:

[recognize_upgrade_get-devices.json](https://github.com/user-attachments/files/16996487/recognize_upgrade_get-devices.json)

Attached the dock after the OS start (or doing a hot plug):

[not-recognize_upgrade_get-devices.json](https://github.com/user-attachments/files/16996524/not-recognize_upgrade_get-devices.json)

Thanks,

hughsie, 2024-09-16:
Huh, weird -- but it supports what you're saying:

	@@ -20,7 +20,6 @@
	       "Flags" : [
		 "updatable",
		 "require-ac",
	-        "supported",
		 "registered",
		 "dual-image",
		 "signed-payload"

Okay this next step is more involved. Can you do:

* insert dock
* sudo /usr/libexec/fwupd/fwupd -vv --no-timestamp &> coldplug.txt
* ctrl+c
* remove dock

then:

* sudo /usr/libexec/fwupd/fwupd -vv --no-timestamp &> hotplug.txt
* insert dock
* wait 5 seconds
* ctrl+c

...then attach the both (big!) files to this issue please.

rcilto1, 2024-09-17:
Hi guys!!

**Step 1 - Ubuntu 22.04 OEM**
I tried to capture the logs using the Ubuntu 22.04 OEM with fwupd version 1.9.24 installed by snap, and I had some issues:

I followed the steps mentioned in the previous comment but the binary doesn't exist, I found the binary in other folder "/snap/fwupd/6368/libexec/fwupd/fwupd" but didn't work "error while loading shared libraries: libfwupdengine.so: cannot open shared object file: no such file or directory". 

Than I tried to use the fwupdmgr located in the folder "/snap/bin/fwupdmgr" with the parameter "-vv", this package not supported the parameter "--no-timestamp",  and some errors happened (picture attached).

[image]

**Step 2 - Ubuntu 24.04 Live**
I started the Ubuntu 24.04 Live and updated the fwupd to 1.9.24 using Ubuntu repositories, now I found the binary in the folder "/usr/libexec/fwupd".

Follows the logs and information that I analysed using the commands to capture the logs:

Logs:
[coldplug-2404.txt](https://github.com/user-attachments/files/17028305/coldplug-2404.txt)
[hotplug-2404.txt](https://github.com/user-attachments/files/17028306/hotplug-2404.txt)

Information:
1 - I identified the issue, with a little difference, when we doing the hotplug the dock is not recognized more by the fwupd, using the fwupdmgr installed by snap on Ubuntu 22.04  the dock is recognized but not show the updates.

[image]

2 - If I run the command to capture the logs or only run the binary "fwupd" located in the folder "/usr/libexec/fwupd", after that run the "fwupdmgr get-updates" the dock found the firmware available to update (Ubuntu 24.04).

[image]

Thanks for help,

hughsie, 2024-09-19:
@rcilto1 thank you for the debugging logs --with those I managed to work out what was happening. For some reason on hotplug the devices get enumerated in a slightly different order, which is fine -- but we were only re-evaluating the SUPPORTED flag "one way". After some review, I'll backport that patch to 1_9_X so it can be included in the next 1.9.x release that distros are using.

closed 2024-09-19 by hughsie