fwupd/firmware-lenovo #471: Thinkpad T14s Gen 1 Intel not updating since 2022

vuori, 2024-09-28:
Thinkpad T14s Gen 1 Intel firmwares don't appear to be updating on LVFS. Latest on support site is 1.32, latest on LVFS is 1.23. It seems like updates stopped in early 2022: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2YET.firmware

Meanwhile the AMD version of the same model is up to date: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1CET.firmware

Updating to 1.32 from Windows didn't cause any immediately apparent problems on Linux side. Lenovo fixed LogoFail in 1.29 so getting these rolling on LVFS would be good for users who don't check Vantage or the support site.

mrhpearson, 2024-09-30:
Thanks for reporting and my apologies - it's not acceptable that they've not delivered the updates. Looks like EC has not been updated in four years.

I've created ticket LO-3320 for internal tracking. It's a holiday in China this week so I won't get a reply until next week - but I'm escalating this as a priority.

vuori, 2024-10-08:
Thanks for the update. I already applied the update manually so I can't test the LVFS release, but at least version BIOS 1.32 is now showing up in `fwupdmgr get-releases` for me.

closed 2024-10-08 by vuori