fwupd/firmware-lenovo #472: ThinkPad X1 2-in-1 Gen 9 missing firmware update

siwei-luo, 2024-10-06:
- Looks like BIOS `1.36` with ECP `1.24` has been released on the website already: https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-2-in-1-gen-9-type-21ke-21kf/21ke/21kecto1ww/pf52s3b5/downloads/driver-list/component?name=BIOS%2FUEFI
- Weren't there Linux compatible artifacts in the past for manual download before?

mrhpearson, 2024-10-07:
Thanks for the note.

FW is on LVFS - but unfortunately still under embargo. It looks like it has been tested, so I've raised internal ticket LO-3334 asking for it to be moved to stable channel.

ChiWei-Chen, 2024-10-09:
Hi,
BIOS and EC firmware released to LVFS stable today, please let us know if this doesn't work for you. Thanks! 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3YETXXW.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3YHTXXW.firmware

siwei-luo, 2024-10-09:
Thank you all for the quick resolution! Had I known it would be this fast, I wouldn't have done it in advance.

Since the latest firmware includes CVE fixes, I went ahead and used a Windows booted from a USB stick to complete those updates.

So I can't test them anymore myself :(

closed 2024-10-09 by ChiWei-Chen