fwupd/firmware-lenovo #474: [Lenovo ThinkBook] FWUPD Request to Lenovo

Cyber-Typhoon, 2024-10-15:
Hi -

I see the thinkpad and even legion family of firmware available in the FWUPD. Can Lenovo please add Thinkbook firmware updates in the FWUPD too?

Thanks

hughsie, 2024-10-15:
I don't think it's in the Linux support agreement, but I can ask Mark for a more public statement.

mrhpearson, 2024-10-16:
I can't take this as a general request I'm afraid (sadly). 

If a platform is in our certified Linux program (www.lenovo.com/linux) then doing FW updates on LVFS is a requirement - and if the FW team aren't delivering those I do have a hammer I can use (with a caveat that for older platforms the hammer is sometimes replaced with a foam noodle...)

We do certify some Thinkbooks - but not the whole portfolio. So it's a bit platform by platform I'm afraid.

closed 2024-10-16 by mrhpearson

Cyber-Typhoon, 2024-12-31:
> We do certify some Thinkbooks - but not the whole portfolio. So it's a bit platform by platform I'm afraid.

Could you please point which Thinkbook model has support?

I'm afraid this isn't even the case? Please consider escalate this request with Lenovo so we have future models supported. If we don't bring those customers requests I'm afraid we will never see improvements.

mrhpearson, 2024-12-31:
Check www.lenovo.com/linux for which platforms have been certified with Linux. We keep that page up to date as certification completes.

Cyber-Typhoon, 2024-12-31:
As I mentioned, no ThinkBooks so, prior information that there are ThinkBooks supported is incorrect.

mrhpearson, 2024-12-31:
Huh...that's strange. Not sure why they've not been added, I'll ask the team to check and update the page in the New Year
.
As an alternative you can see which Thinkbooks have been Ubuntu certified:
https://ubuntu.com/certified/laptops?q=thinkbook&limit=20&category=Laptop

Cyber-Typhoon, 2024-12-31:
Quite weird, the ThinkBooks are a single yearly new hardware release rollout. Different from Thinkpads, keeping the fwupd support for it should be a smooth generation path. What is the criteria to skip a gen? Why not have support for the new gens? Thinkbook 14 gen 6 for example. Next year we should have gen 7 and how we can find out if it will have support? What is the agreement and commitment that Lenovo is putting of getting Linux onboard of the Thinkbook customers? Should we just avoid Thinkbooks and aim to get Thinkpads because it is officially guaranteed that it will have support?

Danny3, 2025-02-11:
Well since, proper Linux support seems to be ignored or avoided, I have decided that from now on for new purchases I will start with the Framework laptop and the ones that KDE recommends:
https://kde.org/hardware/

Thanks Lenovo for not caring at all about proper security and Linux support!

hughsie, 2025-02-11:
@Danny3 I'm not sure that's completely fair; Lenovo are one of the biggest uploaders to the LVFS now.

Danny3, 2025-02-13:
> [@Danny3](https://github.com/Danny3) I'm not sure that's completely fair; Lenovo are one of the biggest uploaders to the LVFS now.

Hmm, it seems you are right:
https://fwupd.org/lvfs/vendors/
The only thing I have to do now is to meet with 2 friends I helped buy Lenovo laptops on which I've helped them install Linux + Windows in duall-boot mode.
I need to see if their laptops got any firmwares through LVFS before I make my mind completely about this vendor for future purchases and recommendations.

mrhpearson, 2025-12-09:
Saw your posting in the forum - I'll reply there but as a note you posted it in the Linux pages so you're going to hit the Linux team and not the Thinkbook team. It should get picked up and analysed by the Lenovo tools at least.
Confirmed the Feedback tool is broken (Gah!) - I'll let the web team know about that.

I did also get an update as to why we don't list the previously Ubuntu certified Thinkbooks on the lenovo.com/linux page - I'd actually forgotten the reason. I asked for them to be pulled from the page a while back - because of lack of LVFS updates. There are some long (and frankly poor) reasons for that, related to internal process and product requirement specs - but it was a miss, and they didn't/couldn't fix it. This means they aren't recommended as systems for Linux users.

The requirements are fixed for future platforms - but I'm not allowed to comment on which platforms exactly that will be I'm afraid.

I'll comment on the brand shame and other parts on the forum - as that's not LVFS related.