fwupd/firmware-lenovo #475: UEFI/BIOS two versions behind on Thinkpad T14s Gen3 (AMD)

dal00, 2024-10-16:
The Thinkpad T14s Gen3 (AMD) lags two versions behind with the UEFI/BIOS updates.
I currently have 1.40 installed but the latest is 1.42. There are some security issues fixed that would be good to roll out.

https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t14s-gen-3-type-21cq-21cr/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C

mrhpearson, 2024-10-16:
I'm seeing version 1.42 on LVFS and it's been there ~1 month.

https://fwupd.org/downloads/b7a3f73b284479cb98e241827dc6786d266462f3158b072a830661dcedb01f60-Lenovo-Thinkpad-X13-T14s-AMD-SystemFirmware-1.42.cab

Can you confirm that you're not being offered the update? If you're not can you provide the output from 'fwupdmgr get-devices' please?

Thanks
Mark

dal00, 2024-10-16:
Here it is, as far as I can tell I only have 1.40

```
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [ -                                     ]
Successfully downloaded new metadata: 5 local devices supported
# fwupdmgr get-devices 
LENOVO 21CQCTO1WW
│
├─ELAN0678:00 04F3:3195:
│     Device ID:          f8bebdac7d9634f3890f6f6cf3eec0908c2c700b
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404105
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        fb2f6a4f66c638829126eef33d281666a63c1e64
│ │     Summary:          Rembrandt Generic VBIOS
│ │     Current version:  032.017.000.055.000000
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUID:             6d1757cb-7371-5d9a-9b7a-4be4b21ef0b9 ← AMD\113-REMBRANDT-X37
│ │     Device Flags:     • Internal device
│ │   
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.28.01.6f
│ │     Bootloader Version:00.28.01.6f
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9844da3e-1df2-52fe-9413-d4378af6221e ← PCI\VEN_1022&DEV_1649
│ │                       fbbfd456-73f4-5488-a520-f5e4aaa5b9e1 ← PCI\VEN_1022&DEV_1649&SUBSYS_17AA50B4
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  69.62.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             79307ae6-a2ea-52e1-bf56-6abbaf3547ad ← /sys/devices/platform/AMDI0007:00
│       Device Flags:     • Internal device
│     
├─EM05-G:
│     Device ID:          593e609e6f7b35b0fe53f583b7f5548094b04c4f
│     Summary:            Mobile broadband device
│     Current version:    EM05GFAR07A07M1G_01.021.01.021
│     Vendor:             Quectel (USB:0x2C7C)
│     GUIDs:              7d697f50-0b44-559a-a735-3478250e6053 ← USB\VID_2C7C&PID_030A&REV_0200&CARRIER_ROW_GENERIC_3GPP&NAME_EM05GFA
│                         c97e7af4-06f0-5e0e-8fe0-631bceeec6a6 ← USB\VID_2C7C&PID_030A&REV_0200&CARRIER_ROW_GENERIC_3GPP
│                         0322d845-8497-5cbe-bfd5-2c2a6c9ea129 ← USB\VID_2C7C&PID_030A&REV_0200
│                         df71f261-0f4c-55d5-8c0c-e6d2840feee5 ← USB\VID_2C7C&PID_030A
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Signed Payload
│   
[... 178 lines cut ...]
│     GUID:               75d78aef-5e7f-0afe-b68a-aef4c52ccddd
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          e02dffc81c45a5b410a6b30eeb6648a828a9bd65
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    16781317
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               a9e5fe08-4318-4f97-a930-91001c6af104
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   

```

mrhpearson, 2024-10-16:
Hmm, you are on 1.40, and you should be offered that update :/

Can you provide the output from 'sudo fwupdtool -vv --plugins uefi_capsule update' please? 

Mark

dal00, 2024-10-16:
Interesting, that offered me to upgrade to 1.42 for some reason

```
# fwupdtool -vv --plugins uefi_capsule update
15:01:41.987 FuDebug              verbose to debug (on console 1)
15:01:41.987 FuEngine             starting fwupd 1.9.25…
15:01:41.988 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]15:01:42.001 FuConfig             loading config /etc/fwupd/fwupd.conf
15:01:42.001 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
15:01:42.001 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
15:01:42.001 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
15:01:42.001 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
15:01:42.001 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
15:01:42.001 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
15:01:42.002 FuConfig             ::configuration loaded
15:01:42.002 FuRemoteList         loading /usr/share/fwupd/metainfo
15:01:42.002 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:01:42.002 XbSilo               file: b3550df7-b17d-9c57-fe3a-e261eec532c9, current:b3550df7-b17d-9c57-fe3a-e261eec532c9
15:01:42.002 XbSilo               loading silo with existing file contents
15:01:42.002 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:01:42.002 FuRemoteList         ::remote_list changed
15:01:42.002 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:01:42.003 FuRemoteList         ::remote_list changed
15:01:42.003 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:01:42.003 FuRemoteList         ::remote_list changed
15:01:42.003 FuEngine             priority local and vendor-directory is not download remote, so bumping
15:01:42.003 FuRemoteList         ordering lvfs-testing=lvfs+1
15:01:42.003 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]15:01:42.003 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:01:42.003 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:01:42.003 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:01:42.003 Jcat                 loaded 1 certificates
15:01:42.003 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:01:42.003 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:01:42.003 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:01:42.003 Jcat                 loaded 1 certificates
15:01:42.003 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
15:01:42.003 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:01:42.003 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]15:01:42.003 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:01:42.004 FuHistory            got schema version of 12
Loading…                 [*                                      ]15:01:42.011 FuPlugin             load(/usr/lib64/fwupd-1.9.25/libfu_plugin_modem_manager.so)
15:01:42.012 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:01:42.012 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
15:01:42.012 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:01:42.012 XbSilo               file: 1065f168-160c-fb91-9e8a-d04be0965cdc, current:1065f168-160c-fb91-9e8a-d04be0965cdc
15:01:42.012 XbSilo               loading silo with existing file contents
Loading…                 [*                                      ]15:01:42.014 FuConfig             loading config /etc/fwupd/fwupd.conf
15:01:42.014 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
15:01:42.014 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
15:01:42.014 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
15:01:42.014 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
15:01:42.014 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
15:01:42.014 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
15:01:42.014 FuConfig             ::configuration loaded
15:01:42.014 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x7e
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x3
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xbc3
  structure_table_addr: 0x73712000
15:01:42.014 FuStruct             SmbiosStructure:
  type: 0x86
  length: 0xd
  handle: 0x0
[... 1405 lines cut ...]
15:01:42.504 FuRelease            remote lvfs has kind=download and so marking as trusted
15:01:42.504 FuRelease            remote lvfs has kind=download and so marking as trusted
15:01:42.504 FuRelease            remote lvfs has kind=download and so marking as trusted
15:01:42.504 FuEngine             add trusted-report to com.lenovo.ThinkPadR22ET.firmware:0.1.21 as trusted:   Flags:                from-oem

[... cut ...]
║ • Enhancement to address security vulnerability.                             ║
║ • Enable TW WIFI6E for TP products                                           ║
║                                                                              ║
║ LENOVO 21CQCTO1WW must remain plugged into a power source for the duration   ║
║ of the update to avoid damage.                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
```

dal00, 2024-10-16:
Seems like the update was a success and I've uploaded the report as well.

closed 2024-10-16 by dal00