fwupd/firmware-lenovo #477: Bricked lenovo p14s gen 1 laptop after updating the bios

mouadriyad, 2024-10-31:
Hi,

3 weeks ago, i updated the bios from the fwupd because i had two usb c ports not working with my smartphone. As far as I know, the update seemed to be successful  since the laptop booted normally.

The morning after, I observed that the red led on the back was blinking slowly even if i turned off the laptop at night. When i checked, the laptop was unresponsive with a blinking power button and keyboard light  at a low pace. 

I removed the battery and cmos with no success, but it started once, and i downgraded the bios from 1.5 to 1.49, and i ended up with the same result.

hughsie, 2024-10-31:
Hmm, that sounds more like a hardware failure than a firmware update failure. @mrhpearson any debugging tips for this situation please?

mrhpearson, 2024-11-01:
I'm a bit confused with title saying 'bricked'  - yet BIOS can be downgraded? That suggests something is running.
Also not sure where a smartphone is involved. Bit confused I'm afraid :)

The red led blinking slowly normally means it is suspended. Which doesn't fit with the description. What happens is press and hold the power button for 30s?

This does sound more like a HW issue - but afraid will need more details as to what exactly isn't working.

 - Can you power up the system? If so does it get to the BIOS setup screen? Grub? Linux?

 - If it powers up can you give some logs from a power up and shutdown cycle (boot after doing the cycle and run 'journalctl -k -b -1'). You can send the logs to mpearson-lenovo at squebb dot ca with this issue referenced.

Thanks
Mark