fwupd/firmware-lenovo #480: fwupd fails to identify the computer correctly when updating the UEFI firmware.

rochfort1, 2024-11-13:
fwupd does not identify the host computer correctly when attempting to update the UEFI firmware.

**Sequence of events**
1. GNOME advises that a "ThinkPad P16 Gen2 System Update" is available.
2. I run fwupdmgr update and wait for the operating system to restart.
3. I run fwupdmgr get-history to see the following:
LENOVO 21KXCTO1WW
│
└─System Firmware:
  │   Device ID:          ff73260cd8152eb987360fa09de2c45801dc954c
  │   Previous version:   0.1.13
  │   Update State:       Failed
  │   Update Error:       failed to update to 117474352: error-auth-error
  │   Last modified:      2024-11-13 03:42
  │   GUID:               a7a2491c-9ab6-41ff-a539-3635b27f85c2
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P16 Gen 2 System Update:
        New version:      0.1.55
        Remote ID:        lvfs
        Release ID:       96082
        Summary:          Lenovo ThinkPad P16 Gen 2 System Firmware
        License:          Proprietary
        Size:             28.8 MB
        Created:          2023-03-14
        Urgency:          High
        Tested by Lenovo:
          Tested:         2024-07-23
          Distribution:   ubuntu 20.04
          Old version:    0.1.52
          Version[fwupd]: 1.7.9
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
                          • Tested by trusted vendor
        Description:      
        Lenovo ThinkPad P16 Gen 2 System Firmware Version 1.55[New functions or enhancements]Support Lenovo Memory Self Repair(PPR) function.[Problem fixes]Fixed an issue Dock Mac address isn't cloned to system Mac address.Fixed an issue unexpected Windows Auto Repair action might started after BIOS update
        Issues:           CVE-2023-45237
                          CVE-2024-0762
                          LEN-154756
                          CVE-2023-38575
                          CVE-2023-39368
                          CVE-2023-28746
                          CVE-2023-45236
                          CVE-2023-45235
                          CVE-2023-45234
                          CVE-2023-45233
                          CVE-2023-45232
                          CVE-2023-45231
                          132380
                          CVE-2023-45230
                          CVE-2023-45229
                          CVE-2023-49141
                          CVE-2023-45733
        Checksum:         f9f47c8f91184ce7405bbc87eaf67d381aa85e9b382aba7817e11c35f7bd26aa

**Expected behaviour**
The firmware should update without error.

**fwupd version information**
1.9.26-2 (fwupd-amd64-signed 1:1.7+1)

fwupd was installed as part of the Debian distribution installation.

**Result**
get-history shows that the system firmware failed to update and reports the following error message:

> failed to update to 117474352: error-auth-error

**Comment**
I notice that get-history is reporting the wrong computer and the wrong UEFI firmware version.
At the top of the dump, get-history correctly identifies the computer as a Lenovo 21KXCTO1WW which is a ThinkPad P16v Gen 2 (Intel). But elsewhere it refers to a ThinkPad P16 Gen 2.
The dump says that the previous firmware version is 0.1.13 and that the new version is 0.1.55. That is curious. I trawled through Lenovo's BIOS/UEFI support pages to find the following:

P16 Gen 2 (Type 21FA, 21FB) Laptop (ThinkPad) uses UEFI firmware version 1,56
P16v Gen 1 (Type 21FE, 21FF) Laptop (ThinkPad) uses UEFI firmware version 1.55/1.09. (I don't know what the 1.09 part means).
P16v Gen 2 (Type 21KX, 21KY) Laptops (ThinkPad) uses UEFI firmware version 1.13. The current version, as reported by GNOME is N44ET30W (1.13)

I suspect that fwupd is not identifying the computer correctly. It might be attempting to update the UEFI firmware to the version for the P16v Gen 1 (Type 21FE, 21FF). Luckily it is failing.

**Additional questions**

- Operating system and version: Debian 13 (trixie/sid)
- Have you tried rebooting? Many times
- Is this a regression?  No
- Are you using an NVMe disk?  No
- Is secure boot enabled?  No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Off

mrhpearson, 2024-11-13:
Strange - your GUID does match with the P16 G2 FW, so that's why it is being offered as an update.
I've created internal ticket LO-3400 to ask the FW team for guidance on next steps.

mrhpearson, 2024-12-06:
Should be fixed now - P16 G2 and P16v G2 FW has been updated on LVFS.
Please let me know if you still see problems.

rochfort1, 2024-12-07:
I think the problem has been fixed. _fwupdmgr get-updates_ now reports that the latest firmware versions are installed on the computer.

closed 2024-12-07 by rochfort1