fwupd/firmware-lenovo #483: T16 Gen 1: Failed ME firmware. No sound on Ubuntu 22.04/24.04

pandancode, 2024-11-16:
I applied the last firmware update on Ubuntu 22.04 which caused:
1. 'Failed ME firmware' update screen on boot
2. Integrated audio (speakers + mic) stopped working

I tried various things with no success, then updated to 24.04 altogether. The following fixed driver issue with the integrated speakers, but the microphone is still not working:

```
$ nano /etc/modprobe.d/alsa-base.conf

options snd-hda-intel dmic_detect=0
```

This is similar to [this](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135), [this](https://github.com/fwupd/firmware-lenovo/issues/280), and various other threads. 

```
$ fwupdmgr get-history
LENOVO 21BV008UHH
│
├─Embedded Controller:
│ │   Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
│ │   Previous version:   0.1.13
│ │   Update State:       Success
│ │   Last modified:      2024-08-22 14:56
│ │   GUID:               c6fdd612-fdf3-4bff-bc77-347bcdb0219f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─(null) Update:
│       New version:      0.1.13
│       Remote ID:        lvfs
│       Description:      
│       The vendor did not supply any release notes.
│     
├─Intel Management Engine:
│ │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│ │   Previous version:   1.32.2418
│ │   Update State:       Success
│ │   Last modified:      2024-08-22 14:57
│ │   GUID:               cb05acd1-54fe-4afe-958d-b1d515825077
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─(null) Update:
│       New version:      1.32.2418
│       Remote ID:        lvfs
│       Description:      
│       The vendor did not supply any release notes.
│     
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Previous version:   10.01.3478575
│ │   Update State:       Success
│ │   Last modified:      2024-08-22 14:57
│ │   GUID:               d67c4925-e30f-53e6-abc6-e7af0e4c5884
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Is in bootloader mode
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus Fingerprint Reader Update:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10696
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425.5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Tested by Wistron:
│         Tested:         2024-06-06
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3333223
│         Version[fwupd]: 1.9.15
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Release Flags:    • Trusted metadata
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│       Checksum:         da76c521b478c2c1c03eaf56a91e8498c4abafc0fa6a8d4fd4b067277219e1a4
│     
├─TPM:
│ │   Device ID:          9b4ca17462f3f409e1d22aa2a862cbbeffc6a649
│ │   Previous version:   0.15.22
│ │   Update State:       Success
│ │   Last modified:      2024-08-22 14:57
│ │   GUID:               8d5056e5-7a0a-4bcd-bf92-7d16212b72aa
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─(null) Update:
│       New version:      0.15.22
│       Remote ID:        lvfs
│       Description:      
│       The vendor did not supply any release notes.
│     
└─System Firmware:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Previous version:   0.1.20
  │   Update State:       Success
  │   Last modified:      2024-08-22 15:18
  │   GUID:               30fbf24a-e1da-4840-84f6-6d7deb630e1d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─(null) Update:
        New version:      0.1.20
        Remote ID:        lvfs
        Description:      
        The vendor did not supply any release notes.
```

```
$ fwupdmgr update --show-all
Devices with no available firmware updates: 
 • CT1000P5PSSD8
 • Intel Management Engine
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Integrated Camera
 • Intel Management Engine
 • Prometheus IOTA Config
 • TPM
 • Prometheus
 • UEFI dbx
```

Any advice on how to fix these?

kmauleon, 2024-11-18:
hi @pandancode let me forward this to ODM CO and will feedback once they replied... thank you very much

kmauleon, 2024-11-18:
hi @pandancode can you please try below solution if it will work for you as advised by ODM CO

Changing the file /etc/modprobe.d/alsa-base.conf
And adding: options snd-hda-intel model=auto

kmauleon, 2024-11-18:
@pandancode already forwarded your results to ODM CO

BTW not a good workaround though but a user tried this one and it fixed the issue >>> https://github.com/fwupd/firmware-lenovo/issues/280#issuecomment-1406307948

if nothing else works, would you be willing to update MEFW via USB EFI?  i can ask ODM CO to provide the necessary files and procedure for you... please advise.  thank you very much.

pandancode, 2024-11-19:
I don't even have Windows installed on this machine...

Sure, wouldn't mind going the MEFW route. Thank you!

kmauleon, 2024-11-20:
noted... requesting ODM CO to provide files and procedure... thank you!

kmauleon, 2024-11-21:
@pandancode can you please email me at kmauleon at lenovo dot com.... thank you very much!

kmauleon, 2025-01-02:
hi @pandancode not sure but have you tried this one >> [#295](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135)

pandancode, 2025-01-18:
Thanks - it's finally working. I believe the path to fixing this is as follows:
1. Run all under `fwupdmgr reinstall`, no need to restart after each step; just run 1-6 and then reboot. I have to say, I have done this in the past, but today, it (successfully) removed the `ME firmware update failed` line at startup
2. I reinstalled the following:
```
sudo apt install --reinstall firmware-sof-signed
sudo apt install --reinstall alsa-firmware
sudo apt install --reinstall alsa-utils
```
3. Removed all custom lines from `/etc/modprobe.d/alsa-base.conf` like
```
options snd-hda-intel dmic_detect=0
```

This was a bit bizarre. Thanks so much @kmauleon! <3

closed 2025-01-18 by pandancode