fwupd/firmware-lenovo #484: X1 Nano Gen1 Stuck at Self Healing backup progressing 100%

darthcloud, 2024-12-03:
I started the BIOS update from KDE Discover Updates so not sure what was my original version.

I recall the BIOS update process at one point did a memory training step IIRC.
After that it got to the "BIOS Self Healing backup progressing" screen.

It got stuck at 100%, I let the laptop run for an hour+ at that screen and it didn't progress futher.
Turning the laptop off & on again resulted in going back to the "BIOS Self Healing backup progressing" screen and getting stuck again at 100% all the time.

I tried the pinhole reset with no success, remove battery didn't help either, also removed the NVMe card didn't help. The laptop was basicly a brick.

In the end I had to desolder the GD25R256D serial flash IC from the motherboard and reflash it with a BIOS image I found online with my minipro programmer.

That did revive the laptop!

Here the logs I got back from fwupdmgr after fixing it:
fwupdmgr get-history
LENOVO 20UN0008US
│
├─Embedded Controller:
│ │   Device ID:          4d37570329b29c27ef29119339a362db210978c4
│ │   Previous version:   0.1.10
│ │   Update State:       Success
│ │   Last modified:      2024-11-30 20:04
│ │   GUID:               9cba69ff-4c8b-4398-aede-7c39450174a3
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Nano Gen 1  Embedded Controller Update:
│       New version:      0.1.47
│       Remote ID:        lvfs
│       Release ID:       51228
│       Summary:          Lenovo ThinkPad X1NanoGen1 Embedded controller Firmware
│       Variant:          Machine types: 20UN, 20UQ
│       License:          Proprietary
│       Size:             767.4 kB
│       Created:          2023-06-28
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│       Description:
│       -Add extra check power supplying condition to minimize and avoid 2 FET GATE_VSYS_VBUS are ON at the same time.*Multiple power supply(AC or docking) attach detach rapidly case.-Sign tool changes (CS21, CS22)
│       Checksum:         a6029d8905c2b06e6207b514ff9fb8798878b18ddeba50f12d71bd7945f9a415
│
├─Intel Management Engine:
│ │   Device ID:          c8da521d50beec4d279829f0de0fbe43aeeb11ab
│ │   Previous version:   240.10.1618
│ │   Update State:       Transient failure
│ │   Update Error:       boot entry missing; perhaps 'Boot Order Lock' enabled in the BIOS: no 'Linux Firmware Updater' entry found
│ │   Last modified:      2024-11-30 20:04
│ │   GUID:               6779d186-76a8-4588-bea8-d5b03480f77e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Nano Gen 1 ME Update:
│       New version:      240.49.2573
│       Remote ID:        lvfs
│       Release ID:       98332
│       Summary:          Lenovo ThinkPad X1 Nano Gen 1 Consumer ME Firmware
│       License:          Proprietary
│       Size:             4.0 MB
│       Created:          2024-07-16
│       Urgency:          Medium
│       Tested by Lenovo:
│         Tested:         2024-11-21
│         Distribution:   ubuntu 23.10
│         Old version:    240.45.2411
│         Version[fwupd]: 1.9.23
│       Tested by Lenovo:
│         Tested:         2024-10-15
│         Distribution:   ubuntu 23.10
│         Old version:    240.45.2411
│         Version[fwupd]: 1.9.23
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N2TRN52W
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│                         • Is upgrade
│                         • Tested by trusted vendor
│       Description:
│       Lenovo ThinkPad X1 Nano Gen 1 Consumer ME Firmware
│
│       • Updated PMC to 150.1.20.1042
│       • Updated IOM to 17.24.0.0000
│       • Updated NPHY to 11.225.276.2043
│       • Updated MEFW binary to 15.0.49.2573 production for QS
│       Checksum:         15faa7424c85092ec59798c45afb4c375a0518fbb096cb6437de25ec433632a1
│
└─System Firmware:
  │   Device ID:          cf9797ae74c38c23e726b666588aa9b2bbf2f223
  │   Previous version:   0.1.12
  │   Update State:       Transient failure
  │   Update Error:       boot entry missing; perhaps 'Boot Order Lock' enabled in the BIOS: no 'Linux Firmware Updater' entry found
  │   Last modified:      2024-11-30 20:04
  │   GUID:               6ca01c9d-91d3-4a48-b3d6-e3668c420a21
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Nano Gen 1 System Update:
        New version:      0.1.41
        Remote ID:        lvfs
        Release ID:       8092
        Summary:          Lenovo ThinkPad X1 Nano Gen 1 System Firmware
        Variant:          W-BIOS for Machine types: 20UN, 20UQ
        License:          Proprietary
        Size:             18.3 MB
        Created:          2021-03-15
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is downgrade
        Description:
        Lenovo ThinkPad X1 Nano Gen 1 System Firmware
        Checksum:         8fe23521609c542d05c9e2e3ef9e649c747bb63e360768f2558732d0c1544494

I made a dump of the BIOS chip in the bad state before rewriting it in case you want to take a look.

hughsie, 2024-12-03:
> In the end I had to desolder the GD25R256D serial flash IC from the motherboard

Wowsa, well done for managing that. There's not many of us that can do hot-air rework these days :) Could you upload the old image somewhere please (e.g. zip on github) and Mark can ask the ODM to look.

darthcloud, 2024-12-03:
Here the dump of the serial flash in the bad state:
[x1_nano_g1_bad.bin.zip](https://github.com/user-attachments/files/17991853/x1_nano_g1_bad.bin.zip)

mrhpearson, 2024-12-03:
Kudos on recovering the system.
Internal ticket LO-3432 created - I've asked the FW team for input on what caused the problem.

ChiWei-Chen, 2024-12-12:
Thanks for reporting this. 
The FW team has added a restriction in the metadata to prevent this jump. Also, the latest BIOS 1.64 has been released to the stable channel.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2TET.firmware

darthcloud, 2024-12-21:
Updated everything from fwupd in the shell this time one by one.

Everything worked fine this time. On BIOS 1.64 now.

Thank you!!

Uploaded all the reports.

prismindanalytics, 2025-04-18:
Hi all, I ran into this exact same issue and a Google search led me to this thread — thank you to everyone who shared your experience here.

I have a ThinkPad X1 Nano Gen 1 (Type 20UN), and today after a BIOS update through Fedora update, I’m now stuck on the “BIOS Self Healing backup progressing… 100%” screen. It has remained frozen there for hours.

So far I’ve tried reset hole and unplugged battery and nothing works.

Since my machine is out of warranty, I’m now preparing to use a CH341A programmer and flash the BIOS chip directly, but before I go that route:

Are there any other last-ditch recovery methods I should try first?
And for those who went the direct flash route — is there a .bin file I can use?

Would really appreciate any final tips or confirmations. Thanks again for all the insight in this thread!

prismindanalytics, 2025-04-21:
Here are the logs:

fwupdmgr get-history

LENOVO 20UN0007US
│
├─Intel Management Engine:
│ │   Device ID:          c8da521d50beec4d279829f0de0fbe43aeeb11ab
│ │   Previous version:   240.10.1618
│ │   Update State:       Transient failure
│ │   Update Error:       boot entry missing; perhaps 'Boot Order Lock' enabled in the BIOS: no 'Linux Firmware Updater' entry found
│ │   Last modified:      2025-04-17
│ │   GUID:               6779d186-76a8-4588-bea8-d5b03480f77e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Nano Gen 1 System Update:
│       New version:      240.49.2573
│       Remote ID:        lvfs
│       Release ID:       98332
│       Summary:          Lenovo ThinkPad X1 Nano Gen 1 Consumer ME Firmware
│       License:          Proprietary
│       Size:             4.0 MB
│       Created:          2024-07-16
│       Urgency:          Medium
│         Tested:         2025-04-01
│         Distribution:   ubuntu 23.10
│         Old version:    240.45.2411
│         Version[fwupd]: 1.9.23
│         Tested:         2024-10-15
│         Distribution:   ubuntu 23.10
│         Old version:    240.45.2411
│         Version[fwupd]: 1.9.23
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=N2TRN52W
│       Vendor:           Lenovo
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       Lenovo ThinkPad X1 Nano Gen 1 Consumer ME Firmware
│      
│       • Updated PMC to 150.1.20.1042
│       • Updated IOM to 17.24.0.0000
│       • Updated NPHY to 11.225.276.2043
│       • Updated MEFW binary to 15.0.49.2573 production for QS
│       Checksum:         15faa7424c85092ec59798c45afb4c375a0518fbb096cb6437de25ec433632a1
│    
├─Embedded Controller:
│ │   Device ID:          4d37570329b29c27ef29119339a362db210978c4
│ │   Previous version:   0.1.12
│ │   Update State:       Success
│ │   Last modified:      2025-04-17
│ │   GUID:               9cba69ff-4c8b-4398-aede-7c39450174a3
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Nano Gen 1  Embedded Controller Update:
│       New version:      0.1.47
│       Remote ID:        lvfs
│       Release ID:       51228
│       Summary:          Lenovo ThinkPad X1NanoGen1 Embedded controller Firmware
│       Variant:          Machine types: 20UN, 20UQ
│       License:          Proprietary
│       Size:             767.4 kB
│       Created:          2023-06-28
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       -Add extra check power supplying condition to minimize and avoid 2 FET GATE_VSYS_VBUS are ON at the same time.*Multiple power supply(AC or docking) attach detach rapidly case.-Sign tool changes (CS21, CS22)
│       Checksum:         a6029d8905c2b06e6207b514ff9fb8798878b18ddeba50f12d71bd7945f9a415
│    
└─System Firmware:
  │   Device ID:          cf9797ae74c38c23e726b666588aa9b2bbf2f223
  │   Previous version:   0.1.15
  │   Update State:       Transient failure
  │   Update Error:       boot entry missing; perhaps 'Boot Order Lock' enabled in the BIOS: no 'Linux Firmware Updater' entry found
  │   Last modified:      2025-04-17
  │   GUID:               6ca01c9d-91d3-4a48-b3d6-e3668c420a21
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Nano Gen 1 System Update:
        New version:      0.1.41
        Remote ID:        lvfs
        Release ID:       8092
        Summary:          Lenovo ThinkPad X1 Nano Gen 1 System Firmware
        Variant:          W-BIOS for Machine types: 20UN, 20UQ
        License:          Proprietary
        Size:             18.3 MB
        Created:          2021-03-15
        Vendor:           Lenovo
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        Lenovo ThinkPad X1 Nano Gen 1 System Firmware
        Checksum:         8fe23521609c542d05c9e2e3ef9e649c747bb63e360768f2558732d0c1544494

ChiWei-Chen, 2025-04-25:
@prismindanalytics Thanks for reporting this! It looks like your BIOS and EC are quite old. Could you update them one by one using fwupd in the shell?
Let me know if you need any help!

pganguli, 2025-04-27:
Hi @darthcloud and @prismindanalytics ! I ran into this exact same issue and now my Thinkpad X1 Nano doesn't post anymore. I have ordered a CH341A programmer and will try to flash the EEPROM with a known good BIOS image.

Could you please share the firmware that you guys flashed and which revived the laptop?

Also, after flashing and (hopefully reviving) the currently bricked laptop BIOS, how exactly do I use fwupd from the shell to update it from there to BIOS 1.64 one by one?

prismindanalytics, 2025-04-29:
Thank you @ChiWei-Chen! I'm currently traveling with my laptop but will try upgrading one by one once I'm back in a few days. I noticed in the Badcaps thread that another user is experiencing the same bricking issue, so I wonder if there might be something wrong with the latest firmware.

@pganguli I used the .bin file shared by folks in this thread: https://www.badcaps.net/forum/troubleshooting-hardware-devices-and-electronics-theory/troubleshooting-laptops-tablets-and-mobile-devices/bios-requests-only/3546312-lenovo-x1-nano-gen-1-bricked-where-to-flash-the-bios-fw

prismindanalytics, 2025-05-01:
quick update - Upgrade one by one in the shell works for me!

closed 2025-06-06 by mrhpearson