fwupd/firmware-lenovo #486: ThinkPad X1 2-in-1 Gen 9 missing firmware update

siwei-luo, 2024-12-08:
Looks like BIOS `1.37` with ECP `1.25` has been released on the website already:
https://download.lenovo.com/pccbbs/mobiles/n3yuj12w.html

I'll promise to wait this time, Thanks!

mrhpearson, 2024-12-09:
They're on LVFS and under stable - try doing 'fwupdmgr refresh' to refresh your cache?

closed 2024-12-09 by mrhpearson

mrhpearson, 2024-12-09:
As a note, it generally takes just a little bit longer to get FW on LVFS - just because of the way the FW teams process works (I'm not a fan, but it is what it is). They test and release the LVFS updates once Lenovo site release is complete. I didn't do anything for this case other than to check on the latest status.

Recommend holding off a few days before raising a ticket here for future releases ;)