fwupd/firmware-lenovo #487: unable to update system firmware (0.1.53 to 0.1.55) on T470s until disabling Rollback Prevention

heis2201, 2024-12-10:
Follow-up from https://bugzilla.redhat.com/show_bug.cgi?id=2331279

`sudo fwupdmgr update` found the firmware, downloaded it, asked to reboot, but on reboot BIOS denies updating with a message: "*Secure Flash Authentication failed!*"

Setup:
 - AC power connected, 
 - SecureBoot off,
 - Flash BIOS Updating by End-Users 'Enabled',
 - Secure RollBack Prevention 'Enabled', 
 - Windows UEFI Firmware Update 'Enabled'.

After setting Secure RollBack Prevention to `Disabled`, the firmware updates correctly.

This seems to be some sort of bug as the v.0.1.55 firmware is newer than the v.0.1.53 that the system ran on, so no rollback.

Log file: https://bugzilla-attachments.redhat.com/attachment.cgi?id=2061804

HtwoO, 2026-01-19:
I'm having similar issue trying to upgrade my ThinkPad T16 from 0.1.45 to 0.1.46.

``` shell
CLI> sudo LANG=C.UTF-8 fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.45 to 0.1.46?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release has the following updates:                               ║
║                                                                              ║
║ • Cannot roll back to previous BIOS version after flash this BIOS 1.46.      ║
║ • Enhancement to address security vulnerability.                             ║
║ • Fix issue that system may have SSD corruption.                             ║
║                                                                              ║
║ 21*****018 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: n
Devices with the latest available firmware version:
 • KXG8AZNV512G LA KIOXIA
Devices with no available firmware updates: 
 • Integrated Camera
 • KEK CA
 ...
 • UEFI dbx
 • Windows Production PCA
```

HtwoO, 2026-05-25:
I was having problem with suspend and resuming for my ThinkPad T16 [0]. Finally I decided to upgrade the firmware and in my case the option I needed to change to make the upgrade successful was to change `Security -> UEFI BIOS Update Option -> Flash BIOS Updating by End-Users` to `On` and then I was able to go through this upgrade path `0.1.45 -> 0.1.46 -> 0.1.47 -> 0.1.48`. Although the latest firmware version (as of 2026-05-25) is `0.1.49`, I decided to stay at `0.1.48` for now. I disable the option after firmware upgrade.

Current value for the option can be read while one is on a Linux system with `fwupdmgr`:

```shell
>>> sudo fwupdmgr get-bios-settings | rg --context=1 'biosUpdate'

BIOSUpdateByEndUsers:
  Setting type:         Enumeration
  Current Value:        Disable
  描述:                 BIOSUpdateByEndUsers
  Read Only:            False
```

[0]: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1137142