fwupd/firmware-lenovo #494: After selfhealing bios backup progressing my laptop shutdown. Won't turn back on

Thermoximus, 2025-01-28:
The laptop won't turn back on. The charging light is still in even though the charger is unplugged. Plugged back in, the light above the on button won't flash like usual. I'm worried about the battery, and I have school today. I'll bring to IT, but if they can't do anything please help. I have heard about the paper clip trick but will that make me lose everything in the laptop? 

[image]

hughsie, 2025-01-28:
What exact laptop model? Can you describe what exactly you did please?

Thermoximus, 2025-01-29:
> What exact laptop model? Can you describe what exactly you did please?

Lenovo X13 Yoga Gen 2 (bought 2021). Since the start of this year, it has been booting up occasionally with the message "System has self-healed by restoring BIOS from backup. Please check BIOS version and then update BIOS if need" with a big BEEP. I did update at some point then it disappeared, I notice it mostly happens when my laptop overheats.

However after one gaming session yesterday (my laptop overheated), it keeps popping up every time I reboot the next day even though my laptop is normal temp. This morning, I tried to install an update from Lenovo Vantage, BIOS Utility Tool 10/11. However during installation, it restarts then it boots up into a SCT Flash Utility for Lenovo. Then after it goes into "Self-Healing BIOS backup progressing ... %" I wasn't paying attention, but I think it hit 100% then shutdown thats when the issue above happened.

The light magically turned off during school, and i could finally turn it on. I think this is because it was full when I updated it, but the update needed to be plugged in, so it used more energy than the battery could handle. But the main issue still occurs even though the update was successful this time (it failed multiple times previously which also caused me to not be able to turn it on but not the same charger light issue). Brought it to it, she took one look at it and told my to claim warranty in the BYOD, and still try even though the warranty expired.

I thought hard about the root of the problem. I'm not sure, but last year i tried connecting my Wii Remote to my laptop and the guy said to hop into blue screen, trouble shoot, advance settings, and to disable driver signature enforcement, which almost caused me my drivers if i didn't know my bitlocker code. I havent re-enabled it, nor do i know how to, so could that be the problem?

I researched other people that had the same issue and they said that, entering BIOS Setup, Security, UEFI BIOS update option, turn off secure rollback prevention, then save and exit. However im scared that this will do the same thing and trigger the bitlocker. I don't want to risk it, as I still have to do school, and make this laptop last for another 2 years.

hughsie, 2025-01-29:
> the paper clip trick

I think this is your best best, and no, it won't affect anything on the SSD.

> I notice it mostly happens when my laptop overheats

I think you possibly have a hardware problem, I think you should get it into the service center and let them have a look.

> The light magically turned off during school, and i could finally turn it on.

I think it was inhibiting the system power button, and it probably ran out of battery power. I don't think this is anything you've done with software, it sounds like a hardware device issue I'm afraid.

> However im scared that this will do the same thing and trigger the bitlocker

No idea I'm afraid; I think you should have the recovery key somewhere? Sorry that's more helpful. I'll close this issue as the tracker is really for people having Lenovo firmware update issues with the Linux firmware updater. Good luck!

closed 2025-01-29 by hughsie