fwupd/firmware-lenovo #495: Thinkpad P15v Gen3 Intel - ✘ csme manufacturing mode: Unlocked

tanriol, 2025-02-05:
According to `fwupd security`, my laptop has CSME manufacturing mode unlocked. Is this normal / expected?
<details>
<summary>`fwupd security` partial output</summary>
HSI-1                                                                                                                
✔ BIOS firmware updates:         Enabled                                                                             
✔ csme override:                 Locked
✔ csme v0:16.1.32.2418:          Valid                                                                               
✔ Platform debugging:            Disabled                                                                            
✔ SPI write:                     Disabled                                                                            
✔ SPI lock:                      Enabled                                                                             
✔ SPI BIOS region:               Locked                                                                              
✔ Supported CPU:                 Valid                                                                               
✔ TPM empty PCRs:                Valid                                                                               
✔ TPM v2.0:                      Found                                                                               
✔ UEFI bootservice variables:    Locked                                                                              
✔ UEFI platform key:             Valid                                                                               
✘ csme manufacturing mode:       Unlocked                                                                            
                                                                                                                     
HSI-2                                                                                                                
✔ BIOS rollback protection:      Enabled                                                                             
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ CET Platform:                  Supported
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported
</details>

The laptop `product` in lshw is "21D8A00ACD (LENOVO_MT_21D8_BU_Think_FM_ThinkPad P15v Gen 3)"

hughsie, 2025-02-06:
>  org.freedesktop.fwupd 2.0.3

Can you try with fwupd 2.0.5 please? There were some CSME fixes in 2.0.4 IIRC.

tanriol, 2025-02-06:
Assuming you meant `-vv` instead of `--vv`, here it is

[fwupd-security.out.txt](https://github.com/user-attachments/files/18689890/fwupd-security.out.txt)

hughsie, 2025-02-06:
I really think this machine is in manufacturing mode:

	12:24:13.945 FuStruct             FuMeiCsme11Hfsts1:
	  mfg_mode: 0x1
	  operation_mode: 0x0 [normal]
	12:24:13.945 FuStruct             FuMeiCsme11Hfsts6:
	  force_boot_guard_acm: 0x1
	  error_enforce_policy: 0x3 [shutdown-30mins]
	  verified_boot: 0x1
	  boot_guard_disable: 0x0
	  fpf_soc_lock: 0x1

@mrhpearson what's the best course of action here? @tanriol can you share the machine service tag and tell us when and where you purchased it from please?

tanriol, 2025-02-06:
> [@tanriol](https://github.com/tanriol) can you share the machine service tag and tell us when and where you purchased it from please?

Not sure where I should find the service tag
<details><summary>What I do know</summary>

    product: 21D8A00ACD (LENOVO_MT_21D8_BU_Think_FM_ThinkPad P15v Gen 3)
    vendor: LENOVO
    version: ThinkPad P15v Gen 3
    serial: PF3Z9Q78

Purchased in 2022 in Moscow, Russia. Likely in December, some small retailer I don't remember.
</details>

mrhpearson, 2025-02-06:
Recommend contacting Lenovo support for this one - it's possible you got an unofficial device that had not completed the proper manufacturing release process. Support should be able to tell you the origins of your machine.
My understanding is that sales were stopped in Russia due to the political situation. I'm afraid I can't provide any support for this case.
Mark

hughsie, 2025-02-06:
Thanks Mark, sorry that's not more helpful @tanriol

closed 2025-02-06 by hughsie