fwupd/firmware-lenovo #496: Audio not working and unable to update ME Firmware for Thinkpad X1 Yoga gen 6 running Ubuntu

hcnery08, 2025-02-23:
```fwupdmgr get-devices``` reterns:
```LENOVO 20Y0S2X500
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              ab16bce2-e219-52f5-89a6-fd6b1c4244c0 ← HIDRAW\VEN_04F3&DEV_3187
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e ← HIDRAW\VEN_04F3&DEV_3187&MOD_0002
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2 ← ELANTP\ICTYPE_14&MOD_0002
│                         31a158de-a671-50bf-b787-3bba647a8c49 ← ELANTP\ICTYPE_14&MOD_0002&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Unsigned Payload
│   
├─11th Gen Intel Core™ i7-1185G7 @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ba
│     Vendor:             Intel
│     GUIDs:              2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          34210774738cce62e77e32d3a08a71619df371ba
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.36
│     Minimum Version:    0.1.36
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Intel Management Engine:
│     Device ID:          23156eb4e14af860c0d4e04a401cc82b89298733
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    240.50.2633
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               c1b2be54-d7ed-4e24-a577-7c5f32bb7587
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      88445900443179
│ │   GUID:               448868f0-e05d-5849-8fc4-b8fa1ec16bf5 ← USB\VID_06CB&PID_00FC
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
[... 278 lines cut ...]
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • Prometheus
 • System Firmware
 • UEFI dbx```

When I boot I get ```Failed to recover ME firmware``` then it restarts and eventually boots to ubuntu. But audio does not work and booting takes a while.

I tried the reset button on the bottom, but it doesn't do anything.

I tried downloading the .cab file from the LENOVO website but I get this error: ```requirement 'firmware with version' also needs org.freedesktop.fwupd >= 1.1.0```

But ```fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.27
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.27
runtime   org.kernel                    6.11.0-17-generic```

kmauleon, 2025-02-24:
hi @hcnery08 please share your BIOS Menu screenshot... I'll coordinate with ODM CO.  Thanks!

kmauleon, 2025-02-24:
upon checking you already have the latest MEFW.  Have you updated the BIOS/EC as well?
Please try to execute "fwupdmgr reinstall"

hcnery08, 2025-02-25:
[image]

Heres my BIOS menu screen,

hcnery08, 2025-02-25:
I ran "fwupdmgr reinstall" for all firmware, but still get that "Failed to recover ME firmware" on boot.

Audio is working somehow now.

kmauleon, 2025-02-25:
good to know your audio is working now.
for MEFW, can you please try to downgrade and reinstall it again?

kmauleon, 2025-03-13:
hi @hcnery08 any update from your side?  please advise.  thank you very much.

hcnery08, 2025-03-13:
Still getting that message on boot. Did everything you said. I also tryed using the emergency reset button, switching to windows and using the windows recovery tools from lenovo, i switched back to linux and installed all the combinations of firmware i could, is there a way to directly flash the firmware via usb or somthing im not rlly sure. i also let the battary complely run out and that didnt change anything either. not sure whats going on but its a pain whenever it boots

kmauleon, 2025-03-13:
you may try to reflash BIOS/EC via USB as below...

[readme](https://download.lenovo.com/pccbbs/mobiles/n32ur30w.html)
[image download](https://download.lenovo.com/pccbbs/mobiles/n32ur30w.iso)

hcnery08, 2025-03-13:
That didn't work either: An update is not necessary at this time. Process canceled.

is there a way for force it?

kmauleon, 2025-03-13:
i think via EFI can force it... but have you tried downloading 1 version lower then update to latest again?

kmauleon, 2025-12-05:
please try procedure from this case if it helps... https://github.com/fwupd/firmware-lenovo/issues/360#issuecomment-2395047396
closing for now due to no update... thank you!