fwupd/firmware-lenovo #498: I am experiencing the same even after installing one at time and rebooting . . . . After installing both Debian 12 and now Fedora 41

mylesmorales, 2025-02-25:
This issue is ongoing for my ThinkPad x390. Initially I thought the system update issue was only happening with Debian 12 but now it is also happening with Fedora Workstation 41. 

1. After installing Fedora 41 immediately received a message to update Device Firmware:

ThinkPadT490/x390 System Update: 0.1.75 ---> 0.1.80

ThinkPad x390/T490s Corporation ME update

There is one titled system and I installed this one first. After install via the software manager for Fedora 41 a message appeared to restart. I opted to restart and during the restart the screen went black, and I left this way overnight. In the morning the laptop was off, there was no battery juice left. I tried key suggestions, pressing the power button, there is no hole on the side or back to insert an opened paper clip to restart.  I plugged in the adapter turned on the laptop and all was fine, at least that is what I thought.

Today, the message appeared again to update the system files (the same three) but this time I typed: **fwupdmgr update** and the updates installed one at a time.  

Again a pop-up message appeared to restart/reboot, and now I am sitting here with a black x390 screen, tried a suggested key combination to restart, and it did not work. Did not bring a screwdriver to remove the back panel to detach the battery, so I am waiting for the battery to drain.  Yes, I booted into bios, ran diagnostics and all is well.

kmauleon, 2025-02-25:
please share BIOS Menu screenshot.  and please try to update firmware via Gnome Software one firmware at a time with Secure Boot Disabled.

update BIOS restart system
update EC restart system
update MEFW restart system

mylesmorales, 2025-02-26:
@kmauleon I appreciate the update. One of the three installed and today I tried this one Thinkpad T490/x3990 System update 1.75 ---> 1.80 and a popup suggested a restart. It has been more than hour for the restart. Second I did not turnoff Secure Boot. I am going to wait a little longer before I power off the laptop. Then I will follow your suggestions and update.

Thanks

kmauleon, 2025-02-27:
@mylesmorales you may also want to try to update BIOS/EC via USB

mylesmorales, 2025-02-27:
@kmauleon Yes I agree. The next step is to use the dd command to write the iso to a usb. The restart is still active and in a moment I am going to power the x390 down, wait a few, turn the laptop back on and wait for messages. If the message is: warning no updates to process. exiting in 10 seconds then I am to try the usb method. Will update.

mylesmorales, 2025-02-27:
@kmauleon created the usb bootable iso. Booted with the usb, followed the screen suggestions, received a message the system updated and select restart. The restart is still ongoing. I cannot power off the laptop and have to wait for the power to drain. Will update soon.,

kmauleon, 2025-02-27:
thank you @mylesmorales will wait for your update...

kmauleon, 2025-03-13:
hi @mylesmorales any update from your side?  please advise.  thank you very much.

mylesmorales, 2025-03-13:
Thanks for the follow-up. A couple of days ago it was suggested by
Canonical to submit a bug report that I will share  you, and Lenovo as soon
as I receive a reply from the Canonical Bug team. I am also sending an
email to Lenovo Marketing team about conerns of this issue happening in the
future. Although I power down the laptop I still have to press the power
button to turn off the laptop. This is a major issue.

Peace and Struggle🤟🏿

kmauleon, 2025-03-13:
thank you for the updates @mylesmorales. However, as mentioned earlier, this particular system is already EOS.  if you happen to encounter the issue in a different platform, please open an issue against that one since we have different ODM Owners per platform.
can we agree to close this ticket for now... hoping for your understanding.  thank you very much.

mylesmorales, 2025-03-13:
Yes the ticket can be closed.

kmauleon, 2025-03-13:
thanks again @mylesmorales !

closed 2025-03-13 by kmauleon

mylesmorales, 2025-03-13:
Always welcome