fwupd/firmware-lenovo #501: Firmware upgrade fails with "failed to run update on reboot"

julienbenjamin, 2025-03-02:
**Describe the bug**
Upgrading Embedded Controller firmware from 0.1.15 to 0.1.22 does not work.

**Steps to Reproduce**
- run `fwupdmgr upgrade`.
- say yes to "perform operation".
- after reboot, notice the "failed to run update on reboot" in the "Update Error" field from `fwupdmgr get-history`

**Expected behavior**
I would expected the upgrade to suceed and that the Embedded Controller firmware to be updated to 0.1.22 version.

**fwupd version information**
```
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.0-31-amd64
```

```shell
fwupdtool get-report-metadata
Loading?                 [*****************                      ]19:16:27.748 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]
Waiting?                 [*************************************  ]
BootTime:               1740855695
CompileVersion(com.hughsie.libjcat):0.1.9
CompileVersion(com.hughsie.libxmlb):0.3.10
CompileVersion(org.freedesktop.fwupd):1.8.12
CompileVersion(org.freedesktop.gusb):0.3.10
CpuArchitecture:        x86_64
CpuModel:               Intel Core? i7-7500U CPU @ 2.70GHz
DistroId:               debian
DistroName:             Debian GNU/Linux
DistroPrettyName:       Debian GNU/Linux 12 (bookworm)
DistroVersion:          12
FwupdSupported:         True
HostBaseboardManufacturer:LENOVO
HostBaseboardProduct:   20HQS10P00
HostBiosMajorRelease:   01
HostBiosMinorRelease:   3e
HostBiosVendor:         LENOVO
HostBiosVersion:        N1MET77W (1.62 )
HostEnclosureKind:      a
HostFamily:             ThinkPad X1 Carbon 5th
HostFirmwareMajorRelease:01
HostFirmwareMinorRelease:0f
HostProduct:            20HQS10P00
HostSku:                LENOVO_MT_20HQ_BU_Think_FM_ThinkPad X1 Carbon 5th
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.1.0-31-amd64
RuntimeVersion(com.dell.libsmbios):2.4
RuntimeVersion(org.freedesktop.fwupd):1.8.12
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.3.10
RuntimeVersion(org.kernel):6.1.0-31-amd64
DeviceId:               a45df35ac0e948ee180fe216a5f703f32dda163f
  pre:
      CapsuleApplyMethod:nvram
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x0
DeviceId:               349bb341230b1a86e5effe7dfe4337e1590227bd
[... 156 lines cut ...]
?                                                                              ?
? 20HQS10P00 must remain plugged into a power source for the duration of the   ?
? update to avoid damage.                                                      ?
????????????????????????????????????????????????????????????????????????????????
Perform operation? [Y|n]: 
Waiting?                 [***************************************] Less than one minute remaining?
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Successfully uploaded 1 report
Devices with the latest available firmware version:
 ? Intel Management Engine
 ? MZVLW512HMJP-000L7
 ? System Firmware
An update requires a reboot to complete. Restart now? [y|N]:
```
(Here, the machine reboots.)
- `fwupd-efi version 1.4` is displayed on screen after POST
- `Reset system` is then display below
(Another reboot happens.)
- the logo `Lenovo` appears on screen
- `fwupd-efi version 1.4` is displayed after POST
- `'WARNING: No updates in process, exiting in 10 seconds` is then displayed below
(And the machine reboots to the usual bootloader).

Hope that helps.

hughsie, 2025-03-02:
> fwupd-efi version 1.4 is displayed after POST

This means we set up the update correctly.

> WARNING: No updates in process, exiting in 10 seconds

I *think* this is from the lenovo updater binary; it's been downloaded thousands of times with 96% success rate so I'm not entirely sure why it doesn't work for your specific machine. @mrhpearson any ideas? the firmware is https://fwupd.org/lvfs/firmware/7992 I think.

mrhpearson, 2025-03-03:
Afraid I don't know off the top of my head. It's possible it's because it's quite a big jump from 1.15 to 1.22.  1.15 is really old (mind you 1.22 is no spring chicken)

Maybe  try going to 1.19 first?
https://fwupd.org/downloads/e5ab3b757b5ebe9542bc795beabc2a1e99394521-Lenovo-ThinkPad-X1Carbon5th-ECFirmware-1.19.cab

I'd honestly be a bit cautious at this point - I believe the X1 Carbon 5 is out of support (it's old enough that I don't have one). It will be really hard for me to get answers for it from the FW team. If you don't need the fix referenced I'd be inclined to leave it alone, but up to you.

Mark

Jackaed, 2025-03-20:
I'm experiencing the same category of issue with a ThinkPad T14s Gen 3 AMD - much more recent machine. Should I open a new issue @mrhpearson?

mrhpearson, 2025-03-20:
Yes please - new ticket with details on your current FW, and version of fwupd please.
Mark

julienbenjamin, 2025-03-29:
This might a stupid question but, how can we rule out hardware issues?

ptr727, 2025-07-06:
Similar issue on T14 Gen1 with ME firmware, fingerprint and BIOS firmware applied without issue.

`Update Error:       failed to run update on reboot: expected 225.75.2420 and got 225.70.2228`

```
$ sudo fwupdmgr get-history
LENOVO 20S1S3VT16
│
├─Prometheus:
│ │   Device ID:          71fc4ca26c1cd81e2d82496b0d428c221c88fd47
│ │   Previous version:   10.01.3273255
│ │   Install Duration:   16 seconds
│ │   Update State:       Success
│ │   Last modified:      2025-07-06
│ │   GUID:               8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │ 
│ └─Prometheus Fingerprint Reader Update:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10694
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425.5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│         Tested:         2023-05-05
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3273255
│         Version[fwupd]: 1.7.9
│         Tested:         2023-02-28
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3273255
│         Version[fwupd]: 1.7.9
│         Tested:         2022-07-07
│         Distribution:   ubuntu 22.04
│         Old version:    10.01.3158804
│         Version[fwupd]: 1.7.5
│       Vendor:           Synaptics
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│       Checksum:         9a3173547a0f44cd369c03b561018ce1d86834d6fef24e1ebb95fa1f8391521a
│     
└─Intel Management Engine:
  │   Device ID:          d9166c223a3152f6deacdf4944cbf5e4df8cc620
  │   Previous version:   225.70.2228
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot: expected 225.75.2420 and got 225.70.2228
  │   Last modified:      2025-07-06
  │   GUID:               b36438d8-9128-49d2-b280-487be02d948b
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Corporate ME Update:
        New version:      225.75.2420
        Remote ID:        lvfs
        Release ID:       111432
        Summary:          Lenovo ThinkPad T14 Gen 1/ P14s Gen 1/ T15 Gen 1/ P15s Gen 1/ T14 Gen 1 Healthcare Edition Corporate ME Firmware
        License:          Proprietary
        Size:             11.0 MB
        Created:          2025-04-09
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N2XRM16W
        Vendor:           Lenovo
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        • 1 Intel Platform Update
        
        Version 14.1.75.2420 (LVFS: 225.75.2420)
        
        Problem Fixes
        
        • Intel CSME IPU 2025.1:
        • Mitigated the following security vulnerabilities under issues.
        Issues:           CVE-2024-38307
                          CVE-2024-30211
                          CVE-2024-26021
        Checksum:         817dba79c293ac635b5370faa434eac36636af88276289838f24194b89d5d610
```

kmauleon, 2025-07-06:
hi all... i am not sure if this will help but please do the following...
     disable secure boot
     update firmware one at a time:  BIOS  then EC first

kmauleon, 2025-08-26:
how about reflashing BIOS/EC via USB?

closed 2025-09-22 by kmauleon