fwupd/firmware-lenovo #502: Thinkpad T14 Gen 5 Embedded Controller 1.0.8 breaks WiFi & Bluetooth

denzilferreira, 2025-03-19:
After updating today to 1.0.8, I was left with a laptop without WiFi and Bluetooth. Not sure what is wrong with this version of the Embedded Controller, but it's clearly broken with Linux.

I was able to downgrade back to 1.0.5 by downloading the .cab and using --allow-older on the command line with:
`sudo fwupdmgr install 6c6d552f8af619f597b019258817cb1af95389fb1d3195ab5f14bf060f062b07-Lenovo-ThinkPad-T14-P14s-AMD-Gen5-EmbeddedControllerFirmware-1.05.cab --allow-older`

Post here in case it helps someone.

mrhpearson, 2025-03-19:
Thanks for the report - I've created internal ticket LO-3654 to get input from the FW team.
Mark

mrhpearson, 2025-03-19:
I reproduced the issue on my P14s G5 AMD. I've demoted the FW to embargo :(

hughsie, 2025-03-19:
Thanks Mark, much appreciated.

tury325re, 2025-03-19:
@mrhpearson @hughsie I have the same exact issue after updating my Embedded Controller to 1.0.8 on my Thinkpad T14 Gen 5 as well. However, I noticed on the Lenovo driver support website, the latest version for the ECP is ECP ID 1.06 (R2LHT**23**W). Upon going into my BIOS, it shows the embedded controller version is 1.08 (R2LHT**25**W). Does fwupd get access to newer ECP firmware versions before it's officially released on the lenovo website? Because ECP 1.08 (R2LHT25W) is nowhere to be found on Lenovo's official driver website. Also, how do I go about fixing this issue? Thank you for your help!

noahlvb, 2025-03-19:
I have encountered the same issue. @denzilferreira were can I download this .cab file?

mrhpearson, 2025-03-19:
wget https://fwupd.org/downloads/6c6d552f8af619f597b019258817cb1af95389fb1d3195ab5f14bf060f062b07-Lenovo-ThinkPad-T14-P14s-AMD-Gen5-EmbeddedControllerFirmware-1.05.cab

Then the steps from above (I had to use --force on my system FWIW).

I'm not sure on 1.06 vs 1.08 I'm afraid - hopefully will get some answers from the FW team in Asia this evening
Mark

noahlvb, 2025-03-19:
I have encountered the same issues. If I understand correctly the UEFI and embedded controller run separate firmware with separate versions. My UEFI is at 0.1.11 and my embedded controller at 0.1.8. Were can I find the correct .cab file for my laptop with these versions or is it realistic to wait for a fix?

mrhpearson, 2025-03-19:
I think you answered your question - but the R2LHT link above is the right one (the other one is for the Intel platform).

The issue is in the EC FW, not the BIOS (aka UEFI in your post). You just need to downgrade the EC to 1.05 to recover. 

I would do this rather than wait as even once the FW team have determined the cause (I think it's mucking up the PCIe bus) it will take time to deliver the fix. There's a release process, including testing, that has to be done.

noahlvb, 2025-03-20:
Downgrading to 1.05 worked perfectly on my system. I also discovered on 1.08 that my laptop could not boot of usb drives when plugged in to the onboard usb-a ports, the usb drives would not shown on the boot selection screen (f12). Using a usb-c dongle did worked. Booting over the onboard usb-a works on 1.05 again. Does this deserve a separate issue?

mrhpearson, 2025-03-20:
Quick update: Issue is related to the fact that BIOS needs to be updated to 32W (1.13) for the new EC to work. That's not released yet, which is what triggered this problem (the EC metadata really needs a check adding that the BIOS is at the right level too - I've requested that to be added before it's released again).

Please stick with the 1.05 EC until this is fixed, and apologies for the muck up

@noahlvb - I did a quick test on mine, and (new BIOS and EC) is working fine for booting from USB-A connected stick. Let's wait for that to roll out first, and if you still see the problem we can do a new ticket for that separately. For now I'm going to assume it's related.

Mark

denzilferreira, 2025-03-21:
Thanks everyone! Love to see all the hard work that goes into making Linux a first class citizen on our laptops! You peeps rock! Cheers from 🇫🇮

03x6, 2025-06-04:
just to bump this, according to lenovos site 1.14 released
https://download.lenovo.com/pccbbs/mobiles/r2luj33w.html

does this mean im safe to update now or am i misinterpreting what @mrhpearson said?

mrhpearson, 2025-06-09:
I'd hold off - I can see both BIOS 1.14 and 1.15 on LVFS (under embargo). They should release to stable once the Linux testing has been done so unless you're desperate for the update I'd wait until their testing is done.

I've sent a nudge to the FW team to check on the exact status.

denzilferreira, 2025-06-25:
Looks like we got pushed BIOS 1.14 now but not the embedded controller so no WiFi. Bluetooth is working...

> [   32.836338] ath11k_pci 0000:02:00.0: BAR 0 [mem 0x90600000-0x907fffff 64bit]: assigned
[   32.836354] ath11k_pci 0000:02:00.0: enabling device (0000 -> 0002)
[   32.836430] ath11k_pci 0000:02:00.0: MSI vectors: 1
[   32.836437] ath11k_pci 0000:02:00.0: wcn6855 hw2.1
[   54.255944] ath11k_pci 0000:02:00.0: failed to power up mhi: -110
[   54.255955] ath11k_pci 0000:02:00.0: failed to start mhi: -110
[   54.255960] ath11k_pci 0000:02:00.0: failed to power up :-110
[   54.266254] ath11k_pci 0000:02:00.0: failed to create soc core: -110
[   54.266260] ath11k_pci 0000:02:00.0: failed to init core: -110
[   54.382352] ath11k_pci 0000:02:00.0: probe with driver ath11k_pci failed with error -110

03x6, 2025-06-25:
fyi there is currently a regression in the linux kernel regarding the WCN7850/X Network controller as someone seemingly pushed a faulty update. Im personally on kernel 6.13.12 which still works, the regression was introduced in the 6.14.x range, iirc .5 or .6

sofar kernel 6.15.3 didnt include the fixes but they will prolly be in 6.16

03x6, 2025-06-25:
here for a src 
https://bugzilla.redhat.com/show_bug.cgi?id=2365882

03x6, 2025-06-25:
i know this is ath12k for the wifi 7 NCM825 but i switched to kernel 6.15.3 real quick to show you what i mean

```
[   31.804895] ath12k_pci 0000:02:00.0: failed to receive wmi unified ready event: -110
[   31.806353] ath12k_pci 0000:02:00.0: failed to start core: -110
[   31.813123] failed to send QMI message
[   31.813167] ath12k_pci 0000:02:00.0: qmi failed to send mode request, mode: 4, err = -5
[   31.813170] ath12k_pci 0000:02:00.0: qmi failed to send wlan mode off%    
```

it is acting the same way with you tho, namely bluetooth working but wifi being broken

yours may be unrelated as its a different chip, atleast i knew what i got into when buying that chip, citing some linux mailing list entry that this chip (the NCM825) is the only reason they cannot get the linux certification thus far. 

For reference i got the `21MDS3D500` T14 Gen 5 and i still have not updated the bios to 1.14 so this regression may be the linux kernel one, not the bios'es fault. id try switching to either the lts kernel or incase you may be on nixos setting boot.kernelPackages = pkgs.linuxPackages_6_13 for a slightly newer one and see if that fixes it for now.

denzilferreira, 2025-06-25:
That was not the case for me. I was on 6.15.3 Fedora 42 and I had WiFi. Only lost WiFi when installing the latest BIOS 1.14 @03x6

03x6, 2025-06-25:
just checked before i do the update, the linux-testing nixpackage 6.16-rc2 fixes it for the ncm825 too, no longer crashes upon startup 

```
[   26.171865] wlp2s0: authenticate with AB:CD:EF:01:23:45 (local address=54:32:10:FE:DC:BA)
[   26.171876] wlp2s0: send auth to AB:CD:EF:01:23:45 (try 1/3)
[   26.176912] wlp2s0: authenticated
[   26.177508] wlp2s0: associate with AB:CD:EF:01:23:45 (try 1/3)
[   26.181239] wlp2s0: RX AssocResp from AB:CD:EF:01:23:45 (capab=0x1011 status=0 aid=28)
[   26.202204] wlp2s0: associated
[   29.263583] wlp2s0: Limiting TX power to 23 (23 - 0) dBm as advertised by AB:CD:EF:01:23:45
```

ChiWei-Chen, 2025-07-01:
Hi,
It looks like the issue has been resolved.
If there are no further concerns related to this task, I’ll go ahead and close it.
Please let me know if anything else comes up.
Thanks!

pgray, 2025-11-06:
downgraded to `1.0.05` from a broken `1.0.8` and fwupd doesn't seem to see any available upgrade versions so I guess I'll stick to this for now
```
Host: 21MCCTO1WW (ThinkPad T14 Gen 5)
CPU: AMD Ryzen 7 PRO 8840U
```