fwupd/firmware-lenovo #505: (Thinkpad T14s Gen 3 AMD) System firmware upgrade fails with "Could not find update"

Jackaed, 2025-03-31:
When upgrading the system firmware on my laptop, it succesfully boots to fwupd-efi. However, it then flashes some text for about 200ms, then my system reboots into the normal OS. I took a slow-mo video of the reboot, here's what the fwupd-efi text read:

```
fwupd-efi version 1.6
WARNING: Could not get variable "LastMemoryMapInformation", Not Found
WARNING: Could not populate update info for "LastMemoryMapInformation"
WARNING: Could not find updates: Not Found
```

Device information:

```
LENOVO 21CQCTO1WW
│
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404107
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
│ │
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.28.00.71
│ │     Bootloader Version: 00.28.00.71
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUID:             9844da3e-1df2-52fe-9413-d4378af6221e ← PCI\VEN_1022&DEV_1649
│ │     Device Flags:     • Internal device
│ │                       • Can tag for emulation
│ │
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  69.63.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             79307ae6-a2ea-52e1-bf56-6abbaf3547ad ← /sys/devices/platform/AMDI0007:00
│       Device Flags:     • Internal device
│                         • Can tag for emulation
│
├─ATNA40YK08-0:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     Vendor:             PNP:SDC
│     GUID:               c9b62dde-ef81-599c-84d2-ff42aa9543de ← DRM\VEN_SDC&DEV_417A
│     Device Flags:       • Internal device
│                         • Can tag for emulation
│
├─CT1000P5PSSD8:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    P7CR403
│     Vendor:             Micron/Crucial Technology (PCI:0xC0A9)
│     Serial Number:      22343AF0DC77
│     GUIDs:              5b5825f0-2a84-54df-b60b-8ee09786de93 ← NVME\VEN_C0A9&DEV_5407
│                         cd36ee17-e270-50f3-9468-7bb0b99b5fc5 ← NVME\VEN_C0A9&DEV_5407&SUBSYS_C0A90100
│                         f99dba73-0ec7-5fb0-b34f-3566d7e85bc2 ← CT1000P5PSSD8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Can tag for emulation
│
├─Embedded Controller:
│     Device ID:          86099bd0eda2dcdb566a62c828f02ba94fbba000
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.32
│     Minimum Version:    0.1.32
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
[... 201 lines cut ...]
│
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Platform Key
      GUID:               be27c8c1-68ba-5fbf-adf4-09401c1c8928 ← UEFI\CRT_C86E6979D66B9FEBEA58C58BE04319F049302D8A
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation
```

This is with fwupd from NixOS's nixpkgs unstable. `fwupdmgr --version`:

```
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.7
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.12.20
runtime   org.freedesktop.fwupd         2.0.7
```

Note that I have successfully used fwupd to do firmware updates in the past without issues. Let me know what other information I can provide.

hughsie, 2025-04-22:
> my system no longer detects "system firmware" as a device

What does: `sudo fwupdtool get-devices --plugins uefi_capsule -vv` print please?

Jackaed, 2025-04-22:
About 6000 lines.

https://pastebin.com/uhJV8aU0

hughsie, 2025-04-22:
Eigh? What does `tree /sys/firmware/efi/esrt` say? Is there more than one device there?!

Jackaed, 2025-04-22:
```
/sys/firmware/efi/esrt
├── entries
│   └── entry0
│       ├── capsule_flags
│       ├── fw_class
│       ├── fw_type
│       ├── fw_version
│       ├── last_attempt_status
│       ├── last_attempt_version
│       └── lowest_supported_fw_version
├── fw_resource_count
├── fw_resource_count_max
└── fw_resource_version
```

hughsie, 2025-04-23:
@mrhpearson help! Have you seen this before?

mrhpearson, 2025-04-23:
I haven't I'm afraid
Created internal ticket LO-3707 and will ask for input from the FW team.

mrhpearson, 2025-04-25:
Bit of a confusing message - but FW team said to retry the update now. The FW got pushed to the stable channel so I think they fixed something?
Let me know if you still see the issue.

Jackaed, 2025-04-25:
> Bit of a confusing message - but FW team said to retry the update now. The FW got pushed to the stable channel so I think they fixed something? Let me know if you still see the issue.

The issue is I can't even see the device anymore - possibly a fwupd issue rather than a lenovo issue? I'll try and go to an older fwupd version and see if that helps.

Jackaed, 2025-04-25:
Older fwupd versions don't help - something has gone wrong here. Either an update was botched, or something else has happened. Either way, I'm very confused.

mrhpearson, 2025-04-25:
The update got moved to testing for a bit, wonder if that is why?
Try 'fwupdmgr refresh' ?

hughsie, 2025-04-25:
@mrhpearson it's a bit weird; the normal devices have disappeared from the ESRT and there only seems to be 54cb734c-975c-4a74-9556-57ae6d9a2229 now -- perhaps some kind of "recovery" device?

Jackaed, 2025-04-25:
@hughsie could it be worth me updating via the USB ISO that Lenovo provide? Might bring everything back to normal if it's just a failed update causing problems.

mrhpearson, 2025-04-28:
Apologies - I hadn't understood the issue properly.
Yeah - only one ESRT entry is strange, not seen that before. Asking for clarification from the FW team.

Jackaed, 2025-05-04:
> Apologies - I hadn't understood the issue properly. Yeah - only one ESRT entry is strange, not seen that before. Asking for clarification from the FW team.

Did you hear anything back? Should I try the ISO upgrade?

jbiel, 2025-05-11:
I'm having the same issue as OP's original comment on an X1 Carbon 12th gen. If you'd rather I create a discrete issue I will do so, but I figured this issue was similar enough.

Fwupdmgr prompts for two updates to be installed, and the "Not found" message is shown on boot. I'm also on Nixos and I've uploaded a report to fwupd.org (`"MachineId" : "031264adda02fc4f0cfefb1dfb93f4a283b0ddf3904f8cabb3a1bea9c58972ae"`.)

Note that I made a "manual" boot entry for this as it seemed that it wasn't installing updates. When I use this the system hangs on the error so it's easier to see what's going on.

```
# /boot/loader/entries/firmware-update.conf 
title Firmware Update
efi /EFI/nixos/fwupdx64.efi
```

```
$ fwupdmgr --version
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.8
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.5
runtime   org.freedesktop.fwupd         2.0.8
```

Jackaed, 2025-05-16:
> It's golden week in Asia this week - I won't hear back from anybody until the second half of the week I'm afraid.
> 
> I think trying the ISO update is worth a go.
> 
> Mark

I ran the ISO update successfully, but still don't see a "System Firmware" device on my machine when using `fwupdmgr get-upgrades` or `fwupdmgr get-devices`. I have some exams right now but after they're done I'll see if I can run fwupd from a Fedora LiveCD or something and see if I can see it on there.

kkflt, 2025-05-22:
I am also facing this issue on my T14s gen3 AMD

ChiWei-Chen, 2025-05-27:
@Jackaed , Could you let me know which version of the old fwupd you tested?
Also, when you get a chance, could you try it with version 1.9.x and let me know how it goes? Thanks!

Jackaed, 2025-05-27:
> [@Jackaed](https://github.com/Jackaed) , Could you let me know which version of the old fwupd you tested? Also, when you get a chance, could you try it with version 1.9.x and let me know how it goes? Thanks!

Can't remember what I tried originally, just tried with 1.9.25. Output of `sudo fwupdmgr get-devices`:

```
LENOVO 21CQCTO1WW
│
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404107
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        fb2f6a4f66c638829126eef33d281666a63c1e64
│ │     Summary:          Rembrandt Generic VBIOS
│ │     Current version:  032.017.000.055.000000
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUID:             6d1757cb-7371-5d9a-9b7a-4be4b21ef0b9 ← AMD\113-REMBRANDT-X37
│ │     Device Flags:     • Internal device
│ │   
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.28.00.72
│ │     Bootloader Version:00.28.00.72
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9844da3e-1df2-52fe-9413-d4378af6221e ← PCI\VEN_1022&DEV_1649
│ │                       fbbfd456-73f4-5488-a520-f5e4aaa5b9e1 ← PCI\VEN_1022&DEV_1649&SUBSYS_17AA50B4
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  69.65.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             79307ae6-a2ea-52e1-bf56-6abbaf3547ad ← /sys/devices/platform/AMDI0007:00
│       Device Flags:     • Internal device
│     
├─CT1000P5PSSD8:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    P7CR403
│     Vendor:             Micron/Crucial Technology (NVME:0xC0A9)
│     Serial Number:      22343AF0DC77
│     GUIDs:              5b5825f0-2a84-54df-b60b-8ee09786de93 ← NVME\VEN_C0A9&DEV_5407
│                         cd36ee17-e270-50f3-9468-7bb0b99b5fc5 ← NVME\VEN_C0A9&DEV_5407&SUBSYS_C0A90100
│                         f99dba73-0ec7-5fb0-b34f-3566d7e85bc2 ← CT1000P5PSSD8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│     Current version:    10.22
│     Vendor:              (USB:0x174F)
│     Serial Number:      0001
│     GUID:               0cf3aef8-c29e-5cb8-997c-d498ed52b666 ← USB\VID_174F&PID_1812
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      266085307572408
[... 42 lines cut ...]
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    20241101
      Minimum Version:    20241101
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUIDs:              020aa71b-0e5b-5d35-99bc-76105fae9c6e ← UEFI\CRT_03503049E14ABCF967B56618513E60DF9EC5C7840AD046ACF62D75D784BC7F6E&ARCH_X64
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
```

Still no "System Firmware" device.

Jackaed, 2025-05-27:
The only folder in `/sys/firmware/efi/esrt/entries/` is `entry0`. It's `fw_class` output is `54cb734c-975c-4a74-9556-57ae6d9a2229`.

Jackaed, 2025-06-07:
@mrhpearson any luck in recreating this issue? Supposedly my original issue was due to using gnu-efi 4.0.0 with fwupd-efi 1.6 (see https://github.com/fwupd/fwupd/issues/8714). I've since updated to fwupd-efi 1.7, but that hasn't given me my system firmware entry back - I suppose this implies I've managed to break something? Is there any way to reset the whole firmware back to factory settings for me to upgrade from there?

mrhpearson, 2025-06-09:
Hi - afraid I don't have any break-thru updates.

The FW engineer looking at this says he reproduced it once, but updated to the latest fwupdmgr (he's using Ubuntu so installed the snap) and it worked fine after that. It doesn't make any sense or match what you're experiencing so we've asked for more clarification.

I saw kkflt is on Nix. Jackage - what distro are you on?

closed 2025-06-12 by mrhpearson

hughsie, 2025-06-12:
@mrhpearson I thought we had something like that in plugins/uefi-esrt/fu-uefi-esrt-plugin.c already?

mrhpearson, 2025-06-16:
> [@mrhpearson](https://github.com/mrhpearson) I thought we had something like that in plugins/uefi-esrt/fu-uefi-esrt-plugin.c already?

Yeah - you're right. I'll see if I can go and figure out why it wasn't triggering in this case.

mrhpearson, 2025-06-16:
> [@hughsie](https://github.com/hughsie) [@mrhpearson](https://github.com/mrhpearson) The setting was indeed off, unsure how that happened but I'm glad it was an easy fix! Sorry for the troubles, you guys do great work. Device appears as normal.

Awesome - glad we've solved it.
Make me wonder if Nix is doing something strange - peculiar to have two different reports in such a close period of time.

mariolpantunes, 2025-10-15:
Hi, have the same error, already checked, and my BIOS has the option "Windows UEFI Firmware Update" enabled.
However, whenever I try to update, I get the same error.
Using Slackware64-current on a lenovo T41s AMD gen 6.

mariolpantunes, 2025-11-27:
Any help with this?

Here is a screenshot (of a video, from my computer booting) of the error.
And another screenshot with the BIOS options are enabled.

[image]

[image]