fwupd/firmware-lenovo #506: Lenovo thinkpad t14 gen 2 not charging battery stuck at 0% ubuntu 22.04

mcandeo, 2025-04-06:
Hello,

I'm having trouble with my battery not charging on thinkpad t14 gen 2.

I'm attaching the history for fwupdmgr. 

[history.txt](https://github.com/user-attachments/files/19622262/history.txt)

Could you help me with that?

hughsie, 2025-04-07:
@mcandeo I'm not sure what `Update State:       Transient failure` means -- have you tried powering the device off (not just restart) and then booting back up? Did the battery firmware update complete successfully?

mcandeo, 2025-04-07:
I did, but no results

hughsie, 2025-04-07:
Can you `fwupdmgr downgrade` the battery firmware? @mrhpearson any ideas?

mrhpearson, 2025-04-07:
From the logs it looked like the battery update didn't happen as the system thinks it wasn't plugged in. Does that make sense? Have you tried triggering that again with power connected?
I don't think downgrading is an option - it doesn't look like it's updated yet.

If your system isn't charging any more, I think that's likely a HW related issue (dead battery?) and you'll have to call Lenovo support. I'm assuming you haven't changed your battery settings (charge threshold's etc). 
Did anything in particular trigger this, or did you start looking at updates because it wasn't charging?

Mark