fwupd/firmware-lenovo #509: ThinkPad L13 Yoga Gen 2 - Not always waking from suspend

rgrove-rsu20, 2025-05-05:
Forgive me if this is the wrong place to report this (Lenovo forums got nowhere), but it seems that since at least firmware rev 1.27, this model sometimes does not wake properly from suspend (s2idle/s0ix). The status indicator light on the power button indicates the device is still suspended, but it will not work via lid switch, fn key, or power button. Our devices are currently on Debian stable/12.10, but I've seen this behavior on other distributions, such as Ubuntu, Fedora, and even Chrome OS Flex, which leads me to believe this is firmware related. Connecting the power adapter and hitting the power button will wake the device FWIW.

hughsie, 2025-05-06:
It sounds like a bug, but it's not really the right place I'm afraid. @mrhpearson might be able to help if you can link the Lenovo forum post.

rgrove-rsu20, 2025-05-06:
Thanks, my google/duckduckgo search failed me for a place other than the forums. The original forum post is [here](https://forums.lenovo.com/t5/ThinkPad-L-R-and-SL-series-Laptops/ThinkPad-L13-Yoga-Gen-2-Not-always-waking-from-suspend-on-firmware-1-32/m-p/5330693).

mrhpearson, 2025-05-07:
Hi @rgrove-rsu20 - I'll pick up the ticket in the forum - it wasn't in the Linux channels so hadn't seen it :)

Usually Linux related issues get moved to the Linux channels (any of the ones in forums.lenovo.com/linux where my team will see them). Doesn't seem to have happened in this case.

rgrove-rsu20, 2025-05-07:
Thank you. I'll close here and continue on that forum.