fwupd/firmware-lenovo #510: Lenovo P360 Tiny Workstation shows updates drivers/firmware but never updates them

ftmiranda, 2025-05-07:
Using a Lenovo P360 Tiny workstation and Ubuntu 24.04 LTS
When I run $ sudo fwupdmgr update command it shows updates and it does show the process of updating, but after a recommended reboot, no updates were applied.

Steps to Reproduce
Use the Ubuntu 24.04 LTS command $ sudo fwupdmgr update to update the outdated components:
$ sudo fwupdmgr update
Devices with no available firmware updates:
• SSD 990 EVO Plus 2TB
• UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.22 to 0.1.24? ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues: ║
║ ║
║ • Fix some EC Setup issues. ║
║ ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the ║
║ update to avoid damage. ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Waiting… []
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 10.0.15.1662 to 10.1.30.2330? ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues: ║
║ ║
║ • Enhancement to address security vulnerability. ║
║ ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the ║
║ update to avoid damage. ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating Intel Management Engine… ]
Waiting… []
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.37 to 0.1.50? ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues: ║
║ ║
║ • Fix some BIOS Setup issues. ║
║ ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the ║
║ update to avoid damage. ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating System Firmware…[ - ]
Waiting… [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
• UEFI dbx

Then reboot

Expected behavior
After the reboot all the updated devices are running the updated version, but instead it remains on the old versions

fwupd version information
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.29
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.29
runtime   org.kernel                    6.11.0-25-generic

Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: M4GKT25A
BiosMajorRelease: 1
BiosMinorRelease: 37
FirmwareMajorRelease: 01
FirmwareMinorRelease: 16
Manufacturer: LENOVO
Family: ThinkStation P360 Tiny
ProductName: 30FBS39B01
ProductSku: LENOVO_MT_30FB_BU_Think_FM_ThinkStation P360 Tiny
EnclosureKind: 23
BaseboardManufacturer: LENOVO
BaseboardProduct: 330E
Hardware IDs
------------
{0250582b-f857-55bb-b0d9-e5be68610f49}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{9a375b0c-1701-594d-bfa3-5ae58bc94334}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{ad527b69-1af3-5bb0-a100-03671ef159e2}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{b3097405-d7c7-5f54-ba3b-e33f6d479a00}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{36488102-7f7a-5f08-b3b0-a6f190f3d84c}   <- Manufacturer + Family + ProductName + ProductSku
{4504b509-4d4f-5221-8a87-e21262eee395}   <- Manufacturer + Family + ProductName
{75b4ed92-b339-5ed8-89fc-8310dc0d833c}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{c24763d8-8fdc-5c79-978b-2cec383ed9e8}   <- Manufacturer + ProductSku
{7cac3995-e3a7-5965-a488-f0558498fc73}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{4b4306a1-f8da-5055-9824-f17917d52e01}   <- Manufacturer + ProductName
{b3a93ac3-1722-5397-8a19-9e89d344b222}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{8bd04657-1fb8-598a-a2e2-903a0d6e08b6}   <- Manufacturer + Family
{78cfbbd7-c5d4-5d24-9ad2-df6114c7d37d}   <- Manufacturer + EnclosureKind
{55db0253-c51a-527d-a7ec-6a8fd1bd07eb}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer
Extra Hardware IDs
------------------
{c8b191f4-c457-5c18-b66f-77f11efe6a7b}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{7c495e92-1d95-5d11-9c74-70d3d0d3dcff}   <- Manufacturer + Family + ProductName + BiosVendor
{99431f53-09a1-5869-be79-65e2fa3f341d}   <- Manufacturer + BiosVendor

ftmiranda, 2025-05-07:
Tagging @mrhpearson  here as well :-)

mrhpearson, 2025-05-07:
Oh - this is different to what was reported in ticket 8696.
Your DBX is updated fine - it's your other updates that are not triggering. Different problem - good thing we did a new ticket :)

Can you try just doing one of the updates, and check if anything is displayed on reboot - it might be worth videoing it as the messages can fly by pretty fast. The fact that all three fail makes me wonder if the FW updater isn't being triggered on reboot.

Mark

ftmiranda, 2025-05-07:
@mrhpearson good catch !
well I tried many different ways - even manually download each update and try to update each item individually. 
The issue from what I can tell is that, once you complete the " update"  the boot entry is not created.
So if you run the command efibootmgr -v, no entry is created:

BootCurrent: 000A
Timeout: 2 seconds
BootOrder: 000A,000B,000C,0000,0003,0002
Boot0000* Windows Boot Manager	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000000000100000010000000040000007fff0400
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
    data: 57 49 4e 44 4f 57 53 00 01 00 00 00 88 00 00 00 78 00 00 00 42 00 43 00 44 00 4f 00 42 00 4a 00 45 00 43 00 54 00 3d 00 7b 00 39 00 64 00 65 00 61 00 38 00 36 00 32 00 63 00 2d 00 35 00 63 00 64 00 64 00 2d 00 34 00 65 00 37 00 30 00 2d 00 61 00 63 00 63 00 31 00 2d 00 66 00 33 00 32 00 62 00 33 00 34 00 34 00 64 00 34 00 37 00 39 00 35 00 7d 00 00 00 00 00 01 00 00 00 10 00 00 00 04 00 00 00 7f ff 04 00
Boot0002* Generic Usb Device	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot0003* CD/DVD Device	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
      dp: 01 04 14 00 e7 75 e2 99 a0 75 37 4b a2 e6 c5 38 5e 6c 00 cb / 7f ff 04 00
Boot000A* Ubuntu	HD(1,GPT,0c23b8d1-c106-43ba-a3cc-1c46779c13b9,0x800,0x219800)/File(\EFI\UBUNTU\SHIMX64.EFI)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 98 21 00 00 00 00 00 d1 b8 23 0c 06 c1 ba 43 a3 cc 1c 46 77 9c 13 b9 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 55 00 42 00 55 00 4e 00 54 00 55 00 5c 00 53 00 48 00 49 00 4d 00 58 00 36 00 34 00 2e 00 45 00 46 00 49 00 00 00 / 7f ff 04 00
Boot000B* UEFI: PXE IPv4 Realtek USB Ethernet Controller	PciRoot(0x0)/Pci(0x14,0x0)/USB(21,0)/MAC(6c1ff71be551,0)/IPv4(0.0.0.00.0.0.0,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 15 00 / 03 0b 25 00 6c 1f f7 1b e5 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f
Boot000C* UEFI: PXE IPv6 Realtek USB Ethernet Controller	PciRoot(0x0)/Pci(0x14,0x0)/USB(21,0)/MAC(6c1ff71be551,0)/IPv6([::]:<->[::]:,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 15 00 / 03 0b 25 00 6c 1f f7 1b e5 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f

I also tried to use this option as well suggested by a person on the previous ticket:
CapsuleApplyMethod:cod

According to the system firmware, the firmware wants us to use Capsule-on-Disk rather than using fwupd-efi.

You might want to try DisableCapsuleUpdateOnDisk=true -- e.g. see man fwupd.conf -- if that works I can ping the ThinkStation team and ask if they want to change the BIOS setting or add the quirk via a DMI match.

Tried that, rebooted, tried to update, no success..

mrhpearson, 2025-05-07:
Disabling CoD was going to be one of my suggestions, but you're right that if the efi variable isn't created that will stop it working.

In the BIOS menu can you check that you have 'Windows UEFI Firmware Update' enabled in the security menu?

You could try creating a boot entry and see if that works? Just as a test. e.g.:
efibootmgr --create --label "Delete me"
(then delete with 'efibootmgr -B -b <entry ID>')

Mark

ftmiranda, 2025-05-07:
@mrhpearson 
I checked the BIOS and yes  'Windows UEFI Firmware Update'  is enabled

I tried to use that example command:

$ sudo efibootmgr --create --label "Delete me" -v
Could not prepare Boot variable: No such file or directory
error trace:
 creator.c:174 efi_va_generate_file_device_path_from_esp(): could not open device for ESP: No such file or directory
 creator.c:335 efi_generate_file_device_path_from_esp(): could not generate File DP from ESP: No such file or directory
 efi.c:225 make_linux_load_option(): efi_generate_file_device_path_from_esp() = -1 (failed): No such file or directory
 efibootmgr.c:295 make_var(): make_linux_load_option() failed: No such file or directory
 efibootmgr.c:344 make_var(): Could not set variable: No such file or directory

no success (maybe this is why the entries does not get created by the fwupdmgr ?

hughsie, 2025-05-07:
> could not open device for ESP: No such file or directory

Very odd. What does `sudo fwupdtool efiboot-info` and `sudo fwupdtool esp-list` say please?

ftmiranda, 2025-05-07:
@hughsie Thank you

Here it is:

$ sudo fwupdtool esp-list
Loading…                 [************************************** ]
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/ubuntu/grubx64.efi
/boot/efi/EFI/ubuntu/shimx64.efi
/boot/efi/EFI/ubuntu/mmx64.efi
/boot/efi/EFI/ubuntu/BOOTX64.CSV
/boot/efi/EFI/ubuntu/grub.cfg
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/mmx64.efi
/boot/efi/EFI/UpdateCapsule/fwupd-15fae072-6129-4306-bd08-22194b7855ba.cap

the other one I get a command not found: (I did a --help and I could not find the efiboot-info or anything efi related on the options)
$ sudo fwupdtool efiboot-info
Command not found
Use fwupdtool --help for help

ElCoyote27, 2025-05-10:
Same thing on my Lenovo P350 ThinkStation Tiny (Fedora42) which can't update either:

#  fwupdtool esp-list
Loading…                 [************************************** ]
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p4
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/fbia32.efi
/boot/efi/EFI/BOOT/BOOTIA32.EFI
/boot/efi/EFI/fedora/grubia32.efi
/boot/efi/EFI/fedora/shimx64.efi
/boot/efi/EFI/fedora/grub.cfg
/boot/efi/EFI/fedora/gcdia32.efi
/boot/efi/EFI/fedora/grub.cfg.orig
/boot/efi/EFI/fedora/mmx64.efi
/boot/efi/EFI/fedora/shim.efi
/boot/efi/EFI/fedora/BOOTX64.CSV
/boot/efi/EFI/fedora/gcdx64.efi
/boot/efi/EFI/fedora/grubx64.efi
/boot/efi/EFI/fedora/grubenv.rpmsave
/boot/efi/EFI/fedora/grub.stub
/boot/efi/EFI/fedora/mmia32.efi
/boot/efi/EFI/fedora/shimia32.efi
/boot/efi/EFI/fedora/grub.cfg.stub
/boot/efi/EFI/fedora/BOOTIA32.CSV
/boot/efi/EFI/UpdateCapsule/fwupd-e68fd560-549f-4690-a018-ccecb0fda8af.cap
/boot/efi/mach_kernel
/boot/efi/System/Library/CoreServices/SystemVersion.plist
/boot/efi/RECOVERY.CAP

# fwupdtool efiboot-info
Command not found
Use fwupdtool --help for help

ElCoyote27, 2025-05-10:
On my P350 Tiny, it's the same symptom: update is detected, but when I try to apply it (like I do on my other Lenovo machines), nothing happens during the reboot:

```
# fwupdmgr update
Devices with no available firmware updates: 
 • SSD 970 EVO 1TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.64 to 1.65?                                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║ • Fix boot order exclusion under Windows Advanced Startup.                   ║
║ • Fix Lenovo Cloud Boot failure issue.                                       ║
║ • Enhance the compatibility of NVMe SSD.                                     ║
║ • Support Class WMI Lenovo systemDeploymentBootMode.                         ║
║                                                                              ║
║ 30EF0009GE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]: 
```
(the only way I managed to update the BIOS previously was to use a Windows PE env and apply the Lenovo Windows BIOS update from there. This is less than convenient.

```
Applying the BIOS update using a USB stick would result in:
Reading flash .............done
- Convert BIOS Guard files to Normal Rom Bugger... done
- Error message: BIOS ROM file is older than BIOS ROM image or project ID unm[....]
Update process successfully completed.

```
(This wasn't truy, I was trying to update BIOS from 1.63 to 1.64)

hughsie, 2025-05-10:
If `fwupdtool efiboot-info` then the fwupd version is kinda old. Can you retest with 2.0.x please?

ElCoyote27, 2025-05-10:
this is what I have (Fedora 41):

```
# rpm -qa fw\*
fwupd-efi-1.6-1.fc41.x86_64
fwupd-plugin-flashrom-1.9.29-1.fc41.x86_64
fwupd-plugin-uefi-capsule-data-1.9.29-1.fc41.x86_64
fwupd-1.9.29-1.fc41.x86_64
fwupd-plugin-modem-manager-1.9.29-1.fc41.x86_64

```

ElCoyote27, 2025-05-10:
I' m unable to rebuild fwupd-2.0.7 (from Fedora42) on Fedora41 due to too many dependencies:

```
# rpmbuild -ba fwupd.spec 
setting SOURCE_DATE_EPOCH=1742860800
error: Failed build dependencies:
        ModemManager-glib-devel >= 1.10.0 is needed by fwupd-2.0.7-2.fc41.x86_64
        flashrom-devel >= 1.2-2 is needed by fwupd-2.0.7-2.fc41.x86_64
        gi-docgen is needed by fwupd-2.0.7-2.fc41.x86_64
        json-glib-devel >= 1.1.1 is needed by fwupd-2.0.7-2.fc41.x86_64
        libcbor-devel is needed by fwupd-2.0.7-2.fc41.x86_64
        libjcat-devel >= 0.1.0 is needed by fwupd-2.0.7-2.fc41.x86_64
        libmbim-devel is needed by fwupd-2.0.7-2.fc41.x86_64
        libqmi-devel >= 1.22.0 is needed by fwupd-2.0.7-2.fc41.x86_64
        libxmlb-devel >= 0.1.3 is needed by fwupd-2.0.7-2.fc41.x86_64
        passim-devel is needed by fwupd-2.0.7-2.fc41.x86_64
        pkgconfig(bash-completion) is needed by fwupd-2.0.7-2.fc41.x86_64
        polkit-devel >= 0.103 is needed by fwupd-2.0.7-2.fc41.x86_64
        protobuf-c-devel is needed by fwupd-2.0.7-2.fc41.x86_64
        tpm2-tss-devel >= 2.2.3 is needed by fwupd-2.0.7-2.fc41.x86_64
        vala is needed by fwupd-2.0.7-2.fc41.x86_64
```

ElCoyote27, 2025-05-10:
For the record, here is how I successfulky flashed my P350 Tiny machine:
- Boot Hirden Boot CD PE 1.0.8
- run m3jjy41usa.exe (BIOS Update (Operating system version)), obtained from: https://pcsupport.lenovo.com/us/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p350/downloads/
driver-list

K1kc4, 2025-06-16:
Having same problem on 2 P360 Workstations with Fedora41, fwupd 1.9.30 version.

mrhpearson, 2025-06-16:
I got hold of a P360 Tiny and was able to reproduce the issue.
The problem isn't with the EFI variables - I am able to create one fine. 

I see the capsules created on the EFI filesystem - but the update doesn't trigger. I need to review with the FW team on that

If I disable CoD the FW update efi isn't included which is odd. Still checking if I've messed something up there (@hughsie - let me know if there's anything in particular I should be looking at)

ftmiranda, 2025-07-05:
> EFI filesystem - but the update doesn't trigger. I need to review with the FW team on that
First of all thank you for doing this!
@mrhpearson any progress on this front? Any new from the firmware team?
thank you !

stefanlindstroem-afk, 2025-09-04:
I had same issue with Lenovo ThinkCentre-M70q Gen 4. I could never get the update to trigger - hopeless. However, if you are satisfied with a workaround, rather that solving the problem the right way, this is how I finally managed:

On a Windows PC:

1. Create "Hiren's BootCD PE" bootable USB stick (I used an old 4GB stick)

1a. Download Hiren's BootCD PE ISO image from https://www.hirensbootcd.org/
1b. Download and install rufus.exe from https://rufus.ie/en/
1c. Write ISO image to USB stick using Rufus

On the target Linux system:

2. Download flash BIOS update from Lenovo

2a. Download BIOS Update (Operating System Version) EXE file from (in my case)
https://pcsupport.lenovo.com/tc/en/products/desktops-and-all-in-ones/thinkcentre-m-series-desktops/thinkcentre-m70q-gen-4/downloads/ds561027-flash-bios-update-thinkcentre-m70q-gen-4
2b. Copy EXE file to root of USB stick

3. Apply flash BIOS update

3a. Boot from the bootable USB stick into Windows Environment
3b. Run BIOS update EXE file as Administrator
3c. Answer questions and select "Extract" (do not choose "Install"; that failed without error message)
3d. Choose target extraction folder to, e.g., Y:\DRIVERS\FLASH\Lenovo, if Y is your bootable USB stick
3e. Open folder Y:\DRIVERS\FLASH\Lenovo and run, e.g. wFlashGUIX64.exe
    (that executable might vary and could be a command line tool wflash-something)
3f. Answer questions in the GUI and reboot

Success!

For next update, just start from item 2 armed with your Windows PE-bootable stick.

I suppose this is not the elegant solution you're hoping for, but maybe useful anyway.

ElCoyote27, 2025-09-06:
Another alternative way (for people who have extra hardware):
- Unplug the Monitor from your P350/P360.
- Connect USB + HDMI from a PiKVM (using a V4 Mini)
- Download and then re-upload the Lenovo ISO image to the PiKVM USB storage and connect the ISO.
- Reboot your P350/P360 and select to boot from the PiKVM ISO image.
- Follow the Lenovo promps.

[image]

ElCoyote27, 2026-03-09:
@mrhpearson Any hope for the P350 too?