fwupd/firmware-lenovo #511: Firmware update not available for Thunderbolt 4 40B1

bitonic, 2025-05-11:
Hi,

I'm trying to update the firmware for my Thunderbolt 4 40B1 dock (in the hope of fixing #191), but I don't get any updates, even if my fw version is out of date:

```
├─ThinkPad Universal Thunderbolt 4 Smart Dock:
│ │   Device ID:          64d62e2af417c24295829c53e56618e701ab2720
│ │   Current version:    10.6
│ │   Vendor:             Lenovo (USB:0x17EF)
│ │   Serial Number:      XXX
│ │   GUIDs:              8d30b09f-bcc5-5379-bc65-9ccceeece1f4 ← USB\VID_17EF&PID_30B4
│ │                       72918203-63fd-5064-b704-85200e0f3b72 ← USB\VID_17EF&PID_30B4&CID_40B1
│ │                       6e78e123-f3e0-5243-b320-217cc66c2e81
│ │                       b17dff10-684e-50b5-861d-0150e8171f3a ← USB\VID_17EF&PID_30B4&CID_40B1&REV_0100
│ │   Device Flags:       • Updatable
│ │                       • System requires external power source
│ │                       • Device stages updates
│ │                       • Signed Payload
│ │   Device Requests:    • Message
```

Any idea of what this might be due to? I'm using fwupd version 1.9.25 on NixOS, and my laptop is a Framework 13, Intel, 13th generation.

bitonic, 2025-05-12:
@norru presumably you disabled the PCI Ethernet by disabling PCIe tunnelling in the BIOS? I can't find an equivalent option on the framework...

Also, do you know if the other TB4 docks work better on Linux?

norru, 2025-05-12:
@bitonic 

Yes, I've disabled PCI tunnelling from the BIOS. The USB LAN is stable (but slower).

I have seen some community reports of the PCI LAN of the "non-smart" version of the dock not working correctly either, even after updating the FW (which appears to be up-to date on LVFS).

bitonic, 2025-05-12:
@norru honestly I'd be OK with the Ethernet being over USB, but I can't seem to disable it on my BIOS in any way... which makes the dock effectively useless. Hence me wanting to find a way to force it to use the USB interface.

In any case, thanks a lot for the info.