fwupd/firmware-lenovo #512: fwupd 2.0.9: Updates not applied on ThinkPad X1 Carbon Gen 8, Arch Linux

shawnyeager, 2025-05-13:
Related: #56 

Since buying my second-hand X1 last Fall, I've been unable to apply firmware updates successfully using `fwupd`. Those which I have applied successful have been by creating bootable Windows USB sticks. However, the two most recent updates for firmware and Intel Management Engine do not offer this option, so I'm in a bind.

I'm running Arch Linux with LUKS encryption, systemd-boot, UKI, Plymouth, and secure boot, should these be relevant.

In every case, whether using `fwupdmgr` directly or Gnome Firmware, after reboot, I tap `F12` to select boot partition, select `Linux Firmware Updater` (having assured that it's listed first in boot devices), but then after a very brief delay, I'm taken to Plymouth and then to encryption password prompt.

kmauleon, 2025-05-13:
please try to disable secure boot and update firmware one by one

shawnyeager, 2025-05-13:
Thanks, @kmauleon. That did work. I had tried unsuccessfully with secure boot off using previous releases of `fwupd`, so it didn't occur to try again.

FYI that both updates applied in one go.

It might be useful to advise or even refuse to update with secure boot on, just as as it does when not on AC power.

kmauleon, 2025-05-13:
great!  thanks for the update @shawnyeager ... closing this ticket ^_^

closed 2025-05-13 by kmauleon