fwupd/firmware-lenovo #513: Thinkpad P1 Gen 6 shows "No matching device" error on update

iqbalabd, 2025-05-13:
When I try to update the firmware after, I get a "No matching device" error even though `fwupdmgr` tells me there is an update available and downloading it.

The device in question:
```
$ fwupdmgr get-devices

├─Intel Management Engine:
│     Device ID:          32de03ef370e5da5582976b375b3946e03236bb2
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    1.32.2418
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               4e78ce68-5389-4a95-bf10-e3568c30caf8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
```

I'm using Ubuntu 24.04

My system:
```
$ for d in system-manufacturer system-product-name bios-release-date bios-version \
do \
echo "${d^} : " $(sudo dmidecode -s $d) \
done

System-manufacturer :  LENOVO
System-product-name :  21FWS11X00
Bios-release-date :  01/02/2025
Bios-version :  N3ZET47W (1.34 )
```

`fwupdmgr` tells me an update is available to version `1.35.2557` for the ME device.

```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • Fingerprint Sensor
 • Integrated RGB Camera
 • SKHynix HFS001TEJ9X162N
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB4 host controller
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
LENOVO 21FWS11X00
│
└─Intel Management Engine:
  │   Device ID:          32de03ef370e5da5582976b375b3946e03236bb2
  │   Summary:            UEFI System Resource Table device (updated via NVRAM)
  │   Current version:    1.32.2418
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               4e78ce68-5389-4a95-bf10-e3568c30caf8
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  └─ThinkPad P1 Gen 6 Corporate Corporate ME Update:
        New version:      1.35.2557
        Remote ID:        lvfs
        Release ID:       110785
        Summary:          Lenovo ThinkPad P1 Gen 6 Corporate ME Firmware
        License:          Proprietary
        Size:             10.0 MB
        Created:          2025-03-28
        Urgency:          High
        Tested by Lenovo:
          Tested:         2025-03-28
          Distribution:   ubuntu-core 24
          Old version:    1.32.2418
          Version[fwupd]: 2.0.5
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3ZRG03W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
                          • Tested by trusted vendor
        Description:      
        Intel Platform Update 2024.3 Product Version Maintenance Release
        
        Corporate Version 16.1.35.2557 (LVFS: 1.35.2557)
        Checksum:         7ffb8de49f99542d82618fbefa30a939d8d0a6a047aeae73e10a4767b4c67ed7
```

But when I actually run the update, it says "No supported devices found"

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • Fingerprint Sensor
 • Integrated RGB Camera
 • SKHynix HFS001TEJ9X162N
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB4 host controller
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.32.2418 to 1.35.2557?                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Intel Platform Update 2024.3 Product Version Maintenance Release             ║
║                                                                              ║
║ Corporate Version 16.1.35.2557 (LVFS: 1.35.2557)                             ║
║                                                                              ║
║ 21FWS11X00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Decompressing…           [***************************************]
No supported devices found
```

This is what it looks like from the Firmware Updater GUI:

[image]

kmauleon, 2025-05-14:
hi @iqbalabd it seems your system GUID does not match that of the firmware from the LVFS site.  Will contact ODM and update you later... thank you very much for reporting

kmauleon, 2025-05-15:
hi @iqbalabd ODM CO ask to update system BIOS to [1.35](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3ZET.firmware) first.  can you please check.  thank you very much

iqbalabd, 2025-05-15:
Hi,
I upgraded the BIOS to 1.35 and can also now update the ME firmware.
Thank you for your pointer!

closed 2025-05-15 by iqbalabd

hughsie, 2025-05-15:
@kmauleon if there's a dependency on a different ESRT component then it should have an additional requirement -- e.g. https://lvfs.readthedocs.io/en/latest/metainfo.html#other-firmware-version

kmauleon, 2025-05-15:
@hughsie will remind ODM CO about this again... thank you very much!