fwupd/firmware-lenovo #514: [Thinkpad X1 Carbon Gen 12] "Could not find updates: Not found" message

jbiel, 2025-05-28:
Hello,

I'm running `fwupdmgr` on my laptop and attempting to perform three updates that it says are available. When I boot into the EFI entry manually it flashes the following error very quickly before going back to the EFI boot selector.

[image]

I created a `systemd-boot` entry for it in order to try a different way, and it shows a similar (but slightly different) error.

[image]

Debug output:
```
$ fwupdmgr --version     
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.9
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.8
runtime   org.freedesktop.fwupd         2.0.9

# tree /boot
/boot
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Linux
│   ├── nixos
│   │   ├── 9j6m02vwsslp4l39xcxg5gzc7cchp4dc-linux-6.14.7-bzImage.efi
│   │   ├── fw
│   │   │   ├── fwupd-420e7cf1-cedf-4b5f-9ef0-f1896a946d8f.cap
│   │   │   ├── fwupd-573c8caf-fbdb-41a5-8f1a-c87d6695d39a.cap
│   │   │   └── fwupd-76ab0e88-4c8a-488e-9764-e3ab0bf35fac.cap
│   │   ├── fwupdx64.efi
│   │   ├── nrd1sf5aqgyhnbjqi30ip99w8xpcx1hw-linux-6.14.8-bzImage.efi
│   │   ├── wxz8vz64hwab8wqn41ghmr9jv3vd7ji5-initrd-linux-6.14.7-initrd.efi
│   │   └── z2hkp96nfg4qniw1q4ldkc9dzy9grj5x-initrd-linux-6.14.8-initrd.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   ├── firmware-update.conf
    │   ├── nixos-generation-205.conf
    │   ├── nixos-generation-206.conf
    │   ├── nixos-generation-207.conf
    │   ├── nixos-generation-208.conf
    │   └── nixos-generation-209.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed

9 directories, 19 files

# fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 0.10.2269 to 0.15.2515?                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Meteorlake 2025-WW09 HF Release                                              ║
║                                                                              ║
║ Corporate Version 18.0.15.2515 (LVFS: 0.15.2515)                             ║
║                                                                              ║
║ Fixed an issue wherein system may encounter system hang after reboot while   ║
║ LAN cable is connected.                                                      ║
║                                                                              ║
║ Fixed an issue wherein Intel AMT fails to resume the CIRA connection after   ║
║ it disconnects due to network instability.                                   ║
║                                                                              ║
║ 21KCCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.39 to 0.1.40?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.40                                          ║
║                                                                              ║
║ Important updates                                                            ║
║                                                                              ║
║ • Enhancement to address security vulnerabilities                            ║
║                                                                              ║
║ New functions or enhancements                                                ║
║                                                                              ║
║ • ThinkShield secure wipe can work with AC adapter even with less than       ║
║ 25% Battery Capacity.                                                        ║
║                                                                              ║
║ 21KCCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Updating System Firmware…[  -                                    ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI Device Firmware from 0.1.26 to 0.1.27?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.27                                          ║
║                                                                              ║
║ Problem fixes                                                                ║
║                                                                              ║
║ • Fixed an issue where specific monitor, or a monitor on specific docking    ║
║ station is not recognized through type-C port.                               ║
║                                                                              ║
║ 21KCCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Updating UEFI Device Firmware…                                   ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • KXG8AZNV1T02 LA KIOXIA
 • UEFI dbx
Devices with no available firmware updates: 
 • Integrated Camera
 • KEK CA
 • KEK CA
 • Prometheus
 • Prometheus IOTA Config
 • ThinkPad Product CA
 • UEFI CA
 • UEFI CA
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • Windows UEFI CA
An update requires a reboot to complete. Restart now? [y|N]: n

# /etc/nixos/configuration.nix

  boot.loader = {
    systemd-boot = {
      enable = true;
      extraEntries = {
        "firmware-update.conf" = ''
          title Firmware Update
          efi /EFI/nixos/fwupdx64.efi
        '';
      };
    };
    efi.canTouchEfiVariables = true;
  };

  services.fwupd.enable = true;
```

kmauleon, 2025-05-28:
hi @jbiel can you please check entries in the ESRT... will send them to BIOS dev for reference.  thank you very much...
grep . /sys/firmware/efi/esrt/entries/*/*

jbiel, 2025-05-29:
```
# grep . /sys/firmware/efi/esrt/entries/
grep: /sys/firmware/efi/esrt/entries//: Is a directory

# tree /sys/firmware/efi/esrt/entries/
/sys/firmware/efi/esrt/entries/
├── entry0
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry1
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry10
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry11
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry12
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry13
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry14
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry15
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry16
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry17
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry2
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry3
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry4
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry5
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry6
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry7
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
├── entry8
│   ├── capsule_flags
│   ├── fw_class
│   ├── fw_type
│   ├── fw_version
│   ├── last_attempt_status
│   ├── last_attempt_version
│   └── lowest_supported_fw_version
└── entry9
    ├── capsule_flags
    ├── fw_class
    ├── fw_type
    ├── fw_version
    ├── last_attempt_status
    ├── last_attempt_version
    └── lowest_supported_fw_version

19 directories, 126 files
```

kmauleon, 2025-05-29:
@jbiel I already forwarded the issue to BIOS dev... let's wait for feedback ... i'll update as soon as I heard from them... thank you

kmauleon, 2025-06-04:
hi @jbiel BIOS dev was not able to recreate the issue.  however, they said BIOS 1.42 will be available soon.  would it be OK to wait for this version and hopefully it can resolve your issue... will update you again by next week.  thank you very much.

kmauleon, 2025-06-04:
btw, can you please run this command again... thank you... 

[image]

avdv, 2025-06-04:
I am using a Thinkpad X1 Carbon Gen 9, and I am experiencing the "LvbrBoot: Not Found" error when trying to install an update. I am also on NixOS, BTW.

Here's the output for `sudo grep . /sys/firmware/efi/esrt/entries/*/*`:
```
/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:7956e676-a3c9-4d8d-9378-4938fdc35c04
/sys/firmware/efi/esrt/entries/entry0/fw_type:2
/sys/firmware/efi/esrt/entries/entry0/fw_version:2295799
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:76ca0ad8-4a14-4389-b7e5-fd88791762ad
/sys/firmware/efi/esrt/entries/entry1/fw_type:2
/sys/firmware/efi/esrt/entries/entry1/fw_version:1
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:626d93db-2c42-48c3-915a-71f968a81b04
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:1
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry3/fw_class:3dd84775-ec79-4ecb-8404-74de030c3f77
/sys/firmware/efi/esrt/entries/entry3/fw_type:2
/sys/firmware/efi/esrt/entries/entry3/fw_version:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry4/fw_class:69585d92-b50a-4ad7-b265-2eb1ae066574
/sys/firmware/efi/esrt/entries/entry4/fw_type:2
/sys/firmware/efi/esrt/entries/entry4/fw_version:1
/sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry5/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry5/fw_class:4e88068b-41b2-4e05-893c-db0b43f7d348
/sys/firmware/efi/esrt/entries/entry5/fw_type:2
/sys/firmware/efi/esrt/entries/entry5/fw_version:69145
/sys/firmware/efi/esrt/entries/entry5/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry5/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry5/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry6/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry6/fw_class:5fd44f34-6614-47e2-96f4-c80c8a0270e2
/sys/firmware/efi/esrt/entries/entry6/fw_type:2
/sys/firmware/efi/esrt/entries/entry6/fw_version:4130
/sys/firmware/efi/esrt/entries/entry6/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry6/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry6/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry7/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry7/fw_class:d2718981-bb13-45b4-b919-16bd42529a70
/sys/firmware/efi/esrt/entries/entry7/fw_type:2
/sys/firmware/efi/esrt/entries/entry7/fw_version:9568256
/sys/firmware/efi/esrt/entries/entry7/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry7/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry7/lowest_supported_fw_version:57374
/sys/firmware/efi/esrt/entries/entry8/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry8/fw_class:14f3350e-cf63-4e68-a0d9-0af1d5389a17
/sys/firmware/efi/esrt/entries/entry8/fw_type:1
/sys/firmware/efi/esrt/entries/entry8/fw_version:65608
/sys/firmware/efi/esrt/entries/entry8/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry8/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry8/lowest_supported_fw_version:65575
/sys/firmware/efi/esrt/entries/entry9/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry9/fw_class:03c27e00-8229-4c14-ba40-2afe2276555c
/sys/firmware/efi/esrt/entries/entry9/fw_type:2
[... 23 lines cut ...]
/sys/firmware/efi/esrt/entries/entry12/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry12/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry13/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry13/fw_class:aa096a98-94e6-479b-92f7-5771f6f2d96f
/sys/firmware/efi/esrt/entries/entry13/fw_type:2
/sys/firmware/efi/esrt/entries/entry13/fw_version:16915456
/sys/firmware/efi/esrt/entries/entry13/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry13/last_attempt_version:16915456
/sys/firmware/efi/esrt/entries/entry13/lowest_supported_fw_version:1
```
Note that, meanwhile, I also updated the system firmware using the .iso from Lenovo.

fwupdmgr info:
```console
# fwupdmgr --version
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.9
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.7-zen1
runtime   org.freedesktop.fwupd         2.0.9
```

jbiel, 2025-06-06:
Thank you. For others that run into this, [see my comment here](https://github.com/fwupd/firmware-lenovo/issues/515#issuecomment-2950900997) for a workaround until `fwupd-efi` is updated in nixpkgs.