fwupd/firmware-lenovo #526: "Failed successfully" Micron NVMe firmware

r3pek, 2025-07-28:
Strange thing happened to me today while updating my Micron NVMe drive's firmware. 
Did it via KDE Discover (fedora) and it asked me to reboot to finish the update

After reboot, it just printed out a bunch of nvme "not found" errors and it wouldn't boot. Soft-rebooted the laptop, and it wouldn't boot either (but it had the "Linux-Firmware-Updater" boot entry on the EFI boot order). 

Powered it down and back on and everything went well again and the firmware was updated. Nothing was ever shown as updating so I really had no idea of what happened.

The device itself is this:
```
├─Micron MTFDKBA1T0TGD-2BK15ABLT:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    7007P8LN
```
The update was from 7006 to 7007. 

(running a ThinkPad P14s Gen 5)

hughsie, 2025-07-28:
@mrhpearson can you ask Micron if they need the `Flags = needs-shutdown` quirk in https://github.com/fwupd/fwupd/blob/main/plugins/nvme/nvme.quirk#L1 please?

mrhpearson, 2025-08-05:
Apologies - I missed this one in my inbox last week.

I've flagged this to the storage team - internal ticket is LO-3912.

Could you confirm if this is an Intel or AMD platform please (not sure it will matter, but in case it's important)

mrhpearson, 2025-09-09:
Thanks @hughsie

hughsie, 2025-09-09:
There's also a new Fedora build with this fix in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ -- if that helps.