fwupd/firmware-lenovo #528: update loop during boot for ME firmware update

PfalzTom, 2025-08-05:
Similar to issue #303 I have an T15 thinkpad and during each boot i get: `Reading ME Firmware ... Please do not power off`with a counter that goes up to approximately 48%, then `Recovering ME Firmware...Please do not power off` counting up to approximately 10% and then shortly `Failed to recover ME Firmware...`. This delays booting by approximately 3 minutes (not measured).

After that grub and then Debian 13 boots fine, the audio driver fails to load with

    [   13.485452] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read

Other than in #303 `fwupdmgr reinstall` doesn't help.

```
# LC_ALL=en_US.UTF-8 fwupdmgr get-devices
LENOVO 20W5S15L01
│
├─Embedded Controller:
│     Device ID:          90a0f142b37f5e576245d70de0d5b9a7436da795
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.44
│     Minimum Version:    0.1.44
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Transient failure
│     Update Error:       failed to update to 0: error-pwr-evt-batt
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Integrated Camera:
│     Device ID:          28a4aa67dfde939ffdb66e6975c4b676643e27dc
│     Current version:    63.18
│     Vendor:             Chicony Electronics Co., Ltd (USB:0x04F2)
│     Serial Number:      0001
│     GUID:               f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│     Device Flags:       • Updatable
│                         • Can tag for emulation
│   
├─Intel Management Engine:
│     Device ID:          12a1e60fc79b3ec8df56210363e73bc22340fd7a
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    240.50.2633
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               9a03b1ec-b756-4414-bc09-1f573603aeaa
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Prometheus:
│ │   Device ID:          ff75e69eba7f7bc73bf2e2524de4c41d32d61506
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.4028721
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      69442766475143
│ │   GUID:               8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        2581b515c779125bfc75dfdf633c3b6651c500d6
[... 145 lines cut ...]
```
# fwupdmgr reinstall
0.	Abbrechen
1.	90a0f142b37f5e576245d70de0d5b9a7436da795 (Embedded Controller)
2.	12a1e60fc79b3ec8df56210363e73bc22340fd7a (Intel Management Engine)
3.	ff75e69eba7f7bc73bf2e2524de4c41d32d61506 (Prometheus)
4.	2581b515c779125bfc75dfdf633c3b6651c500d6 (Prometheus IOTA Config)
5.	90ff66b6f6f6862361154a1c6d79ede491f2406c (System Firmware)
Gerät auswählen [0-5]: 2
╔══════════════════════════════════════════════════════════════════════════════╗
║ Intel Management Engine auf 240.50.2633 neu installieren?                    ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Mitigated the following security vulnerabilities                           ║
║                                                                              ║
║ 20W5S15L01 muss während der gesamten Dauer der Aktualisierung an eine        ║
║ Stromquelle angeschlossen bleiben, um Schäden zu vermeiden.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: y
Warten …                 [***************************************]]Weniger als eine Minute verbleiben…
Erfolgreich installierte Firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Ein Neustart ist erforderlich, um eine Aktualisierung abzuschließen. Jetzt neu starten? [y|N]: 
```

mrhpearson, 2025-08-05:
Yeah - I think so. @kmauleon - is this something you can help with please? Platform is T15 Gen 2 Intel
I created internal ticket LO-3911 for tracking

As a note - it looks like the BIOS and EC are a bit out of date. Don't know if that is relevant or not

ukleinek, 2025-08-05:
Regarding the note that BIOS and EC are out of date: That laptop also runs Windows 11, and one of the tries to get that issue solved was to completely update the machine using Windows. When we reported the issue the machine was completely up to date according to windows update, and (after reporting the issue) we also installed the Intel Management Engine Software from https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t15-gen-2-type-20w4-20w5/downloads/driver-list/component?name=Motherboard%20Devices%20%28Backplanes,%20core%20chipset,%20onboard%20video,%20PCIe%20switches%29&id=FB762CAD-DE8B-49C2-A04B-809B4958F608 version 2512.7.3.0 dating 2025-08-01. That update was successful, but the issue with reading / recovering ME firmware at each boot still persists.

ukleinek, 2025-08-06:
> while waiting can you please try this one [#295 (comment)](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135) thank you very much!

we tried `fwupdmgr reinstall` several times already. This offers a menu of (IIRC) 5 selections. We tried them all, doesn't help.

> please update [BIOS](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34ET.firmware) first to latest then fwupdmgr reinstall MEFW... let us know how it goes... thank you very much...

How would updating the bios work? I would guess `fwupdmgr update 90ff66b6f6f6862361154a1c6d79ede491f2406c`?

kmauleon, 2025-08-06:
you can download BIOS from the link I provided, navigate to the location of the BIOS cab file then install via

[image]

PfalzTom, 2025-08-06:
I tried to install the linked cab file, I had to wait twice for the ME firmware stuff during the following reboot; between the two runs fwupd-efi appeared shortly, didn't spot a problem there.

The output of 'LC_ALL=en_US.UTF-8 fwupdmgr get-devices` now has 
```
├─System Firmware:
│     Device ID:          90ff66b6f6f6862361154a1c6d79ede491f2406c
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.58
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Transient failure
│     Update Error:       failed to update to 117474352: error-pwr-evt-batt
│     GUID:               973a10fa-79c0-453e-b78c-27b55cc0e6eb
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
```

So where we had a 0 before now there is a 117474352. Am I in the position now to reinstall the ME firmware?

PfalzTom, 2025-08-07:
> [@PfalzTom](https://github.com/PfalzTom) please make sure AC attached and battery at least 25% and SecureBoot disabled before updating the firmware.... you are still in low level BIOS... can you please try updating to 1.64... same procedure please download [here](https://fwupd.org/downloads/af4a3da7dde027267e4323c8745ca710930b3a6274a0d700595ad96b186b8918-N34ET64W.cab)

Secure boot is (and already was) disabled. I tried to install 1.64, the update process looked fine, battery at > 60% and AC attached, but I still have:
```
# LC_ALL=en_US.UTF-8 fwupdmgr get-devices
LENOVO 20W5S15L01
│
├─11th Gen Intel Core™ i5-1145G7 @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000bc
│     Vendor:             Intel
│     GUIDs:              2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          90a0f142b37f5e576245d70de0d5b9a7436da795
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.44
│     Minimum Version:    0.1.44
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Transient failure
│     Update Error:       failed to update to 0: error-pwr-evt-batt
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Integrated Camera:
│     Device ID:          28a4aa67dfde939ffdb66e6975c4b676643e27dc
│     Current version:    63.18
│     Vendor:             Chicony Electronics Co., Ltd (USB:0x04F2)
│     Serial Number:      0001
│     GUID:               f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│     Device Flags:       • Updatable
│                         • Can tag for emulation
│   
├─Intel Management Engine:
│     Device ID:          12a1e60fc79b3ec8df56210363e73bc22340fd7a
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    240.50.2633
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               9a03b1ec-b756-4414-bc09-1f573603aeaa
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Prometheus:
│ │   Device ID:          ff75e69eba7f7bc73bf2e2524de4c41d32d61506
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.4028721
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      69442766475143
│ │   GUID:               8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │                       • Can tag for emulation
[... 166 lines cut ...]
│     GUID:               f0f8c8b2-9c7d-480c-9c2a-2831f1d7d7c3
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Device Firmware:
      Device ID:          5e32c067118a9d0a0ec2cf5b8d9de3382c70cf04
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    25368
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               4b3d2033-ae0c-4f25-bc3f-96f0ed347569
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message
    
```
indicating that the update didn't take place :-(

kmauleon, 2025-08-07:
I'm guessing because of this error...
│ │   Update State:       Transient failure
│ │   Update Error:       failed to update to 117474352: error-pwr-evt-batt
Could be hardware problem... I'll confirm with ODM....

kmauleon, 2025-08-14:
hi @PfalzTom message from ODM...
Not sure what prevented the machine from upgrading the BIOS and EC. It seems that the customer's BIOS and EC have not been updated for a long time. Did the customer know when they started experiencing issues with not being able to update the BIOS and EC?

hi @hughsie do you know what the error means?
│     Update State:       Transient failure
│     Update Error:       failed to update to 0: error-pwr-evt-batt

hughsie, 2025-08-14:
> error-pwr-evt-batt

This is _UEFI Update Capsule_ language for "battery too low"

ukleinek, 2025-08-14:
> hi [@PfalzTom](https://github.com/PfalzTom) message from ODM... Not sure what prevented the machine from upgrading the BIOS and EC. It seems that the customer's BIOS and EC have not been updated for [...]

IIUC the laptop is a refurbished device that @PfalzTom just recently bought. We installed Debian on it and shortly after that (I guess once the fwupdmgr timer ran) the issue appeared.

kmauleon, 2025-08-14:
@mrhpearson @hughsie this seem to be a hardware issue..right?  fwupd is getting incorrect message of low bat thus firmware won't update.... 

@PfalzTom can i have BIOS Menu screen shot please... thank you very much

kmauleon, 2025-08-19:
it's the owner of the firmware... I will ask them to provide the files... can you please send me a note to kmauleon at lenovo dot com please.... thanks

kmauleon, 2025-09-15:
with this state can you please try to execute the following again...

fwupdmgr refresh --force
fwupdmgr update

PfalzTom, 2025-09-16:
Did that, didn't help :-(

The refresh reported 5 pending updates, the update command then resulted in
```
# LC_ALL=en_US.UTF-8 fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.44 to 0.1.45?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Embedded Controller Firmware N34HT45W version 1.45.Thermal improvement       ║
║ update.                                                                      ║
║                                                                              ║
║ 20W5S15L01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
```
Confirming (with PSU connected) resulted in the expected download, a reboot did end in the fwupd component and another reboot (each reboot accompanied by the ME firmware reading + recovery procedure).

Calling `fwupdmgr update` then suggests to install to 0.1.45 again.

Now the status quo is:
```
christopheros@cts:~$ LC_ALL=en_US.UTF-8 fwupdmgr get-devices
LENOVO 20W5S15L01
│
├─11th Gen Intel Core™ i5-1145G7 @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000bc
│     Vendor:             Intel
│     GUIDs:              2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          90a0f142b37f5e576245d70de0d5b9a7436da795
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.44
│     Minimum Version:    0.1.44
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Transient failure
│     Update Error:       failed to update to 0: error-pwr-evt-batt
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─Integrated Camera:
│     Device ID:          28a4aa67dfde939ffdb66e6975c4b676643e27dc
│     Current version:    63.18
│     Vendor:             Chicony Electronics Co., Ltd (USB:0x04F2)
│     Serial Number:      0001
│     GUID:               f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│     Device Flags:       • Updatable
│                         • Can tag for emulation
│   
├─Intel Management Engine:
│     Device ID:          12a1e60fc79b3ec8df56210363e73bc22340fd7a
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    240.50.2633
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               9a03b1ec-b756-4414-bc09-1f573603aeaa
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 181 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               289be5fa-31a9-44b3-981a-f934377dcc3c
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Device Firmware:
      Device ID:          5e32c067118a9d0a0ec2cf5b8d9de3382c70cf04
      Summary:            UEFI System Resource Table device (updated via NVRAM)
      Current version:    25368
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               4b3d2033-ae0c-4f25-bc3f-96f0ed347569
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message
```

kmauleon, 2025-09-17:
can you please try this one then...
download cab file [here](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34RGM.firmware)
navigate to the location of the cab file then run the following

[image]

kmauleon, 2025-09-22:
hi @PfalzTom as per ODM, you might need to fix battery issue first so you could proceed with the update.  sorry but unfortunately we can't do anything further from our side

ukleinek, 2025-09-22:
Is the battery issue relevant enough that the machine can be handed back to the seller (assuming it still is within warranty)? I have no idea how I'd approach fixing the battery issue. Apart from the diagnosis by fwupd the battery seems to work fine.

Which component determines the battery isn't charged enough for the update? Is that fwupd-efi? Or some part within the firmware? If it's the former, can you please point us to the relevant source code to maybe better understand the issue?

To make the machine less annoying to work with, is there a way to "unschedule" the firmware update such that rebooting the machine doesn't always takes several minutes?

ukleinek, 2025-09-24:
@mrhpearson: I confirm PfalzTom and I are talking about the same laptop. I'm the Linux guy in the office and occasionally support PfalzTom.

I wonder: Where is the relevant data stored that makes the machine try to update the ME firmware at each boot. (Probably) fwupdmgr put the machine in that state. Can that maybe be undone to at least make the machine boot faster again? (Though that might be irrelevant if the laptop is RMAd.)

mrhpearson, 2025-09-26:
Check what is in your EFI partition - usually under '/boot/EFI/$(distro)/fw'

Also check your efi boot params (running 'efibootmgr' should display them). If it has fwupd as the first thing to run you can remove it from the list and that might help?

ukleinek, 2025-09-27:
@mrhpearson I don't have access to the machine right now, so I cannot confirm, but IIRC there is nothing in efibootmgr hinting to fwupd. In my understanding the action happens before the efiboot entries get activated. That at least explains why we see Reading-Recover-Failed then efifwupd and then again Reading-Recover-Failed followed by grub when fwupdmgr scheduled an update.

mrhpearson, 2025-10-03:
@kmauleon - can you confirm with the FW team if there will be some reason the FW update will continuously retry please? Are there some EFI variable we can check (and clear?)

kmauleon, 2025-11-16:
Sorry @PfalzTom .. as mentioned, please reach out to Lenovo Support in your area... closing this for now... thank you very much!

PfalzTom, 2025-11-29:
I sent my T15 device to the seller without a hard drive. 

When it was returned, I received the following information: 

“BIOS repaired. Full functionality of hardware components ensured and tested.”

So far, everything is working and the errors described above are not occurring.

Best regards and many thanks to everyone for all your support!

kmauleon, 2025-11-30:
Thank you very much for the update @PfalzTom !  glad that the issue was fixed ^_^