fwupd/firmware-lenovo #536: unable to update on T480  due to checksum failure

modernNeo, 2025-08-30:
```bash
2025-Aug-30 02:46:48 PM modernNeo@debian-thinkpad:/home/modernNeo
 $ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported
2025-Aug-30 02:46:52 PM modernNeo@debian-thinkpad:/home/modernNeo
 $ sudo fwupdmgr update
Devices with no available firmware updates: 
 • CT1000MX500SSD1
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Thunderbolt host controller
 • Embedded Controller
Upgrade available for Intel Management Engine from 184.86.3909 to 184.96.4657
20L5CTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading…             [***************************************]
checksum invalid, expected 087e2451105c12d4912a533b444a8e044960220d got 5088fe92c2db672a97f9a4a09f7478e50449a63d

```
not sure how to fix this?

hughsie, 2025-08-31:
> checksum invalid, expected 087e2451105c12d4912a533b444a8e044960220d got 5088fe92c2db672a97f9a4a09f7478e50449a63d

That looks like a client download failure -- if you do `wget https://fwupd.org/downloads/2c4c71d3b9950429f83d0c853aadfc134e16975beae46d6361d3f444d31e3dda-Lenovo-ThinkPad-T480-CorporateMEFirmware-11.8.96.4657.cab` and then `sha256sum 2c4c71d3b9950429f83d0c853aadfc134e16975beae46d6361d3f444d31e3dda*` do you then get `2c4c71d3b9950429f83d0c853aadfc134e16975beae46d6361d3f444d31e3dda`? If not, if you open the file do you get an html captive proxy landing page or something?

closed 2025-09-29 by hughsie

hughsie, 2025-10-20:
@modernNeo it's not up to me, you probably need to update to a newer debian version. If that EOL version is in a supported Debian release you should file a bug with them and point out that the fwupd version is LOOONG EOL and that they need to rebase.