fwupd/firmware-lenovo #538: I have a broken touchpad afrer update

cblp0k, 2025-09-15:
Hello. I need your help. I think i am in dead end and i have no idea what is i need to do next. I have a trouble with the thinkpad z13 gen2. Some time ago i updated a haptic trackpad firmware on my laptop and after that i have a broken touchpad. It is does not working corectly, in fact, it is unusable. This firmware is not a component of operation system. It is an embedded software of a part of my PC touchpad so I have problem on Linux and Windows it is not depends. 

A few words about update procedure:
To update i used:
```
fwupdmgr update 
fwupdmgr get-upgrades 
sudo reboot
```
Flashing process has been compleated without any errors.

And this is my update history:

```
├─Touchpad:
│ │   Device ID:          98e2c92219fd82ea7d5efeedb9f038f10e67d56c
│ │   Previous version:   1.3.474
│ │   Update State:       Success
│ │   Last modified:      2025-09-13
│ │   GUID:               940ccbfc-ce9f-4b99-ae99-666db2c06195
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ └─HapticTrackpad Touchpad Update:
│       New version:      1.3.674
│       Remote ID:        lvfs
│       Release ID:       64954
│       Summary:          Firmware for Sensel's Haptic Trackpad
│       License:          Proprietary
│       Size:             110.5 kB
│       Created:          2023-09-27
│       Urgency:          High
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=726f7a2e3bfbddf4069c840619f3c6848c03204a
│       Vendor:           Lenovo
│       Duration:         2 minutes
│       Release Flags:    • Trusted payload
│                         • Trusted metadata
│       Description:      
│       This stable release fixes the following bugs:
│       
│       • Improves initialization stability
│       Checksum:         2ffcb5c71a07502e7b0a844517eceda74ee1e130cefd0cb5037c9ef862885340
```

I tried to do roll out to previous working version but there is no that version in fwupd.org repository same as in Lenovo drivers page for thinkpad z13 gen2. I wrote to the Lenovo support, to try to geting my old firmware working version but they did not answer yet https://forums.lenovo.com/t5/Other-Linux-Discussions/Need-a-driver/m-p/10001693. 

So we have:
1.Version: 1.3.474 Old fine working firmware i do not know how to restore because I have not files to restore
2.Version: 1.3.674 New one broke the touchpad and currently installed

Besides, i found thinkpad z16 gen2 user, who reports about the same problem https://www.reddit.com/r/thinkpad/comments/1mpv0j2/z16_gen_2_sensel_touchpad_issues_after_firmware/. 

z13 gen2 and z16 gen2 both have pretty simmilar touchpads from one manufacturer which makes me think that problem is bigger than it seems.

But maybe i did somethink wrong... I have a pretty small EFI partition. It is only 100Mb and only 65 is free. Can it be the reason?
```
/dev/nvme0n1p1      96M          32M   65M           33% /boot/efi
```

Please, give me some advices or roadmap what i should to do. As a result I want to:
1.Confirm or deny susspitious about version 1.3.674 is broken
2.If it is broken I need to do something to remove it from fwupd to keep safe other users
2.Make my touchpad working correctly without replace it

mrhpearson, 2025-09-15:
I've created internal ticket LO-3999 to get feedback from the FW team on this. Will let you know when we hear back from them.

mrhpearson, 2025-09-22:
Hi @cblp0k 
The team are looking at it, but so far haven't been able to reproduce the issue.
We're seeing if we can get you a mechanism to downgrade the FW to confirm if that fixes it.

Mark

mrhpearson, 2025-09-25:
Thanks for the update - we're checking with the HW team.

WC-william, 2026-09-01:
I just did the exact same thing, updating from 1.3.474 to 1.3.822. My laptop is a Z16gen2. Now the touchpad is behaving erratically and is basically unusable. 

Could you please provide a roll back channel to the 1.3.474 or any other ways to solve this other than replacing the touchpad? Thanks a lot.

mrhpearson, 2026-09-02:
Checking with the FW team - it looks like the investigation as to why it was breaking hit a dead end. So asking for a way to get you back to a working state.

Royc30ne, 2026-10-07:
I’m experiencing the same issue. I updated the touchpad firmware today, and my haptic touchpad stopped working afterward.
Are there any updates from the firmware team on a recovery method? Could you please provide a way to roll back to the previous firmware version? I’d really appreciate a software-based fix rather than having to replace the touchpad.
Please let me know what device information or diagnostic logs would help with the investigation. Thanks!