fwupd/firmware-lenovo #539: ThinkPad P1 Gen 4 – Failed to Recover Intel Management Engine (ME) Firmware After Update

MariusCausemann, 2025-09-22:
I own a ThinkPad P1 Gen 4 (Serial Number: PF342PZ7, BIOS Version: N40ET47W (1.29 )).

After installing recent updates using fwupd, my laptop now shows the following message during startup:

“Finding ME firmware… Please do not power off!”
(progress bar/percentage appears)
then → “Failed to recover ME firmware”

The system then boots eventually, but audio isn't working.

I have already tried the following steps without success:

    Updated BIOS and drivers using fwupd

    Installed the latest Intel ME Firmware package from the Lenovo support website

    Performed standard troubleshooting 

Despite these attempts, the issue persists and the ME firmware does not recover.

Based on Lenovo’s own engineering response in https://github.com/fwupd/firmware-lenovo/issues/458 and https://github.com/fwupd/firmware-lenovo/issues/533, it appears that in cases like mine the only solution is to reflash the Intel ME firmware using a USB-based procedure.

Request:
Could you please provide me with the correct USB MEFW reflash package and procedure for my exact system, @kmauleon ?

Thank you!

kmauleon, 2025-09-22:
hi @MariusCausemann thanks for reporting, will forward to ODM and feedback once I hear from them...

kmauleon, 2025-09-26:
sorry have not heard from them yet.. following up...

kmauleon, 2025-09-26:
@MariusCausemann ODM CO owner is on business trip currently... but they will prepare package for USB MEFW update.  Will forward to you once available.  Thank you very much...

kmauleon, 2025-09-29:
@MariusCausemann in the meantime, could you please update your BIOS to latest [1.32](https://download.lenovo.com/pccbbs/mobiles/n40ur29w.iso) via [USB](https://download.lenovo.com/pccbbs/mobiles/n40ur29w.html)

maxarmah, 2025-10-17:
> @pgerber<https://github.com/pgerber> @MariusCausemann<https://github.com/MariusCausemann> @maxarmah<https://github.com/maxarmah> I have the MEFW with me that you can install via EFI Shell... please send me a note
> kmauleon at lenovo dot com

Good morning,

If you have a fix, I would really appreciate it as this issue has been happening for months now.

My email is ***@***.******@***.***>

Thank you!
Max

kmauleon, 2025-10-20:
@maxarmah can you please send me a personal mail... thanks!

@pgerber already sent an email please check.. thanks!

kmauleon, 2025-10-22:
> [@kmauleon](https://github.com/kmauleon) I've sent you an email, please check, thank you!

@MariusCausemann can you please resend it... i can't seem to find it... thank you

Steve-Newcomb, 2025-11-02:
Dear Marius - fwupdmgr's timely warning that my Lenovo P1 Gen4 needed a firmware update prompted me to find your warning, above, that using fwupdmgr to update the machine is dangerous.  I'm grateful to you for posting the red flag.

Instead of using fwupdmgr to update, I booted Windows and used the Lenovo system maintenance stuff to upgrade the ME firmware (and I don't even know what else; I don't normally use Windows).  It was tedious, of course, but at least it worked, eventually, after 3 long update cycles.  But doing that update rendered my existing Debian installation (with whole-disk encryption) a thing of the past; I could not restore it to operability even with "rescue mode".  All my BIOS settings were gone, too.  I had to relearn the BIOS settings the hard way, by crashing into problems as I installed Debian.  

The lesson of the day for Lenovo owners: Reprogramming an "Intel Management Engine" destroys quite a lot of configuration.   At least I didn't have to re-install Windows; Lenovo seems to understand how to preserve Windows.  An older lesson was reinforced: Keep Windows around even if you never use it.  You'll use it for maintenance, whether you want to or not.