fwupd/firmware-lenovo #542: Cant update from version 0

Jamidd, 2025-09-30:
*Describe the bug*
It is reading the version as 0, but on the bios i get the EUFI BIOS version as 1.13 and the EC as 1.07. I have a lenovo P14s gen6 AMD. After updating, I get the error that Not compatible with firmware version 0, requires >= 1.13.

*Steps to Reproduce*
fwupdmgr refresh --force
fwupdmgr get-updates  
fwupdmgr update 
```shell
❯ fwupdmgr refresh --force

Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: 2 local devices supported
❯ fwupdmgr get-updates

Devices with no available firmware updates: 
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • WD BLACK SN850X 1000GB
Devices with the latest available firmware version:
 • Prometheus
LENOVO 21QLCTO1WW
│
└─System Firmware:
  │   Device ID:          5b6cf33a6a9a32930388b488c6cd066c30cee929
  │   Summary:            UEFI System Resource Table device (updated via NVRAM)
  │   Current version:    0
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               5fd8765e-5bd9-4eaa-bb4f-cc22bb1e764a
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  ├─ThinkPad T14/P14s Gen 6 T16/P16s Gen 4 AMD System Update:
  │     New version:      0.1.14
  │     Remote ID:        lvfs
  │     Release ID:       127984
  │     Summary:          Lenovo ThinkPad T14/P14s Gen 6 T16/P16s Gen 4 AMD System Firmware
  │     License:          Proprietary
  │     Size:             69,4 MB
  │     Created:          2021-07-13
  │     Urgency:          Critical
  │     Vendor:           Lenovo
  │     Release Flags:    • Trusted metadata
  │                       • Is upgrade
  │     Description:      
  │     This stable release fixes the following issues:
  │     
  │     • Combine EC and BIOS into single capsule.
  │     • Prepare supporting Power Delivery firmware update feature in single capsule.
  │     • Fixed issue where infinite boot loop when Logitech Brio 305 camera is connected.
  │     Checksum:         a802bddfd004ea3958c07fa98ba6b07f31488e3a439923af05a26c1f457f089a
  │   
  ├─ThinkPad T14/P14s Gen 6 AMD System Update:
  │     New version:      0.1.13
  │     Remote ID:        lvfs
  │     Release ID:       119939
[... 888 lines cut ...]
├── initrd.img.old -> initrd.img-6.14.0-32-generic
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.14.0-1012-oem
├── System.map-6.14.0-32-generic
├── System.map-6.8.0-84-generic
├── vmlinuz -> vmlinuz-6.14.0-1012-oem
├── vmlinuz-6.14.0-1012-oem
├── vmlinuz-6.14.0-32-generic
├── vmlinuz-6.8.0-84-generic
└── vmlinuz.old -> vmlinuz-6.14.0-32-generic

6 directories, 310 files
```

*Additional questions*

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? yes
- Is this a regression?  what do you mean?
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes, i also tried with it off, same.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Boot order lock is turned off

mrhpearson, 2025-09-30:
Thanks for the report.
This is happening because we've moved to a single capsule update, that will update all FW on the system. It causes some issues when transitioning from the non-single capsule to single capsule approach.

To support this transition, when moving from a non-single capsule BIOS to a single capsule-supported BIOS, we release two packages simultaneously: one for the standard BIOS update and one for the single capsule package.

Somehow what has been released isn't working as expected.

I've created internal ticket LO-4025 to get input from the FW team. My concern is that it's a national holiday in China so not sure I'll get an answer until next week

I'm checking this out on my system - will see if I can find the right path to get this unblocked (just need to wait for the battery to charge a bit...)

mrhpearson, 2025-09-30:
Confirmed on my system that I could go from version 0 to "1.14". 
This is currently in the testing channel - and the reason for that is that after the update the FW is reporting 1.2 instead of 1.14 - so it's getting lots of error reports and demoted. The FW team need to fix this :( (but it does work)

Easiest is to download and install https://fwupd.org/downloads/0e518c64ba3d58a5aad4967662978d4784261c8403ff63d37b3de8c4cc21b7f0-r2xul34w.cab

Once you get to 1.14 (which thinks it is 1.2) then you can update normally to 1.15 (which is in the stable channel)

mrhpearson, 2025-10-01:
Great - glad you're unblocked.

I'll keep this open as we do need to solve the 1.14 vs 1.2 issue, so that version can be moved back to the stable channel.

svenpilz, 2025-10-03:
Thanks for the feedback. I can test the stable channel, once it is ready.

mrhpearson, 2025-10-14:
I think we can ignore the 0.1.0 'system' version number for now - this is related to the update to single capsule update which we've not managed to actually do yet.

The BIOS thinks it's still running 1.14 so no idea why it is telling fwupd 1.15....that's a new one :(

Little hesitant to recommend this - but maybe try a forced update of https://fwupd.org/downloads/2e34d26e063a9b1eb599ff1f369384ded3085750dd8a7f6f51aab717a48b1e34-r2xul35w.cab to get fwupdmgr to offer the UEFI update?

I'd wget download it and then 'fwupdmgr local-install <file> --force'

svenpilz, 2025-10-16:
> It prints the loader version (1.8) as well as "Reset System" (both stay for a couple seconds) and does a longer cold (?) reboot.

Is this maybe the firmware update itself secretly complaining about "Not compatible with firmware version 0.1.0, requires >= 0.1.2"?